coolsnowwolf/lede #7863: 没有预设的.config文件下编译错误

Ho-Lay, 2021-09-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：没有预设的.config文件下编译错误，总是在make[3] -C toolchain/gcc/initial compile之后就出现错误

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：make[2] toolchain/compile
 make[3] -C toolchain/gdb compile
 make[3] -C toolchain/binutils compile
 make[3] -C toolchain/fortify-headers compile
 make[3] -C toolchain/gcc/initial compile
make: *** wait: No child processes.  Stop.
make: *** Waiting for unfinished jobs....
make: *** wait: No child processes.  Stop.
Error: The operation was canceled.

110010110, 2021-09-15:
#7757

Ho-Lay, 2021-09-15:
> 关闭吧，这个是github官方导致的

但是假如把以前编译过的config文件直接拿来用是可以很顺利编译成功

110010110, 2021-09-15:
> > 关闭吧，这个是github官方导致的
> 
> 但是假如把以前编译过的config文件直接拿来用是可以很顺利编译成功

Describes the interrupt condition triggered by the online SSH configuration.

Ho-Lay, 2021-09-15:
> > 但是假如把以前编译过的config文件直接拿来用是可以很顺利编译成功
> 
> Describes the interrupt condition triggered by the online SSH configuration.

我编译的是小米3G的固件，默认的插件，编译到 make[3] -C toolchain/gcc/initial compile 就提示出错退出了，后来我再编译小米mini固件，也一样是默认的插件，但还是编译到make[3] -C toolchain/gcc/initial compile就自动退出
编译小米3G和mini肯定是通过SSH才能选择的

后来我就把上个月成功编译的config文件拿来用，小米3G和mini都顺利编译完成， 因为是用了config文件预设了xiaomi3G和mini，所以这个这个就没有连SSH

不知道是因为连了SSH才导致编译中断还是小米3G和mini的默认插件加了哪个插件才导致编译中断的

Ho-Lay, 2021-09-15:
测试确定是github SSH的问题

closed 2021-09-15 by Ho-Lay

leopardciaw, 2021-09-21:
> 测试确定是github SSH的问题

如何解决哈。

110010110, 2021-09-21:
> 
> 如何解决哈。
Get '.config' in local configuration, and then upload the '.config' file.

LarryLi-cn, 2022-12-13:
我也遇到这个问题了, 有解决方吗?