coolsnowwolf/lede #7865: 编译报错，libselinux提示失败，缺少libpcre.so.1

realconnor, 2021-09-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：更新到最新版（R21.9.18）的代码后，不论是用自己以前的.config，还是用make defconfig生成的默认配置，编译（单线程）都会提示libselinux编译失败，缺少libpcre.so.1库。本地机器环境是Ubuntu 18.04，有代理。

### (2) 路由器型号和固件版本

A：x86_64

### (3) 详细日志

A：
_**:~/Downloads/lede$ cat build_output_error.txt**_
...
autoreconf: Leaving directory 
configure: WARNING: unrecognized options: --disable-nls
configure: WARNING: using cross tools not prefixed with host triplet
configure: WARNING: unrecognized options: --disable-nls
/home/connor/Downloads/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ar: `u' modifier ignored since `D' is the default (see `U')
OpenWrt-libtool: install: warning: remember to run `libtool --finish /usr/lib'
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://github.com/SELinuxProject/selinux/releases/download/3.2/libselinux-3.2.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   627  100   627    0     0    728      0 --:--:-- --:--:-- --:--:--   728
100  201k  100  201k    0     0  85920      0  0:00:02  0:00:02 --:--:--  132k
install: error while loading shared libraries: libpcre.so.1: cannot open shared object file: No such file or directory
make[5]: *** [install] Error 127
make[4]: *** [install] Error 1
make[3]: *** [/home/connor/Downloads/lede/build_dir/target-x86_64_musl/libselinux-3.2/.built] Error 2
make[2]: *** [package/libs/libselinux/compile] Error 1
make[1]: *** [/home/connor/Downloads/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make: *** [world] Error 2

**_:~/Downloads/lede$ cat build_output_**normal**.txt_**
...
Makefile:176: recipe for target 'install' failed
make[5]: Leaving directory '/home/connor/Downloads/lede/build_dir/target-x86_64_musl/libselinux-3.2/src'
Makefile:48: recipe for target 'install' failed
make[4]: Leaving directory '/home/connor/Downloads/lede/build_dir/target-x86_64_musl/libselinux-3.2'
Makefile:154: recipe for target '/home/connor/Downloads/lede/build_dir/target-x86_64_musl/libselinux-3.2/.built' failed
make[3]: Leaving directory '/home/connor/Downloads/lede/package/libs/libselinux'
time: package/libs/libselinux/compile#8.75#1.14#11.80
    ERROR: package/libs/libselinux failed to build.
package/Makefile:114: recipe for target 'package/libs/libselinux/compile' failed
make[2]: Leaving directory '/home/connor/Downloads/lede'
package/Makefile:110: recipe for target '/home/connor/Downloads/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: Leaving directory '/home/connor/Downloads/lede'
/home/connor/Downloads/lede/include/toplevel.mk:228: recipe for target 'world' failed

realconnor, 2021-09-16:
换成github action (ubuntu 20.04)就没事了，目前还不知道是什么原因，先关掉了。

closed 2021-09-16 by realconnor

any35, 2021-10-10:
同样的问题, 花了两天都没解决....

any35, 2021-11-14:
建议首页上加个说明, 不支持在Ubuntu 18.04下面编译.