coolsnowwolf/lede #7873: 首次编译直接报错

ChishFoxcat, 2021-09-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

完全依照README的步骤进行编译操作，首次执行 'make menuconfig' 选择路由器对应配置，没有进行任何额外操作直接开始编译，漫长等待报了一堆错误

### (2) 路由器型号和固件版本

Redmi AC2100-RM2100

### (3) 详细日志

make[3]: *** [Makefile:39: /home/ubuntu/openwrt/lede/build_dir/host/cmake-3.19.1/.configured] Error 9
make[3]: Leaving directory '/home/ubuntu/openwrt/lede/tools/cmake'
time: tools/cmake/compile#125.86#17.72#992.29
    ERROR: tools/cmake failed to build.
make[2]: *** [tools/Makefile:159: tools/cmake/compile] Error 1
make[2]: Leaving directory '/home/ubuntu/openwrt/lede'
make[1]: *** [tools/Makefile:155: /home/ubuntu/openwrt/lede/staging_dir/host/stamp/.tools_compile_yyynyynnyyyynyyyyyynyynyyyynyyyyyyyyyyyyyyyyynynnyyyyyyyyyy] Error 2
make[1]: Leaving directory '/home/ubuntu/openwrt/lede'
make: *** [/home/ubuntu/openwrt/lede/include/toplevel.mk:230: world] Error 2

ChishFoxcat, 2021-09-17:
目前为止尝试过各种云编译和本地编译，均有**Error 2**

EnnawYang, 2021-09-17:
哦

closed 2021-09-17 by AmadeusGhost