coolsnowwolf/lede #7874: 编译后上传网件R7500V2显示成功但不生效，求解。

tsasan, 2021-09-17:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译成功，tftp方式传factory img到R7500V2，显示上传成功。但重启后还是原来的系统，没有更新。

### (2) 路由器型号和固件版本

A：路由器型号：网件R7500V2

编译后的img文件60M左右，个人猜测可能是尺寸太大了？如何破解？试了好多种方式，编译后的映像文件都在50M以上。请大佬指导如何减小映像文件，或改变路由器分区大小使之可以容纳这么大的映像。

搜到lean大一个7500V1扩充ubi分区的commit，不知道是不是解决这个问题的途径，如果是，能否指导一下7500V2应该如何改变？谢谢。

hcym, 2021-09-17:
把默认插件去掉，第一次刷可能要tftp刷哪个factroy过渡一下

tsasan, 2021-09-18:
谢谢hcym指导。能去的都去掉了，现在映像大小是50+M。刷了原厂固件，再刷upgrade过渡，不成功，重启后还是原厂固件。使用tftp直接传编译后的factory.img，上传完毕，自动重启，启动后还是原厂固件。

hcym, 2021-09-18:
50几的还真的没用过，我编译的都是二三十，大版本升级需要过渡，小版本直接刷，和7800一样都是不死之身，刷官方原版21.02的，没有luci，这个行的话再说。

tsasan, 2021-09-27:
刷官方原版21.02成功。但是用Lean大源码编译好的upgrade.bin升级，刷入成功，但是重启，重启，无限重启。

tsasan, 2021-09-27:
发现有个编译选项：static build ; dynamic build.映像文件太大是否这个原因？默认选择是static.

tsasan, 2021-09-27:
发现有个编译选项：static build ; dynamic build.映像文件太大是否这个原因？默认选择是static.
我要试验一下动态编译。

tsasan, 2021-10-05:
使用动态编译，映像缩小到30M，成功刷入。

closed 2021-10-05 by tsasan