coolsnowwolf/lede #7876: 2.5GbE USB网卡 RTL8156B 显示正常但实际限制在150-180M之间

loydcheung, 2021-09-18:
2.5GbE USB网卡 RTL8156B，插上网件R7800，上游路由器显示2.5G全双工，本机ethtool也显示速度2500M全双工，但实际工作却限制在150-180M之间，有没有哪位大神遇到过同样的问题，在线等。

loydcheung, 2021-09-18:
kernel是5.4.143版本，驱动理论上都正常，能用只是无线AP速度被限制到很低只有150M左右，换成接本机的千兆口，无线就能跑满带宽。

EnnawYang, 2021-09-18:
确定不是设备usb瓶颈？

ZeaKyX, 2021-09-18:
你让一个双核1.7G的 arm v7 的硬路由跑2.5G网卡？好吧也不是不行

loydcheung, 2021-09-18:
> 确定不是设备usb瓶颈？

感谢回复，R7800的是USB3.0的，我kmod-usb3驱动也都打上了，理论上不该有瓶颈。而且有瓶颈也不该是150Mbps吧，转换成MB只有20M/s不到

loydcheung, 2021-09-18:
> 你让一个双核1.7G的 arm v7 的硬路由跑2.5G网卡？好吧也不是不行

除了开机稍微比不插慢一丢丢，实际速度无法突破170Mbps，其他一切正常

yinzhefuyun, 2021-09-19:
> > 确定不是设备usb瓶颈？
> 
> 感谢回复，R7800的是USB3.0的，我kmod-usb3驱动也都打上了，理论上不该有瓶颈。而且有瓶颈也不该是150Mbps吧，转换成MB只有20M/s不到

有没有肯设备因为兼容性或者接触不良识别出USB2.0

imy7, 2021-10-03:
usb网卡你先看下是不是过热的问题导致的

imy7, 2021-10-03:
> > > 确定不是设备usb瓶颈？
> > 
> > 
> > 感谢回复，R7800的是USB3.0的，我kmod-usb3驱动也都打上了，理论上不该有瓶颈。而且有瓶颈也不该是150Mbps吧，转换成MB只有20M/s不到
> 
> 有没有肯设备因为兼容性或者接触不良识别出USB2.0

我的建议是看下网卡有没有出现过热的现象

loydcheung, 2021-10-05:
> > > > 确定不是设备usb瓶颈？
> 
> > > 
> 
> > > 
> 
> > > 感谢回复，R7800的是USB3.0的，我kmod-usb3驱动也都打上了，理论上不该有瓶颈。而且有瓶颈也不该是150Mbps吧，转换成MB只有20M/s不到
> 
> > 
> 
> > 有没有肯设备因为兼容性或者接触不良识别出USB2.0
> 
> 
> 
> 我的建议是看下网卡有没有出现过热的现象

没有过热问题，正常温度。ethtool￼查看了也是工作在2500全双工的，但实际就是连200兆都突破不了，还不如路由器自带的千兆口。插电脑上用就正常，已经排除设备硬件问题。

closed 2021-11-02 by AmadeusGhost