coolsnowwolf/lede #7877: 用作翻墙路由时一个奇怪的表现

kleisth, 2021-09-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：原来是用作主路由的，后来由于一些原因，只当做翻墙的路由用，发现了一个奇怪的问题。
用原来做主路由编译出来的固件，正常翻墙。在现在的主路由检测流量，当访问为国内时，流量都是设备本身ip出来，当访问国外时，流量都是从翻墙的路由ip出来，就是说不出国就不走翻墙的路由。
但是当特意去编译一个只包含了openclash的固件时，无论访问国内外，流量全是从翻墙路由出来。
我想知道到底是哪个插件导致的，为何之前的固件这么智能，而后编译的全都失去了这样的功能呢？

### (2) 路由器型号和固件版本

A：x86软路由

### (3) 详细日志

A：无日志，日志上啥也没显示

kleisth, 2021-09-18:
有没大神解答下，现在新编译的固件全都不可以了，只有9月15号编译的可以自动分开走
配置都是一样的，就是不知道取消了哪个插件导致的

ismethr, 2021-09-18:
越来越高端了，还有翻墙路由这种说法呢.......主路由啥问题都没有就用主路由，实在不行有需要的设备自己挂，不要套娃。建议 close

kleisth, 2021-09-18:
> 越来越高端了，还有翻墙路由这种说法呢.......主路由啥问题都没有就用主路由，实在不行有需要的设备自己挂，不要套娃。建议 close

反正人总是要死的，建议你早点达成人的这个最终目的

ismethr, 2021-09-18:
> > 越来越高端了，还有翻墙路由这种说法呢.......主路由啥问题都没有就用主路由，实在不行有需要的设备自己挂，不要套娃。建议 close
> 
> 反正人总是要死的，建议你早点达成人的这个最终目的

你可真会说话，祝你中秋节快乐！

kleisth, 2021-09-18:
源码全部删掉后重新下载编译，只包含openclash又可以了，不知道到底是啥插件影响的这个效果，没人回答也懒得管了，只要新固件能达到这个效果就行了

closed 2021-09-18 by kleisth