coolsnowwolf/lede #7879: 9月19日更新代码后，编译报错

tuhailong, 2021-09-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
9月19日更新代码后，编译报错。

### (2) 路由器型号和固件版本
树莓派4B

### (3) 详细日志
+ info LD vmlinux.o
+ '[' '' '!=' silent_ ']'
+ printf '  %-7s %s\n' LD vmlinux.o
  LD      vmlinux.o
+ modpost_link vmlinux.o
+ local objects
+ objects='--whole-archive                                              arch/arm64/kernel/head.o init/built-in.a usr/built-in.a arch/arm64/built-in.a kernel/built-in.a certs/built-in.a mm/built-in.a fs/built-in.a ipc/built-in.a security/built-in.a crypto/built-in.a block/built-in.a arch/arm64/lib/built-in.a lib/built-in.a  arch/arm64/lib/lib.a  lib/lib.a drivers/built-in.a sound/built-in.a net/built-in.a virt/built-in.a                                           --no-whole-archive                                              --start-group           --end-group'
+ aarch64-openwrt-linux-musl-ld -EL -maarch64elf -r -o vmlinux.o --whole-archive arch/arm64/kernel/head.o init/built-in.a usr/built-in.a arch/arm64/built-in.a kernel/built-in.a certs/built-in.a mm/built-in.a fs/built-in.a ipc/built-in.a security/built-in.a crypto/built-in.a block/built-in.a arch/arm64/lib/built-in.a lib/built-in.a arch/arm64/lib/lib.a lib/lib.a drivers/built-in.a sound/built-in.a net/built-in.a virt/built-in.a --no-whole-archive --start-group --end-group
aarch64-openwrt-linux-musl-ld: drivers/built-in.a: member drivers/dma-buf/heaps in archive is not an object
+ on_exit
+ '[' 1 -ne 0 ']'
+ cleanup
+ rm -f '.btf.*'
+ rm -f .tmp_System.map
+ rm -f '.tmp_vmlinux*'
+ rm -f System.map
+ rm -f vmlinux
+ rm -f vmlinux.o
Makefile:1179: recipe for target 'vmlinux' failed
make[5]: *** [vmlinux] Error 1
make[5]: Leaving directory '/home/tuhailong/0_codes/0_rpi4/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.10.64'
Makefile:33: recipe for target '/home/tuhailong/0_codes/0_rpi4/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.10.64/.modules' failed
make[4]: *** [/home/tuhailong/0_codes/0_rpi4/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.10.64/.modules] Error 2
make[4]: Leaving directory '/home/tuhailong/0_codes/0_rpi4/target/linux/bcm27xx'
Makefile:11: recipe for target 'compile' failed
make[3]: *** [compile] Error 2
make[3]: Leaving directory '/home/tuhailong/0_codes/0_rpi4/target/linux'
time: target/linux/compile#587.08#59.20#641.35
    ERROR: target/linux failed to build.
target/Makefile:23: recipe for target 'target/linux/compile' failed
make[2]: *** [target/linux/compile] Error 1
make[2]: Leaving directory '/home/tuhailong/0_codes/0_rpi4'
target/Makefile:18: recipe for target '/home/tuhailong/0_codes/0_rpi4/staging_dir/target-aarch64_cortex-a72_musl/stamp/.target_compile' failed
make[1]: *** [/home/tuhailong/0_codes/0_rpi4/staging_dir/target-aarch64_cortex-a72_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/tuhailong/0_codes/0_rpi4'
/home/tuhailong/0_codes/0_rpi4/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 2

colour008, 2021-09-18:
本人小白入坑一年多，今天编译x86-64时也出现类似的问题，查阅之前的issue相同的问题里也有出现过，L大回复说是官方补丁pack的出现错误导致，所以等一段时间修复完成了再编译试试吧。

lxl6125, 2021-09-18:
内核从5.4升级到5.10，等修复吧，或者你自己改一下源码，继续用5.4内核

asdertypo, 2021-09-19:
我的树莓派3B+也是这样的错误

EnnawYang, 2021-09-19:
```
  HDRINST usr/include/asm/ioctl.h
  INSTALL /home/xw/openwrt/openwrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.10.64/user_headers/include
make[5]: Leaving directory '/home/xw/openwrt/openwrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.10.64'
net/sched/Kconfig:45: warning: menuconfig statement without prompt
net/Kconfig:480:warning: multi-line strings not supported
*
* Restart config...
*
*
* Performance monitor support
*
ARM CCI PMU driver (ARM_CCI_PMU) [N/m/y/?] n
ARM CCN driver support (ARM_CCN) [N/m/y/?] n
Arm CMN-600 PMU support (ARM_CMN) [N/m/y/?] n
ARM PMU framework (ARM_PMU) [Y/n/?] y
ARM DynamIQ Shared Unit (DSU) PMU (ARM_DSU_PMU) [N/m/y/?] n
Enable support for the ARMv8.2 Statistical Profiling Extension (ARM_SPE_PMU) [N/m/y/?] n
RaspberryPi AXI Performance monitors (RPI_AXIPERF) [N/m/y/?] (NEW) 
Error in reading or end of file.

init/do_mounts.c:478:19: warning: 'mount_ubi_rootfs' defined but not used [-Wunused-function]
 static int __init mount_ubi_rootfs(void)
                   ^~~~~~~~~~~~~~~~
kernel/module.c:1288:12: warning: 'try_to_force_load' defined but not used [-Wunused-function]
 static int try_to_force_load(struct module *mod, const char *reason)
            ^~~~~~~~~~~~~~~~~
fs/binfmt_elf.c: In function 'parse_elf_properties.isra.20':
fs/binfmt_elf.c:818:1: warning: the frame size of 1040 bytes is larger than 1024 bytes [-Wframe-larger-than=]
 }
 ^
aarch64-openwrt-linux-musl-ld: drivers/built-in.a: member drivers/dma-buf/heaps in archive is not an object
make[5]: *** [Makefile:1179: vmlinux] Error 1
```

@AmadeusGhost @Beginner-Go 大佬补一补config？（

closed 2021-09-20 by AmadeusGhost