coolsnowwolf/lede #7881: 老笔记本网卡不能识别和驱动的问题！网卡为Realtek PCIe FE Family Controller 和RTL8187SE

hdrafyy, 2021-09-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
老笔记本网卡不能识别和驱动的问题！网卡为Realtek PCIe FE Family Controller 和RTL8187SE
Q：是否搜索了issue (使用 "x" 选择)
* [X] 没有类似的issue

## 2. 详细叙述
编译后的镜像无法识别网卡，但是我在其他的电脑可以正常连接进入后台的，这能证明固件是没有问题的。
### (1) 具体问题
想请问如何编译这两个网卡的驱动到固件里面去？
A：

### (2) 路由器型号和固件版本
上网本的CPU是N270
A：

### (3) 详细日志

A：

Neverends4, 2021-09-22:
无线简单，kmod 里wireless driver选中rt8187
有线要看你的具体型号，去windows设备管理器找硬件ID，FE默认一般是8111/8168之类的，kmod里network devices选中kmod-r8168/8169就行了

EnnawYang, 2021-09-22:
> 无线简单，kmod 里wireless driver选中rt8187
> 有线要看你的具体型号，去windows设备管理器找硬件ID，FE默认一般是8111/8168之类的，kmod里network devices选中kmod-r8168/8169就行了

他在群里问过了 有线无解

hdrafyy, 2021-09-26:
> 无线简单，kmod 里wireless driver选中rt8187
> 有线要看你的具体型号，去windows设备管理器找硬件ID，FE默认一般是8111/8168之类的，kmod里network devices选中kmod-r8168/8169就行了

有线找到了，是r8169,现在是无线找不到。

hdrafyy, 2021-09-26:
现在的问题是无线网卡没有驱动。

hdrafyy, 2021-09-26:
现在的问题是无线网卡没有驱动。在GitHub找到这个https://github.com/freestyl3r/rtl8187se，不知道是不是驱动程序，不知道怎么编译到固件里面去。

hdrafyy, 2021-09-26:
现在的问题是无线网卡没有驱动。在GitHub找到这个https://github.com/freestyl3r/rtl8187se。不知道是不是驱动程序，不知道怎么编译到固件里面去。

hdrafyy, 2021-09-26:
现在的问题是无线网卡没有驱动。在GitHub找到这个：https://github.com/freestyl3r/rtl8187se，                                                      
不知道是不是驱动程序，不知道怎么编译到固件里面去。

hdrafyy, 2021-09-26:
现在的问题是无线网卡没有驱动。在GitHub找到这个[https://github.com/freestyl3r/rtl8187se]不知道是不是驱动程序，不知道怎么编译到固件里面去。

hdrafyy, 2021-09-26:
[现在的问题是无线网卡没有驱动。在GitHub找到这个https://github.com/freestyl3r/rtl8187se，不知道是不是驱动程序，不知道怎么编译到固件里面去。](url)

hdrafyy, 2021-09-26:
现在的问题是无线网卡没有驱动。在GitHub找到这个[https://github.com/freestyl3r/rtl8187se](url)，不知道是不是驱动程序，不知道怎么编译到固件里面去。

hdrafyy, 2021-09-26:
现在的问题是无线网卡没有驱动。在GitHub找到这个[https://github.com/freestyl3r/rtl8187se](url)不知道是不是驱动程序，不知道怎么编译到固件里面去。

hdrafyy, 2021-09-26:
https://github.com/freestyl3r/rtl8187se

hdrafyy, 2021-09-26:
现在的问题是无线网卡无法驱动，在GitHub里找到这个，https://github.com/freestyl3r/rtl8187se；不知道是怎么编译到固件里面去。

hdrafyy, 2021-09-26:
现在的问题是无线网卡无法驱动，在GitHub里找到这个，https://github.com/freestyl3r/rtl8187se；不知道是怎么编译到固件里面去。[https://github.com/freestyl3r/rtl8187se](url)

hdrafyy, 2021-09-26:
现在的问题是无线网卡无法驱动，在GitHub里面找到这个https://github.com/freestyl3r/rtl8187se
不知道怎么编译到固件里面去。

closed 2021-11-02 by AmadeusGhost