coolsnowwolf/lede #7885: arm-qemu版本编译成功,但是lan接口请求异常

wszgrcy, 2021-09-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
[image]
如图,已经修改了reresolv,加了网关等,目前可以Ping外网网址和Ip

### (2) 路由器型号和固件版本

A：跑在docker中 rk3399 Linux magicsee-n6-max 5.10.20-rockchip64 #trunk SMP PREEMPT Sun Mar 7 10:37:12 CST 2021 aarch64 aarch64 aarch64 GNU/Linux

### (3) 详细日志

A：请求接口报500 http://192.168.31.68/cgi-bin/luci/admin/network/iface_status/lan?_=0.8312354115307148
curl
```
curl 'http://192.168.31.68/cgi-bin/luci/admin/network/iface_status/lan?_=0.8312354115307148' \
  -H 'Connection: keep-alive' \
  -H 'User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/93.0.4577.82 Safari/537.36' \
  -H 'DNT: 1' \
  -H 'Accept: */*' \
  -H 'Referer: http://192.168.31.68/cgi-bin/luci/admin/network/network' \
  -H 'Accept-Language: zh-CN,zh;q=0.9,ja;q=0.8' \
  -H 'Cookie: sysauth=a77dcb4ee61a95bcf0af92974ce89361' \
  --compressed \
  --insecure
```

返回值
```
/usr/lib/lua/luci/dispatcher.lua:469: Failed to execute call dispatcher target for entry '/admin/network/iface_status/lan'.
The called action terminated with an exception:
/usr/lib/lua/luci/model/network.lua:930: bad argument #2 to '?' (no value)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:469: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>
```
- 最小化编译,进去后就是这样,编译成功无报错,不知道啥原因

wszgrcy, 2021-10-06:
有点迷惑.同样的配置x86正常
然后用了原始的openwrt编译,也是能获得到.......
然后目前来说还在用2月份的固件(自己编译的......)
<html>
<body>

OpenWrt R21.2.1 / LuCI Master (git-21.035.34289-13f96d1)
--

<br class="Apple-interchange-newline">
</body>
</html>
不知道是哪里出现的问题........

coolsnowwolf, 2026-09-06:
该报告使用旧 Lua LuCI 接口状态页面并出现 network.lua:930 异常；当前 LuCI 已改用新版 JavaScript 网络状态页面，问题已被后续实现覆盖，关闭为 completed。

closed 2026-09-06 by coolsnowwolf