coolsnowwolf/lede #7888: 最新版openwrt中uu加速器无法使用

yuyongji1987, 2021-09-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：闲来无事，编译了最新版op，发现uu加速器启动之后，switch无法联网，关闭uu加速器之后，switch可以正常联网。

### (2) 路由器型号和固件版本

A：x86软路由
固件版本  OpenWrt R21.9.18 / LuCI Master (git-21.238.35254-83494a9)
内核版本 | 5.4.145

### (3) 详细日志

A：日志中无uu相关log。防火墙中一条如下记录，其中192.168.100.191是switch的ip，不知道有无参考意义：
数据包 | 流量 | 对象 | 协议 | 入口 | 出口 | 源地址 | 目标地址 | 选项
0 | 0.00 B | DROP | udp | br-lan | * | 192.168.100.191 | 0.0.0.0/0 | udp dpt:1900

NagaseKouichi, 2021-09-20:
下面是我编译的2.16.2新版本的IPK，你可以安装试试有没有问题

链接: https://pan.baidu.com/s/1VrcVelBiZmKgrz6mHcOIAw 提取码: h9h8

我已经在L大的github提交了新版本的RP，等L大合并。

yuyongji1987, 2021-09-21:
> 下面是我编译的2.16.2新版本的IPK，你可以安装试试有没有问题
> 
> 链接: https://pan.baidu.com/s/1VrcVelBiZmKgrz6mHcOIAw 提取码: h9h8
> 
> 我已经在L大的github提交了新版本的RP，等L大合并。

谢谢 我试一下

yuyongji1987, 2021-09-22:
> 下面是我编译的2.16.2新版本的IPK，你可以安装试试有没有问题
> 
> 链接: https://pan.baidu.com/s/1VrcVelBiZmKgrz6mHcOIAw 提取码: h9h8
> 
> 我已经在L大的github提交了新版本的RP，等L大合并。

刚才测试了一下，这个版本也是不行，启动加速器之后，switch就无法正常联网，关闭加速器，就可以正常联网。。。

NagaseKouichi, 2021-09-23:
那你可能是设置有问题或者和其他插件冲突了，我用着一直没问题

yuyongji1987, 2021-09-24:
> 那你可能是设置有问题或者和其他插件冲突了，我用着一直没问题。只是以前老的版本经常会出现不认switch的情况，这个新版会不会有这个问题还在观察。

你装ssr plus了吗？可能是和某些插件有冲突

NagaseKouichi, 2021-09-24:
装了的

NagaseKouichi, 2021-09-24:
SSR里面的游戏模式UDP中继服务器我是关了的

yuyongji1987, 2021-09-24:
> SSR里面的游戏模式UDP中继服务器我是关了的

我ssr直接都是关闭的。老哥你是在Lean最新的系统下测试的吗？我5月份编译的那个openwrt是能用的，最近闲着没事用最新的代码编译了，就开始出现断网的问题。。。

yuyongji1987, 2021-09-24:
今天重新编译了L大的openwrt，uu好用了！

closed 2021-09-24 by yuyongji1987

LEGION128, 2022-11-03:
你好请问一下  就是旁路由我开了uu加速switch 但是也想开openclash进行翻墙看油管 会有冲突吗？