coolsnowwolf/lede #7901: R2S编译不成功，重装系统，默认配置也不成功

supppig, 2021-09-21:
log如下：

echo "# CONFIG_KALLSYMS_EXTRA_PASS is not set" >> /home/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.145/.config.target
echo "# CONFIG_KALLSYMS_ALL is not set" >> /home/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.145/.config.target
echo "CONFIG_KALLSYMS_UNCOMPRESSED=y" >> /home/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.145/.config.target
/home/lede/lede/scripts/package-metadata.pl kconfig /home/lede/lede/tmp/.packageinfo /home/lede/lede/.config 5.4 > /home/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.145/.config.override
/home/lede/lede/scripts/kconfig.pl 'm+' '+' /home/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.145/.config.target /dev/null /home/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.145/.config.override > /home/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.145/.config.set
mv /home/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.145/.config.set /home/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.145/.config.old
grep -v INITRAMFS /home/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.145/.config.old > /home/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.145/.config.set
echo 'CONFIG_INITRAMFS_SOURCE=""' >> /home/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.145/.config.set
echo '# CONFIG_INITRAMFS_FORCE is not set' >> /home/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.145/.config.set
rm -rf /home/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/modules
cmp -s /home/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.145/.config.set /home/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.145/.config.prev || { cp /home/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.145/.config.set /home/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.145/.config; cp /home/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.145/.config.set /home/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.145/.config.prev; }
export MAKEFLAGS= ; [ -d /home/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.145/user_headers ] || make -C /home/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.145 KCFLAGS="-fmacro-prefix-map=/home/lede/lede/build_dir/target-aarch64_generic_musl=target-aarch64_generic_musl" HOSTCFLAGS="-O2 -I/home/lede/lede/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Mon Sep 20 03:24:56 2021" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/lede/lede/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/lede/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.4.145 CC="aarch64-openwrt-linux-musl-gcc" INSTALL_HDR_PATH=/home/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.145/user_headers headers_install
grep '=[ym]' /home/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.145/.config.set | LC_ALL=C sort | /home/lede/lede/staging_dir/host/bin/mkhash md5 > /home/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.145/.vermagic
touch /home/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.145/.configured
rm -f /home/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.145/vmlinux /home/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.145/System.map
make -C /home/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.145 KCFLAGS="-fmacro-prefix-map=/home/lede/lede/build_dir/target-aarch64_generic_musl=target-aarch64_generic_musl" HOSTCFLAGS="-O2 -I/home/lede/lede/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Mon Sep 20 03:24:56 2021" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/lede/lede/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/lede/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.4.145 CC="aarch64-openwrt-linux-musl-gcc" Image dtbs modules
make[5]: Entering directory '/home/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.145'
  CALL    scripts/checksyscalls.sh
  CALL    scripts/atomic/check-atomics.sh
  CHK     include/generated/compile.h
  GEN     .version
  CHK     include/generated/compile.h
  LD      vmlinux.o
  MODPOST vmlinux.o
  MODINFO modules.builtin.modinfo
  LD      .tmp_vmlinux.kallsyms1
  KSYM    .tmp_vmlinux.kallsyms1.o
  LD      .tmp_vmlinux.kallsyms2
  KSYM    .tmp_vmlinux.kallsyms2.o
  LD      vmlinux
  SORTEX  vmlinux
  SYSMAP  System.map
  OBJCOPY arch/arm64/boot/Image
  Building modules, stage 2.
  MODPOST 107 modules
make[5]: Leaving directory '/home/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.145'
touch /home/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.145/.modules
make -C image compile TARGET_BUILD=
make[5]: Entering directory '/home/lede/lede/target/linux/rockchip/image'
cp -fpR /home/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.145/COPYING /home/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/COPYING.linux
make[5]: Leaving directory '/home/lede/lede/target/linux/rockchip/image'
make[4]: Leaving directory '/home/lede/lede/target/linux/rockchip'
make[3]: Leaving directory '/home/lede/lede/target/linux'
time: target/linux/compile#13.08#8.88#21.13
make[2]: Leaving directory '/home/lede/lede'
export MAKEFLAGS= ;make -w -r buildinfo
make[2]: Entering directory '/home/lede/lede'
 make[3] diffconfig buildversion feedsversion
make[2]: Leaving directory '/home/lede/lede'
make[2]: Entering directory '/home/lede/lede'
rm -rf /home/lede/lede/staging_dir/target-aarch64_generic_musl/root-rockchip
make[2]: Leaving directory '/home/lede/lede'
make[2]: Entering directory '/home/lede/lede'
make[3]: Entering directory '/home/lede/lede/package/libs/libjson-c'
mkdir -p /home/lede/lede/dl
SHELL= flock /home/lede/lede/tmp/.json-c-0.15-nodoc.tar.gz.flock -c '  	/home/lede/lede/scripts/download.pl "/home/lede/lede/dl" "json-c-0.15-nodoc.tar.gz" "99bca4f944b8ced8ae0bbc6310d6a3528ca715e69541793a1ef51f8c5b4b0878" "" "https://s3.amazonaws.com/json-c_releases/releases/"    '
make[3]: Leaving directory '/home/lede/lede/package/libs/libjson-c'
time: package/libs/libjson-c/host-compile#0.19#0.08#0.23
make[3]: Entering directory '/home/lede/lede/package/libs/libubox'
mkdir -p /home/lede/lede/dl
SHELL= flock /home/lede/lede/tmp/.libubox-2021-05-16-b14c4688.tar.xz.flock -c '  	/home/lede/lede/scripts/download.pl "/home/lede/lede/dl" "libubox-2021-05-16-b14c4688.tar.xz" "7dd1db1e0074a9c7c722db654cce3111b3bd3cff0bfd791c4497cb0f6c22d3ca" "" || (  	echo "Checking out files from the git repository..."; mkdir -p /home/lede/lede/tmp/dl && cd /home/lede/lede/tmp/dl && rm -rf libubox-2021-05-16-b14c4688 && [ \! -d libubox-2021-05-16-b14c4688 ] && git clone  https://git.openwrt.org/project/libubox.git libubox-2021-05-16-b14c4688 && (cd libubox-2021-05-16-b14c4688 && git checkout b14c4688612c05c78ce984d7bde633bce8703b1e && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd libubox-2021-05-16-b14c4688 && git log -1 --format='\''@%ct'\''` && rm -rf libubox-2021-05-16-b14c4688/.git && 	tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c libubox-2021-05-16-b14c4688 | 	xz -zc -7e > /home/lede/lede/tmp/dl/libubox-2021-05-16-b14c4688.tar.xz && mv /home/lede/lede/tmp/dl/libubox-2021-05-16-b14c4688.tar.xz /home/lede/lede/dl/ && rm -rf libubox-2021-05-16-b14c4688;  )    '
make[3]: Leaving directory '/home/lede/lede/package/libs/libubox'
[... cut ...]
    ERROR: package/libs/toolchain failed to build.
make[2]: *** [package/Makefile:116: package/libs/toolchain/compile] Error 1
make[2]: Leaving directory '/home/lede/lede'
make[1]: *** [package/Makefile:110: /home/lede/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lede/lede'
make: *** [/home/lede/lede/include/toplevel.mk:230: world] Error 2

貌似是编译toolchain就出错了。但看了半天没发现怎么能修复~~~
已经重新安装了 ubuntu20.04.3，按照教程做了一遍，全局科学，依然报这个错误。。。
请大神们帮忙看看，谢谢啦~~

supppig, 2021-09-21:
报一下进展：似乎是这句报错了：
/home/lede/lede/staging_dir/host/bin/fakeroot /home/lede/lede/scripts/ipkg-build -m "" /home/lede/lede/build_dir/target-aarch64_generic_musl/toolchain/ipkg-aarch64_generic/libgcc /home/lede/lede/bin/targets/rockchip/armv8/packages

于是执行
cd /home/lede/lede/staging_dir/host/bin/
./fakeroot
/home/lede/lede/scripts/ipkg-build -m "" /home/lede/lede/build_dir/target-aarch64_generic_musl/toolchain/ipkg-aarch64_generic/libgcc /home/lede/lede/bin/targets/rockchip/armv8/packages
提示：
`
/home/lede/lede/scripts/ipkg-build: 33: cannot open CONTROL/control: Permission denied
/home/lede/lede/scripts/ipkg-build: 33: cannot open CONTROL/control: Permission denied
/home/lede/lede/scripts/ipkg-build: 33: cannot open CONTROL/control: Permission denied
sed: can't read /home/lede/lede/build_dir/target-aarch64_generic_musl/toolchain/ipkg-aarch64_generic/libgcc/CONTROL/control: Permission denied
`

于是cd进去报错的目录（fakeroot的状态下进入的）
cd /home/lede/lede/build_dir/target-aarch64_generic_musl/toolchain/ipkg-aarch64_generic/libgcc/CONTROL/
ls -l
输出：
`
total 12
---------- 1 root root 370 Sep 21 06:13 control
-rwxr-xr-x 1 root root 160 Sep 21 00:15 postinst
-rwxr-xr-x 1 root root 117 Sep 21 00:15 prerm
`

似乎是权限问题？control文件没有任何权限。

Her0R0cket, 2021-09-22:
你之前是不是用root用户编译过?

supppig, 2021-09-22:
> 你之前是不是用root用户编译过?

没有，我全新装了一个Ubuntu20.04.3（官方下载的最新安装iso），全新的。然后新建了一个叫lede的账户，除了apt instll用了root，从git clone到编译，一直都用这个账户。
刚刚尝试了云编译，同一个config文件，云编译居然过了！！！
试过换了一个梯子，用美国的节点，还是不行。。
到底什么问题啊。。。。

开始怀疑是不是VMware的问题了（按道理没理由啊！）

supppig, 2021-09-22:
[image]
换了WSL2+Ubuntu20 还是这样，吐血。。。
为啥云编译能过，本地编译不能过啊，啊啊啊啊啊啊啊啊~~

showme2005, 2021-09-23:
我之前也遇到过你这样的问题 但不是提示这个错误..我也是看了很多贴子..有一个人说过.什么出错在编译的之前先把你出错的插件单独编译一次．．make package/feeds/luci/luci-base/compile V=99      luci-base换成你出错的插件名字~
然后在执行make -j8 download V=s和make -j1 V=s

supppig, 2021-09-23:
> 我之前也遇到过你这样的问题 但不是提示这个错误..我也是看了很多贴子..有一个人说过.什么出错在编译的之前先把你出错的插件单独编译一次．．make package/feeds/luci/luci-base/compile V=99 luci-base换成你出错的插件名字~
> 然后在执行make -j8 download V=s和make -j1 V=s

感谢，试过了。仍然不成功。

supppig, 2021-09-23:
结束这个issue了，原因找到了，太羞耻了 T_T
写出来，引以为鉴吧~~

我的ubuntu为了全局翻墙，写了个小脚本，作用是启动一个redsocks，然后用iptables把tcp redirect到redsocks的监听端口，然后转发到翻墙的socks5代理。这是前提。

但是，我写脚本的时候，过于简陋，iptables部分就写了2句：
iptables -tnat -A OUTPUT -d 192.168/16 -j ACCEPT
iptables -tnat -A OUTPUT -p tcp -j REDIRECT --to 2222

于是，问题就出来了：
编译过程中产生了127.x.x.x的内环流量，ipdtables把这部分流量也redirect了！
于是，编译产生了错了。。。

这也解释了为什么云编译能成功，本地编译一直失败。
后来完善了这个脚本，补充了全部的内环网段，问题得以解决。

closed 2021-09-23 by supppig