coolsnowwolf/lede #7907: 编译失败，config/install-sh: Command not found

1368129224, 2021-09-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue
* [x] 搜索了，但解决方法不适用

## 2. 详细叙述

### (1) 具体问题

A：编译失败，提示`config/install-sh: Command not found`

### (2) 路由器型号和固件版本

A：k2p，目前最新代码

### (3) 详细日志

A：

```
making install in e2fsck
make[5]: Entering directory '/home/zzc/share/lede/build_dir/host/e2fsprogs-1.45.6/e2fsck'
        MKDIR_P /home/zzc/share/lede/staging_dir/host/bin /home/zzc/share/lede/staging_dir/host/share/man/man8
make[5]: config/install-sh: Command not found
make[5]: *** [Makefile:522: installdirs] Error 127
make[5]: Leaving directory '/home/zzc/share/lede/build_dir/host/e2fsprogs-1.45.6/e2fsck'
make[4]: *** [Makefile:433: install-progs-recursive] Error 1
make[4]: Leaving directory '/home/zzc/share/lede/build_dir/host/e2fsprogs-1.45.6'
make[3]: *** [Makefile:54: /home/zzc/share/lede/staging_dir/host/stamp/.e2fsprogs_installed] Error 2
make[3]: Leaving directory '/home/zzc/share/lede/tools/e2fsprogs'
time: tools/e2fsprogs/compile#0.09#0.03#0.13
    ERROR: tools/e2fsprogs failed to build.
make[2]: *** [tools/Makefile:159: tools/e2fsprogs/compile] Error 1
make[2]: Leaving directory '/home/zzc/share/lede'
make[1]: *** [tools/Makefile:155: /home/zzc/share/lede/staging_dir/host/stamp/.tools_compile_yyynyynnyyyynyyyyyynyynyyyynyyyyyyyyyyyyyyyyynynnyyyyyyyyyy] Error 2
make[1]: Leaving directory '/home/zzc/share/lede'
make: *** [/home/zzc/share/lede/include/toplevel.mk:230: world] Error 2
```

wuxinna, 2021-09-23:
我和你一样的错误 求解决一下

fengqi, 2021-09-23:
重新来一次吧，大概率是下载的文件不完整

1368129224, 2021-09-24:
使用了proxychains4代理，并且尝试了Ubuntu18和Ubuntu20，都是以上错误。望遇过此问题的大佬解答一下

1368129224, 2021-09-27:
确认是网络问题，使用proxychains-ng还是会有漏网之鱼，个人使用的是虚拟机编译，配合clash网卡全局代理，已编译成功。

1368129224, 2021-09-27:
> 我和你一样的错误 求解决一下

已确认是网络问题。

closed 2021-09-27 by 1368129224