coolsnowwolf/lede #7912: 关于5.4.140正常，5.4.142以上无法正常使用的问题

rongcenter, 2021-09-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

1、esxi+openwrt
2、wan口：正常
3、lan口：2口，vmxnet3（eth1）和直通82576（eth2）
4、有种情况可以正常使用
开机后，编辑network文件，去掉eth2，正常运行，重启后正常无报错。再编辑network加入eth2，重启后eth1和eth2正常无报错，再次重启问题重现，eth2口不通。

完全看不懂！！之前以为mac冲突了，才发现有点关系都没有。

### (2) 路由器型号和固件版本

x86

### (3) 详细日志

A：#7745

chenxuan5858, 2021-09-24:
一样问题

rongcenter, 2021-09-25:
> 一样问题

这应该是上游的问题，跟L大这边无关的，但是没办法，不知道在哪反馈。

alepennyq, 2021-09-27:
> 反馈bug/问题模板，提建议请删除
> 
> ## 1.关于你要提交的问题
> Q：是否搜索了issue (使用 "x" 选择)
> 
> * [x]  没有类似的issue
> 
> ## 2. 详细叙述
> ### (1) 具体问题
> 1、esxi+openwrt
> 2、wan口：正常（eth0）
> 3、lan口：2口，vmxnet3（eth1）和直通82576（eth2）
> 4、有种情况可以正常使用，但是不能重启
> 开机后，编辑network文件，去掉eth2，重启后正常无报错，再编辑network加入eth2，重启后eth1和eth2正常无报错，再次重启问题重现，eth2口不通。
> 
> 完全看不懂！！之前以为mac冲突了，才发现一点关系都没有。5.10.60以上情况一样。
> 
> ### (2) 路由器型号和固件版本
> x86
> 
> ### (3) 详细日志
> A：#7745

一样的，只要有直通口都不行，只能虚拟

kollcn, 2021-11-11:
同样的问题，希望能解决。

kollcn, 2021-11-11:
我是eth1和eth2是直通口，eth1接了网线到光猫，一开机就不断重启接口，无法拨号。

kollcn, 2021-11-13:
有人解决了虚拟机直通网卡工作不正常这个问题吗？

chenxuan5858, 2021-11-15:
> 有人解决了虚拟机直通网卡工作不正常这个问题吗？

前几天尝试编译了，但还是不行，我看到一个issues里提到得手动替换驱动

closed 2021-11-30 by AmadeusGhost