coolsnowwolf/lede #7914: 运营商释放IP后，有V4V6但断网，把WAN重新连接才能恢复

laaArmy, 2021-09-24:
[https://z3.ax1x.com/2021/09/24/4B8Eh6.jpg](url)
[https://z3.ax1x.com/2021/09/24/4B8e1O.jpg](url)
[断网日志.txt](https://github.com/coolsnowwolf/lede/files/7222902/default.txt)

592732377, 2021-09-24:
一样的问题，存在很久了，每48小时强制下线重新拨号，偶发拨号后无法联网，最长不重启记录是8天左右。不过搭配watchcat基本无感！

[image]

Thu Sep 23 22:00:12 2021 daemon.info pppd[16946]: LCP terminated by peer
Thu Sep 23 22:00:12 2021 daemon.info pppd[16946]: Connect time 2875.1 minutes.
Thu Sep 23 22:00:12 2021 daemon.info pppd[16946]: Sent 2347004791 bytes, received 1381737466 bytes.
Thu Sep 23 22:00:12 2021 daemon.err odhcp6c[17145]: Failed to send RS (Permission denied)
Thu Sep 23 22:00:12 2021 daemon.err odhcp6c[17145]: Failed to send RELEASE message to ff02::1:2 (Permission denied)
Thu Sep 23 22:00:12 2021 daemon.notice netifd: Network device 'pppoe-WAN' link is down
Thu Sep 23 22:00:12 2021 daemon.notice netifd: Network alias 'pppoe-WAN' link is down
Thu Sep 23 22:00:12 2021 daemon.notice netifd: Interface 'IPV6' has link connectivity loss
Thu Sep 23 22:00:12 2021 daemon.notice netifd: Interface 'WAN_6' has link connectivity loss
Thu Sep 23 22:00:12 2021 daemon.notice pppd[16946]: Modem hangup
Thu Sep 23 22:00:12 2021 daemon.notice pppd[16946]: Connection terminated.
Thu Sep 23 22:00:12 2021 daemon.info pppd[16946]: Connect time 2875.1 minutes.
Thu Sep 23 22:00:12 2021 daemon.info pppd[16946]: Sent 2347004791 bytes, received 1381737466 bytes.
Thu Sep 23 22:00:12 2021 daemon.notice netifd: Interface 'WAN' has lost the connection
Thu Sep 23 22:00:12 2021 daemon.notice netifd: WAN_6 (17145): Command failed: Permission denied
Thu Sep 23 22:00:12 2021 daemon.info pppd[16946]: Sent PADT
Thu Sep 23 22:00:12 2021 daemon.info pppd[16946]: Exit.
Thu Sep 23 22:00:12 2021 daemon.notice netifd: Interface 'WAN' is now down
Thu Sep 23 22:00:12 2021 daemon.notice netifd: Interface 'WAN' is disabled
Thu Sep 23 22:00:12 2021 daemon.notice netifd: Interface 'WAN' is enabled
Thu Sep 23 22:00:12 2021 daemon.notice netifd: Interface 'WAN' is setting up now
Thu Sep 23 22:00:12 2021 daemon.err insmod: module is already loaded - slhc
Thu Sep 23 22:00:12 2021 daemon.err insmod: module is already loaded - ppp_generic
Thu Sep 23 22:00:12 2021 daemon.err insmod: module is already loaded - pppox
Thu Sep 23 22:00:12 2021 daemon.err insmod: module is already loaded - pppoe
Thu Sep 23 22:00:12 2021 daemon.notice netifd: WAN (28311): uci: Entry not found
Thu Sep 23 22:00:12 2021 daemon.notice netifd: WAN (28311): sh: out of range
Thu Sep 23 22:00:12 2021 daemon.info pppd[28354]: Plugin rp-pppoe.so loaded.
Thu Sep 23 22:00:12 2021 daemon.info pppd[28354]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
Thu Sep 23 22:00:13 2021 daemon.notice pppd[28354]: pppd 2.4.8 started by root, uid 0
Thu Sep 23 22:00:13 2021 daemon.info pppd[28354]: PPP session is 19327
Thu Sep 23 22:00:13 2021 daemon.warn pppd[28354]: Connected to 48:d5:39:42:f6:53 via interface eth0.246
Thu Sep 23 22:00:13 2021 kern.info kernel: [345017.472514] pppoe-WAN: renamed from ppp0
Thu Sep 23 22:00:13 2021 daemon.info pppd[28354]: Renamed interface ppp0 to pppoe-WAN
Thu Sep 23 22:00:13 2021 daemon.info pppd[28354]: Using interface pppoe-WAN
Thu Sep 23 22:00:13 2021 daemon.notice pppd[28354]: Connect: pppoe-WAN <--> eth0.246
Thu Sep 23 22:00:13 2021 daemon.info pppd[28354]: syncppp not active
Thu Sep 23 22:00:13 2021 daemon.err odhcp6c[17128]: Failed to send SOLICIT message to ff02::1:2 (Permission denied)
Thu Sep 23 22:00:13 2021 daemon.notice netifd: Interface 'IPV6' is now down
Thu Sep 23 22:00:13 2021 daemon.info pppd[28354]: Remote message: Authentication success,Welcome!
Thu Sep 23 22:00:13 2021 daemon.notice pppd[28354]: PAP authentication succeeded
Thu Sep 23 22:00:13 2021 daemon.notice pppd[28354]: peer from calling number 48:D5:39:42:F6:53 authorized
Thu Sep 23 22:00:13 2021 daemon.notice pppd[28354]: local  LL address fe80::bdec:dbc3:ab2b:af19
Thu Sep 23 22:00:13 2021 daemon.notice netifd: Network device 'pppoe-WAN' link is up
Thu Sep 23 22:00:13 2021 daemon.notice netifd: Network alias 'pppoe-WAN' link is up
Thu Sep 23 22:00:13 2021 daemon.notice pppd[28354]: remote LL address fe80::4ad5:39ff:fe42:f653
Thu Sep 23 22:00:13 2021 daemon.notice netifd: Interface 'IPV6' has link connectivity
Thu Sep 23 22:00:13 2021 daemon.notice netifd: Interface 'IPV6' is setting up now
Thu Sep 23 22:00:13 2021 daemon.notice netifd: Interface 'WAN_6' has link connectivity
Thu Sep 23 22:00:13 2021 daemon.err odhcp6c[17145]: Failed to send SOLICIT message to ff02::1:2 (Permission denied)
Thu Sep 23 22:00:13 2021 daemon.notice netifd: Interface 'WAN_6' is now down
Thu Sep 23 22:00:13 2021 daemon.notice netifd: Interface 'WAN_6' has link connectivity loss
Thu Sep 23 22:00:13 2021 daemon.notice pppd[28354]: local  IP address 10.0.24.183
Thu Sep 23 22:00:13 2021 daemon.notice pppd[28354]: remote IP address 10.0.0.1
Thu Sep 23 22:00:13 2021 daemon.notice pppd[28354]: primary   DNS address 61.128.128.68
Thu Sep 23 22:00:13 2021 daemon.notice pppd[28354]: secondary DNS address 61.128.192.68
Thu Sep 23 22:00:13 2021 daemon.notice netifd: Interface 'WAN' is now up
Thu Sep 23 22:00:13 2021 daemon.notice netifd: Interface 'WAN_6' is enabled
Thu Sep 23 22:00:13 2021 daemon.notice netifd: Interface 'WAN_6' has link connectivity
Thu Sep 23 22:00:13 2021 daemon.notice netifd: Interface 'WAN_6' is setting up now
Thu Sep 23 22:00:15 2021 daemon.notice netifd: Interface 'WAN_6' is now up
Thu Sep 23 22:00:17 2021 daemon.info watchcat[4781]: Could not reach 114.114.114.114 for 30. Rebooting after reaching 180
Thu Sep 23 22:00:23 2021 daemon.notice netifd: Interface 'IPV6' is now up
[... cut ...]
Thu Sep 23 22:01:28 2021 daemon.info dnsmasq[30831]: Connected to system UBus
Thu Sep 23 22:01:47 2021 daemon.info watchcat[4781]: Could not reach 114.114.114.114 for 120. Rebooting after reaching 180
Thu Sep 23 22:02:17 2021 daemon.info watchcat[4781]: Could not reach 114.114.114.114 for 150. Rebooting after reaching 180
Thu Sep 23 22:02:47 2021 daemon.info watchcat[4781]: Could not reach 114.114.114.114 for 180. Rebooting after reaching 180
Thu Sep 23 22:02:47 2021 daemon.info procd: - shutdown -

laaArmy, 2021-09-24:
> 一样的问题，出现了，每48小时强制下线重拨号，偶发拨号后无法上网，不重启记录是8天左右。
> 
> [image]
> 
> 2021 年 9 月 23 日星期四 22:00:12 daemon.info pppd[16946]：LCP 被对等端终止
> 2021 年 9 月 23 日星期四 22:00:12 daemon.info pppd[16946]：连接时间 2875.1 分钟。
> 2021 年 9 月 23 日星期四 22:00:12 daemon.info pppd[16946]：发送 2347004791 字节，收到 1381737466 字节。
> 2021 年 9 月 23 日星期四 22:00:12 daemon.err odhcp6c[17145]：无法发送 RS（权限被拒绝）
> 星期四 2021 年 9 月 23 日 22:00:12 daemon.err odhcp6c[17145]：无法将 RELEASE：消息发送到:1:2（权限被拒绝）
> 2021 年 9 月 23 日星期四 22:00:12 daemon.notice netifd：网络设备“pppoe-WAN”链接已关闭
> 2021 年 9 月 23 日星期四 22:00:12 daemon.notice netifd：网络别名“pppoe” -WAN' 链接已关闭
> 2021 年 9 月 23 日星期四 22:00:12 daemon.notice netifd：接口“IPV6”有链接连接丢失
> 2021 年 9 月 23 日星期四 22:00:12 daemon.notice netifd：接口“WAN_6”的链路连接丢失
> 星期四 9 月 23 日 22:00:12 2021 年 daemon.notice pppd[16946]：调制解调器挂断
> 9 月 23 日星期四 22:00:12 daemon.notice pppd[16946]：连接终止。
> 2021 年 9 月 23 日星期四 22:00:12 daemon.info pppd[16946]：连接时间 2875.1 分钟。
> 2021 年 9 月 23 日星期四 22:00:12 daemon.info pppd[16946]：发送 2347004791 字节，收到 1381737466 字节。
> 2021 年 9 月 23 日星期四 22:00:12 daemon.notice netifd：接口“WAN”已失去连接
> 2021 年 9 月 23 日星期四 22:00:12 daemon.notice netifd：WAN_6（17145）：命令失败：
> 22 年 9 月 23 日星期四被拒绝:00:12 2021 daemon.info pppd[16946]：发送 PADT
> 星期四 9 月 23 日 22:00:12 2021 daemon.info pppd[16946]：退出。
> 2021 年 9 月 23 日星期四 22:00:12 daemon.notice netifd：“WAN”接口现已关闭
> 2021 年 9 月 23 日星期四 22:00:12 daemon.notice netifd：“WAN”接口已禁用
> 2021 年 9 月 23 日星期四 22:00:12 daemon.notice netifd：接口“WAN”已启用
> 2021 年 9 月 23 日星期四 22:00:12 daemon.notice netifd：现在正在设置“WAN”接口
> 星期四 2021 年 9 月 23 日 22:00:12 daemon.err insmod：模块已经已加载 - slhc
> 2021 年 9 月 23 日周四 22:00:12 daemon.err insmod：模块已加载 - ppp_generic
> Thu Sep 23 22:00:12 2021 daemon.err insmod：模块已加载 - pppox
> Thu Sep 23 22:00： 12 2021 daemon.err insmod: 模块已经加载 - pppoe
> Thu Sep 23 22:00:12 2021 daemon.notice netifd: WAN (28311): uci: Entry not found
> 2021 年 9 月 23 日星期四 22:00:12 daemon.notice netifd：WAN（28311）：sh：超出范围
> 2021 年 9 月 23 日星期四 22:00:12 daemon.info pppd[28354]：插件 rp-pppoe.so 已加载。
> 2021 年 9 月 23 日星期四 22:00:12 daemon.info pppd[28354]：针对 pppd 2.4.8 编译的 RP-PPPoE 插件版本
> 3.8p 2021 年 9 月 23 日星期四 22:00:13 2021 daemon.notice pppd[28352.4 pppd[28352.4]: 8 由 root 启动，uid 0
> Thu Sep 23 22:00:13 2021 daemon.info pppd[28354]: PPP session is 19327
> Thu Sep 23 22:00:13 2021 daemon.warn pppd[28354]: Connected to :39:42:f6:53 通过接口 eth0.246
> Thu Sep 23 22:00:13 2021 kern.info 内核：[345017.472514] pppoe-WAN：从 ppp0
> Thu Sep 23 22:00:13 2021 pppda重命名[28354]：将接口 ppp0 重命名为 pppoe-WAN
> Thu Sep 23 22:00:13 2021 daemon.info pppd[28354]：使用接口 pppoe-WAN
> Thu Sep 23 22:00:13 2021 daemon.notice pppd[28354]：连接：pppoe-WAN <--> eth0。 246
> Thu Sep 23 22:00:13 2021 daemon.info pppd[28354]：syncppp 未激活
> Thu Sep 23 22:00:13 2021 daemon.err odhcp6c[17128]：无法将 SOLICIT 消息发送到 2ff02 （权限被拒绝）
> 2021 年 9 月 23 日星期四 22:00:13 daemon.notice netifd：接口“IPV6”现已关闭
> 星期四 9 月 23 日 22:00:13 2021 daemon.info pppd[28354]：远程消息：身份验证成功，欢迎！
> 2021 年 9 月 23 日星期四 22:00:13 daemon.notice pppd[28354]：PAP 身份验证成功
> 2021 年 9 月 23 日星期四 22:00:13 daemon.notice pppd[28354]：来自呼叫号码 48:D5:39:42:F6 的对等方:53 授权
> 2021 年 9 月 23 日星期四 22:00:13 daemon.notice pppd[28354]：本地 LL 地址 fe80::bdec:dbc3:ab2b:
> af19 2021 年 9 月 23 日星期四 22:00:13 daemon.notice netifd：网络设备 'pppoe ' 链接
> 已于 2021 年 9 月 23 日星期四 22:00:13 daemon.notice netifd：网络别名 'pppoe-WAN' 链接已启动
> 2021 年 9 月 23 日星期四 22:00:13 daemon.notice pppd[28354]：远程 LL 地址 fe80： :4ad5:39ff:fe42:f653
> 2021 年 9 月 23 日星期四 22:00:13 daemon.notice netifd：接口“IPV6”具有链接连接
> 2021 年 9 月 23 日星期四 22:00:13 daemon.notice netifd：接口“IPV6”正在设置现在
> 2021 年 9 月 23 日星期四 22:00:13 daemon.notice netifd：接口“WAN_6”具有链接连接
> 星期四 9 月 23 日 22:00:13 2021 年 daemon.err odhcp6c[17145]：无法向 ff02::1 发送 SOLICIT 消息： 2（权限被拒绝）
> 2021 年 9 月 23 日星期四 22:00:13 daemon.notice netifd：接口“WAN_6”现已关闭
> 2021 年 9 月 23 日星期四 22:00:13 daemon.notice netifd：接口“WAN_6”于
> 9 月 23 日星期四 22:00失去链接连​​接： 13 2021 daemon.notice pppd[28354]: 本地 IP 地址 10.0.24.183
> Thu Sep 23 22:00:13 2021 daemon.notice pppd[28354]: 远程 IP 地址 10.0.0.1
> Thu : 020212da通知 pppd[28354]：主 DNS 地址 61.128.128.68
> Thu Sep 23 22:00:13 2021 daemon.notice pppd[28354]：辅助 DNS 地址 61.128.192.68
> Thu Sep 23 22:002113 “WAN”现已启动
> 2021 年 9 月 23 日星期四 22:00:13 daemon.notice netifd：接口“WAN_6”已启用
> 2021 年 9 月 23 日星期四 22:00:13 daemon.notice netifd：接口“WAN_6”有链路连通性
> 2021 年 9 月 23 日星期四 22:00:13 daemon.notice netifd：接口 'WAN_6' 正在设置 2021 年 9 月 23 日星期四 22:00:15 daemon.notice netifd：接口 'WAN_6' 现在正在设置
> 9 月 23 日
> 星期四 22:00： 17 2021 daemon.info watchcat [4781]：到达后重新引导30.无法达到114.114.114.114 180
> 星期四09月23日22时○○分23秒2021 daemon.notice netifd：接口'IPV6'现在是向上
> 星期四09月23日22:00 :28 2021 user.notice firewall: 由于 WAN (pppoe-WAN) 的 ifup 重新加载防火墙
> 2021 年 9 月 23 日星期四 22:00:28 cron.err crond[28937]: crond (busybox 1.33.1) 启动，日志级别 8
> Thu 2021 年 9 月 23 日 22:00:28 user.notice dnsmasq：DNS 重新绑定保护处于活动状态，将丢弃上游 RFC1918 响应！
[... 4 lines cut ...]
> 2021 年 9 月 23 日星期四 22:00:34 daemon.info dnsmasq[29351]：连接到系统 UBus
> 星期四：03 年 4 月 22 日2021 daemon.info watchcat [4781]：达到180后重新引导60.无法达到114.114.114.114
> 防火墙周四9月23日22时00分55秒2021 user.notice：重新加载防火墙由于ifup的WAN_6（PPPoE的-WAN）的
> 周四09月23 22:00:55 2021 cron.err crond[29685]：crond (busybox 1.33.1) 开始，日志级别 8
> 2021 年 9 月 23 日星期四 22:00:55 user.notice dnsmasq：DNS 重新绑定保护处于活动状态，将丢弃上游 RFC1918 响应！
> 2021 年 9 月 23 日星期四 22:00:55 user.notice dnsmasq：允许 127.0.0.0/8 响应
> 2021 年 9 月 23 日星期四 22:00:58 daemon.info dnsmasq[29859]：连接到系统 UBus 2021 年 9 月 22 日星期四：03:
> 58 2021 user.notice dnsmasq：DNS 重新绑定保护处于活动状态，将丢弃上游 RFC1918 响应！
[... cut ...]
> 星期四：012 年9 月 22 日：22 2021 user.notice dnsmasq：DNS 重新绑定保护处于活动状态，将丢弃上游 RFC1918 响应！
> 2021 年 9 月 23 日星期四 22:01:26 user.notice dnsmasq：允许 127.0.0.0/8 响应
> 2021 年 9 月 23 日星期四 22:01:28 daemon.info dnsmasq[30831]：连接到系统 UBus
> 2021 年 9 月 23 日星期四 22:01:47 daemon.info watchcat[4781]：120 无法到达 114.114.114.114。到达 180 后重新启动
> 2021 年 9 月 23 日星期四 22:02:17 daemon.info watchcat]：[47 无法到达114.114.114.114 为 150。到达 180 后重启
> Thu Sep 23 22:02:47 2021 daemon.info watchcat[4781]: 无法到达 114.114.114.114 为 180。到达后重启：
> 9 月 23 日 9 月 23 日 10:02 .info procd: - 关闭 -

不知道是不是错觉，之前我把WAN6删除后好长一段时间都没遇到过

592732377, 2021-09-24:
这个问题和IPV6有关是已经确定了的，不起用IPV6一切都正常。

laaArmy, 2021-09-24:
> 这个问题和IPV6有关是已经确定了的，不起用IPV6一切都正常。

不，我实测把WAN6删除后，IPV6还是正常的

Neverends4, 2021-09-24:
为什么我从来没遇到过，WAN6，联通每7天/10800分钟刷一次IP，V4 V6都可以自动重新连上。
不过我是硬路由

db-one, 2021-09-24:
确实是老问题了，我自己是遇到过两三次，断网重播以后是正常获取到ip的，但就是无法联网，不清楚是路由问题还是运营商问题，反正重新拨号以后就好了，我的解决方法是自己写了个脚本，添加到计划任务里，每小时运行一次检测一下网络，如果网络不通就延迟20秒再次检测，还不通就直接重启网络
或者直接按楼上的直接编译watchcat搞定

Suwipe, 2021-09-26:
同样遇到这问题，看到别人建议写了个hotplug脚本自动触发检测是否获取到了IPV6地址，但是又出现新问题，最近是V4 V6都获取到IP了，但是路由表上没有v4的0.0.0.0和v6的::0路由，还是要手动重连一次WAN，我的倒是不需要重启，直接在接口界面WAN点一下连接就行，这个问题太烦了，虽然V6说起来应用不太多，但是看nlbwmon里统计的ipv6网络流量占地也有25%了