coolsnowwolf/lede #7921: UU

skaterjuice, 2021-09-24:
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.openwrt.org/uugamebooster-x86_64-v2.16.2.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:02 --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.

Her0R0cket, 2021-09-25:
https://router.uu.163.com/api/plugin?type=openwrt-x86_64

NagaseKouichi, 2021-09-25:
怎么在从openwrt的服务器上下载uu

AmadeusGhost, 2021-09-25:
> 怎么在从openwrt的服务器上下载uu

都是先从总服务器下载的，总服务器没有再去 pkg_source_url 那边下载。

NagaseKouichi, 2021-09-25:
> > 怎么在从openwrt的服务器上下载uu
> 
> 都是先从总服务器下载的，总服务器没有再去 pkg_source_url 那边下载。

原来如此，pkg_source_url 应该是没问题的