coolsnowwolf/lede #7926: Dnspod.cn 的 DDNS错误且提示原因为空

freeedm, 2021-09-24:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：添加dnspod 的DDNS 提示错误但无错误原因明细，仅显示 “ERROR : Error message:[]”

### (2) 路由器型号和固件版本

A：N1 ：5.4.144-flippy-63+o

### (3) 详细日志

A： 045417  note : PID '31704' started at 2021-09-25 04:54
 045417       : ddns version  : 2.7.8-3
 045417       : uci configuration:
ddns.dnspodv6.cacert='IGNORE'
ddns.dnspodv6.domain='n1.frdm.xyz'
ddns.dnspodv6.enabled='1'
ddns.dnspodv6.force_ipversion='1'
ddns.dnspodv6.interface='LAN6'
ddns.dnspodv6.ip_network='LAN6'
ddns.dnspodv6.lookup_host='n1.frdm.xyz'
ddns.dnspodv6.password='***PW***'
ddns.dnspodv6.retry_count='50'
ddns.dnspodv6.service_name='dnspod.cn'
ddns.dnspodv6.use_https='1'
ddns.dnspodv6.use_ipv6='1'
ddns.dnspodv6.use_syslog='1'
ddns.dnspodv6.username='260648'
ddns.dnspodv6=service
 045419       : verbose mode  : 0 - run normal, NO console output
 045419       : check interval: 600 seconds
 045420       : force interval: 259200 seconds
 045420       : retry interval: 60 seconds
 045420       : retry counter : 50 times
 045420       : No old process
 045420       : last update: never
 045420       : Detect registered/public IP
 045420       : #> /usr/bin/host -t AAAA -6 n1.frdm.xyz  >/var/run/ddns/dnspodv6.dat 2>/var/run/ddns/dnspodv6.err
 045421  WARN : NO valid IP found
 045421  WARN : Error in 'expand_ipv6()' - invalid IPv6 found: '' expanded: ''
 045421  info : Starting main loop at 2021-09-25 04:54
 045422       : Detect local IP on 'network'
 045422       : Local IP '2408:824e:2905:*********' detected on network 'LAN6'
 045422       : Update needed - L: '2408:824e:2905:bb80:*********' <> R: 'invalid'
 045423       : parsing script '/usr/lib/ddns/update_dnspod_cn.sh'
 045423       : #> wget --no-hsts -nv -t 1 -O /var/run/ddns/dnspodv6.dat -o /var/run/ddns/dnspodv6.err -6 --no-check-certificate --no-proxy --post-data 'login_token=260648,***PW***&format=json&domain=frdm.xyz&sub_domain=n1&record_type=AAAA' https://dnsapi.cn/Record.List
 045424 ERROR : Error message:[]
 045424  WARN : Transfer failed - 1 Try again in 60 seconds

hxlls, 2021-09-26:
去到对应固件的地方提交问题啊。。源码都不是一个

coolsnowwolf, 2026-09-06:
当前 ddns-scripts 已更新到 2.8.2，Dnspod 脚本使用 curl 并解析 API 返回的 status.message，错误日志不再固定显示空消息；旧版 2.7.8 的问题已被现行实现覆盖，关闭为 completed。

closed 2026-09-06 by coolsnowwolf