coolsnowwolf/lede #7927: TP C2600 210923版本不停重启

hzjnet, 2021-09-24:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
源码版本：
1 commit 61f4bdeb2b85689b1bd61f33f5aa81e03fd3d9d9 (HEAD)
Merge: 50d98983c 3b48d25c4
Author: lean <coolsnowwolf@gmail.com>
Date:   Thu Sep 23 14:48:06 2021 +0800
2  commit 368e31dc7c5022f98abd2bcf2a0eedc0f1fc37bb (HEAD -> master, origin/master, origin/HEAD)
Author: breakings <skygmin@gmail.com>
Date:   Thu Sep 23 22:22:07 2021 +0800

    dnsmasq: update to version 2.86 (#7909)
    
    Co-authored-by: breakings <skygmin@yahoo.com>

硬件是：TP C2600 ipq8064 RAM 512M SPI 32M
 
### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志
U-Boot 2012.07 [Standard IPQ806X.LN,unknown] (Dec 16 2015 - 15:12:29)

smem ram ptable found: ver: 0 len: 5
DRAM:  491 MiB
PCI0 Link Intialized
PCI1 Link Intialized
SF: Detected MX25U25635F with page size 4 KiB, total 32 MiB
  00:01.0     - 17cb:0101 - Bridge device
   01:00.0    - 168c:0040 - Network controller
  02:01.0     - 17cb:0101 - Bridge device
   03:00.0    - 168c:0040 - Network controller
NAND:  ipq_nand: unknown NAND device manufacturer: 0 device: 0
ipq_nand: failed to identify device
SF: Detected MX25U25635F with page size 4 KiB, total 32 MiB
ipq_spi: page_size: 0x100, sector_size: 0x1000, size: 0x2000000
32 MiB
MMC:   
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   MAC1 addr:0:3:7f:ba:db:1
athrs17_reg_init: complete
athrs17_vlan_config ...done
S17c init  done
MAC2 addr:0:3:7f:ba:db:2
eth0, eth1
boot in 2 seconds
FirmwareRecovery: Now doing bootipq
MMC Device 0 not found
MMC Device 0 not found

Loading from nand1, offset 0x1f0000
   Image Name:   ARM OpenWrt Linux-5.4.145
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2947619 Bytes = 2.8 MiB
   Load Address: 42208000
[... 201 lines cut ...]
[    2.563474] s2b: supplied by regulator-dummy
[      2.581278] /dev/root: Can't open blockdev
[    2.581303] VFS: Cannot open root device "(null)" or unknown-block(31,12): error -6
[    2.584258] Please append a correct "root=" boot option; here are the available partitions:
[    2.591923] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(31,12)
[    2.600230] CPU1: stopping
[    2.608836] CPU: 1 PID: 0 Comm: swapper/1 Not tainted 5.4.145 #0
[    2.611348] Hardware name: Generic DT based system
[    2.617539] [<c030f954>] (unwind_backtrace) from [<c030b968>] (show_stack+0x14/0x20)
[    2.622138] [<c030b968>] (show_stack) from [<c090e4d8>] (dump_stack+0x94/0xa8)
[    2.630030] [<c090e4d8>] (dump_stack) from [<c030eb80>] (handle_IPI+0x184/0x1b8)
[    2.637071] [<c030eb80>] (handle_IPI) from [<c05dff24>] (gic_handle_irq+0xb4/0xb8)
[    2.644616] [<c05dff24>] (gic_handle_irq) from [<c0301a8c>] (__irq_svc+0x6c/0x90)
[    2.651977] Exception stack(0xdd46bf18 to 0xdd46bf60)
[    2.659534] bf00:                                                       00000000 00000000
[    2.664596] bf20: 1d05d000 ddb9ea80 dd7edc00 00000001 ddb9de30 00000000 00000000 00000000
[    2.672757] bf40: 9afc05e0 99bbf820 00000015 dd46bf68 c0734548 c073454c 80000013 ffffffff
[    2.680912] [<c0301a8c>] (__irq_svc) from [<c073454c>] (cpuidle_enter_state+0x94/0x498)
[    2.689062] [<c073454c>] (cpuidle_enter_state) from [<c0734994>] (cpuidle_enter+0x30/0x4c)
[    2.696872] [<c0734994>] (cpuidle_enter) from [<c034ad84>] (do_idle+0x1d8/0x240)
[    2.705199] [<c034ad84>] (do_idle) from [<c034b094>] (cpu_startup_entry+0x1c/0x20)
[    2.712749] [<c034b094>] (cpu_startup_entry) from [<423024cc>] (0x423024cc)
[    2.720127] Rebooting in 1 seconds.

A：

1582130940, 2021-09-26:
把这个commit revert掉再试试
https://github.com/coolsnowwolf/lede/commit/56e6c4b8660f6aa5709e6a2efe941c848e47de1a

hzjnet, 2021-09-28:
确实如此，R7800和C2600都正常了 谢谢

closed 2021-10-08 by AmadeusGhost