coolsnowwolf/lede #7928: kernel 5.4.145 incompatible/lack of dependency issue for some luci-app/collected-mod plugin on x86 platform

GreatMichaelLee, 2021-09-25:
Please refer to the following output on console:

Configuring luci-i18n-nlbwmon-zh-cn.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.145-1-6cef6182e7afcb86ea5b037c505eb1de) for kmod-nls-base
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nls-base found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-fs-antfs:
 *      kernel (= 5.4.145-1-6cef6182e7afcb86ea5b037c505eb1de)
 * opkg_install_cmd: Cannot install package kmod-fs-antfs.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.145-1-6cef6182e7afcb86ea5b037c505eb1de) for kmod-nls-utf8
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nls-utf8 found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.145-1-6cef6182e7afcb86ea5b037c505eb1de) for kmod-crypto-hash
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-crypto-hash found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.145-1-6cef6182e7afcb86ea5b037c505eb1de) for kmod-crypto-md4
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-crypto-md4 found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.145-1-6cef6182e7afcb86ea5b037c505eb1de) for kmod-crypto-md5
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-crypto-md5 found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.145-1-6cef6182e7afcb86ea5b037c505eb1de) for kmod-crypto-null
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-crypto-null found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.145-1-6cef6182e7afcb86ea5b037c505eb1de) for kmod-crypto-aead
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-crypto-aead found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.145-1-6cef6182e7afcb86ea5b037c505eb1de) for kmod-crypto-manager
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-crypto-manager found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.145-1-6cef6182e7afcb86ea5b037c505eb1de) for kmod-crypto-hmac
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-crypto-hmac found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.145-1-6cef6182e7afcb86ea5b037c505eb1de) for kmod-crypto-arc4
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-crypto-arc4 found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.145-1-6cef6182e7afcb86ea5b037c505eb1de) for kmod-crypto-ecb
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-crypto-ecb found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.145-1-6cef6182e7afcb86ea5b037c505eb1de) for kmod-crypto-des
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-crypto-des found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.145-1-6cef6182e7afcb86ea5b037c505eb1de) for kmod-crypto-sha256
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-crypto-sha256 found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.145-1-6cef6182e7afcb86ea5b037c505eb1de) for kmod-crypto-cmac
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-crypto-cmac found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.145-1-6cef6182e7afcb86ea5b037c505eb1de) for kmod-crypto-sha512
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-crypto-sha512 found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.145-1-6cef6182e7afcb86ea5b037c505eb1de) for kmod-crypto-rng
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-crypto-rng found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.145-1-6cef6182e7afcb86ea5b037c505eb1de) for kmod-crypto-seqiv
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-crypto-seqiv found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.145-1-6cef6182e7afcb86ea5b037c505eb1de) for kmod-crypto-ctr
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-crypto-ctr found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.145-1-6cef6182e7afcb86ea5b037c505eb1de) for kmod-crypto-ccm
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-crypto-ccm found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.145-1-6cef6182e7afcb86ea5b037c505eb1de) for kmod-crypto-gf128
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-crypto-gf128 found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.145-1-6cef6182e7afcb86ea5b037c505eb1de) for kmod-crypto-ghash
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-crypto-ghash found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.145-1-6cef6182e7afcb86ea5b037c505eb1de) for kmod-crypto-gcm
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-crypto-gcm found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.145-1-6cef6182e7afcb86ea5b037c505eb1de) for kmod-crypto-crc32c
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-crypto-crc32c found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.145-1-6cef6182e7afcb86ea5b037c505eb1de) for kmod-lib-crc32c
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-lib-crc32c found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-fs-ksmbd:
 *      kernel (= 5.4.145-1-6cef6182e7afcb86ea5b037c505eb1de)
 * opkg_install_cmd: Cannot install package kmod-fs-ksmbd.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.145-1-6cef6182e7afcb86ea5b037c505eb1de) for kmod-crypto-authenc
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-crypto-authenc found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-cryptodev:
 *      kernel (= 5.4.145-1-6cef6182e7afcb86ea5b037c505eb1de)
 * opkg_install_cmd: Cannot install package kmod-cryptodev.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.145-1-6cef6182e7afcb86ea5b037c505eb1de) for kmod-dax
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-dax found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-dax.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.145-1-6cef6182e7afcb86ea5b037c505eb1de) for kmod-dm
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-dm found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-dm.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.145-1-6cef6182e7afcb86ea5b037c505eb1de) for kmod-scsi-core
[... 644 lines cut ...]
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for wireguard-tools:
 *      kernel (= 5.4.145-1-6cef6182e7afcb86ea5b037c505eb1de)
 * opkg_install_cmd: Cannot install package wireguard-tools.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-sqm:
 *      kernel (= 5.4.145-1-6cef6182e7afcb86ea5b037c505eb1de)
 * opkg_install_cmd: Cannot install package luci-app-sqm.
[... cut ...]
        But that file is already provided by package  * luci-app-ipsec-vpnd
 * opkg_install_cmd: Cannot install package strongswan-ipsec.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:111: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:230：world] error 2
builder@Build-Server:~/lede_x86$

AmadeusGhost, 2021-09-25:
Just `make clean` and rebuild.

GreatMichaelLee, 2021-09-25:
> Just `make clean` and rebuild.

OK, thanks. will try

closed 2021-09-25 by coolsnowwolf