coolsnowwolf/lede #7938: luci-app-turboacc会反复修改“忽略解析文件”设置

SakuraFallingMad, 2021-09-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：luci-app-turboacc在重启时或者应用设置时，会修改dnsmasq中“忽略解析文件”设置，需求该设置的用户，需要每次重启openwrt或者修改turboacc选项配置时，需重新在DHCP/DNS中选中“忽略解析文件”选项并应用。

### (2) 路由器型号和固件版本

A：x64 OpenWrt R21.9.18 / LuCI Master (git-21.248.43158-92397bb)

### (3) 详细日志

A：
[image]
上图为重启或应用turboacc设置时，“忽略解析文件”选项被复原。

WYC-2020, 2021-10-07:
这个修改一下他的文件，我是直接去掉他的
[image]
这样以后如果你要用它的那个dns的话 你就需要手动去设置忽略解析文件了

smdx, 2021-10-09:
> 这个修改一下他的文件，我是直接去掉他的 [image] 这样以后如果你要用它的那个dns的话 你就需要手动去设置忽略解析文件了

你好，最近一直被这个问题困扰。请问修改的文件路径是？谢谢

smdx, 2021-10-09:
找到这个选项删除了，重启“忽略解析文件”不会被修改了。就是DNS 查询缓存的大小还是会修改，请问这项在那里啊？谢谢

siren202101, 2021-10-17:
> 找到这个选项删除了，重启修改“理解文件”就不会被了。就是DNS查询的大小还是会修改，请问在那里啊？谢谢

仁兄，这个配置文件在哪里？叫什么名字？

smdx, 2021-10-17:
> > 找到这个选项删除了，重启修改“理解文件”就不会被了。就是DNS查询的大小还是会修改，请问在那里啊？谢谢
> 
> 仁兄，这个配置文件在哪里？叫什么名字？

etc/init.d/turboacc

xulizheng66, 2021-11-29:
> 仁兄，这个配置文件在哪里？叫什么名字？

你找到这个了吗

xulizheng66, 2021-11-29:
> 找到这个选项删除了，重启“忽略解析文件”不会被修改了。就是DNS 查询缓存的大小还是会修改，请问这项在那里啊？谢谢
我删除了还是会自动被修改

dfldylan, 2022-06-09:
i have the same issue

sooxt98, 2022-12-31:
> vim /etc/rc.local
把这两行塞进去
``` bash
sed -i "s:option resolvfile '/tmp/resolv.conf.d/resolv.conf.auto':option noresolv '1':g" /etc/config/dhcp
sed -i "s:option noresolv '0'::g" /etc/config/dhcp
```

coolsnowwolf, 2026-09-05:
当前 `luci-app-turboacc` 脚本和 RPC 实现已不再写入 dnsmasq 的 `resolvfile`/忽略解析文件选项，旧 R21.9.18 行为已被当前代码移除，关闭为 completed。

closed 2026-09-05 by coolsnowwolf