coolsnowwolf/lede #7939: 编译后tproxy模块丢失

nJhEqnvK, 2021-09-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译并安装固件后，无法使用`iptables -t mangle`相关命令（无报错，但不能正常转发给xray，相同配置在p2w_r619ac及cm520-79f上测试过没问题），检查发现疑似`kmod-ipt-tproxy`与`iptables-mod-tproxy`未被正常编译，make menuconfig都已勾选上述包

### (2) 路由器型号和固件版本

A：
1. p2w_r619ac-128m (20210801 的代码能正常编译tproxy，20210925 的代码编译后tproxy丢失)
2. xiaomi_ax3600 (20210831 编译后tproxy丢失)
3. x86_64 (20210921 编译后tproxy丢失)
### (3) 详细日志

A：一共五个文件，四个config.buildinfo，一个.config

<details>
  <summary>1. p2w_r619ac-128m 在 20210801 编译得到的 config.buildinfo ，tproxy正常</summary>

```
CONFIG_TARGET_ipq40xx=y
CONFIG_TARGET_MULTI_PROFILE=y
CONFIG_TARGET_DEVICE_ipq40xx_DEVICE_mobipromo_cm520-79f=y
CONFIG_TARGET_DEVICE_ipq40xx_DEVICE_mobipromo_cm520-79f=y
CONFIG_TARGET_DEVICE_ipq40xx_DEVICE_p2w_r619ac-128m=y
CONFIG_DEFAULT_luci-app-pptp-server=y
CONFIG_LIBCURL_COOKIES=y
CONFIG_LIBCURL_FILE=y
CONFIG_LIBCURL_FTP=y
CONFIG_LIBCURL_HTTP=y
CONFIG_LIBCURL_NO_SMB="!"
CONFIG_LIBCURL_OPENSSL=y
CONFIG_LIBCURL_PROXY=y
CONFIG_PACKAGE_6in4=y
CONFIG_PACKAGE_curl=y
CONFIG_PACKAGE_ip-full=y
# CONFIG_PACKAGE_ip-tiny is not set
CONFIG_PACKAGE_ip6tables=y
CONFIG_PACKAGE_iperf3=y
CONFIG_PACKAGE_iptables-mod-tproxy=y
CONFIG_PACKAGE_ipv6helper=y
CONFIG_PACKAGE_kmod-crypto-lib-blake2s=y
CONFIG_PACKAGE_kmod-crypto-lib-chacha20=y
CONFIG_PACKAGE_kmod-crypto-lib-chacha20poly1305=y
CONFIG_PACKAGE_kmod-crypto-lib-curve25519=y
CONFIG_PACKAGE_kmod-crypto-lib-poly1305=y
CONFIG_PACKAGE_kmod-gre=y
CONFIG_PACKAGE_kmod-ip6tables=y
CONFIG_PACKAGE_kmod-ipt-nat6=y
CONFIG_PACKAGE_kmod-ipt-tproxy=y
CONFIG_PACKAGE_kmod-iptunnel=y
CONFIG_PACKAGE_kmod-l2tp=y
CONFIG_PACKAGE_kmod-lib-lzo=y
CONFIG_PACKAGE_kmod-nf-ipt6=y
CONFIG_PACKAGE_kmod-nf-nat6=y
CONFIG_PACKAGE_kmod-nf-reject6=y
CONFIG_PACKAGE_kmod-pppol2tp=y
CONFIG_PACKAGE_kmod-sit=y
CONFIG_PACKAGE_kmod-udptunnel4=y
CONFIG_PACKAGE_kmod-udptunnel6=y
CONFIG_PACKAGE_kmod-wireguard=y
CONFIG_PACKAGE_kmod-zram=y
CONFIG_PACKAGE_libbpf=y
[... 420 lines cut ...]
# CONFIG_PACKAGE_shadowsocksr-libev-ssr-redir is not set
# CONFIG_PACKAGE_strongswan is not set
CONFIG_PACKAGE_wireguard-tools=y
# CONFIG_PACKAGE_zerotier is not set
CONFIG_PACKAGE_zram-swap=y
CONFIG_PACKAGE_kmod-tun=y
CONFIG_PACKAGE_tcping=y
CONFIG_PACKAGE_xray-core=y
# CONFIG_PACKAGE_xray-example is not set
# CONFIG_PACKAGE_xray-geodata is not set
# CONFIG_XRAY_CORE_COMPRESS_GOPROXY is not set
CONFIG_XRAY_CORE_COMPRESS_UPX=y

```
</details>

feeds中引入了两个custom源，之前都是正常的：
```
src-git small https://github.com/kenzok8/small
src-git kenzo https://github.com/kenzok8/openwrt-packages
```
还想麻烦大佬看一看是咋回事，谢谢！

（p2w_r619ac-128m 在 20210925 编译时的config太大，请见附件）
[config.txt](https://github.com/coolsnowwolf/lede/files/7231249/config.txt)

nJhEqnvK, 2021-09-26:
重新编译了xiaomi_ax3600的固件，这次没有添加第三方源，编译后有tproxy，但无法正常使用
```bash
root@OpenWrt:~# ls /lib/modules/5.10.64/ | grep tproxy
nf_tproxy_ipv4.ko
nf_tproxy_ipv6.ko

root@OpenWrt:~# modprobe -v xt_TPROXY
xt_TPROXY is already loaded

root@OpenWrt:~# lsmod | grep -i tproxy
nf_defrag_ipv4         16384  3 xt_socket,xt_TPROXY,nf_conntrack
nf_defrag_ipv6         20480  3 xt_socket,xt_TPROXY,nf_conntrack
nf_tproxy_ipv4         16384  1 xt_TPROXY
nf_tproxy_ipv6         16384  1 xt_TPROXY
x_tables               32768 38 xt_FULLCONENAT,ipt_REJECT,xt_time,xt_tcpudp,xt_state,xt_socket,xt_quota,xt_policy,xt_pkttype,xt_physdev,xt_owner,xt_nat,xt_multiport,xt_mark,xt_mac,xt_limit,xt_esp,xt_conntrack,xt_comment,xt_cgroup,xt_addrtype,xt_TPROXY,xt_TCPMSS,xt_REDIRECT,xt_MASQUERADE,xt_LOG,xt_CT,iptable_raw,iptable_mangle,iptable_filter,ipt_ah,ip_tables,xt_set,ip6t_NPT,ip6table_mangle,ip6table_filter,ip6_tables,ip6t_REJECT
xt_TPROXY              16384  0

root@OpenWrt:~# iptables -t mangle -L
Chain PREROUTING (policy ACCEPT)
target     prot opt source               destination
XRAY       all  --  anywhere             anywhere

Chain INPUT (policy ACCEPT)
target     prot opt source               destination

Chain FORWARD (policy ACCEPT)
target     prot opt source               destination
TCPMSS     tcp  --  anywhere             anywhere             tcp flags:SYN,RST/SYN /* !fw3: Zone wan MTU fixing */ TCPMSS clamp to PMTU
TCPMSS     tcp  --  anywhere             anywhere             tcp flags:SYN,RST/SYN /* !fw3: Zone wan MTU fixing */ TCPMSS clamp to PMTU

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination

Chain POSTROUTING (policy ACCEPT)
target     prot opt source               destination

Chain XRAY (1 references)
target     prot opt source               destination
RETURN     all  --  anywhere             0.0.0.0/8
RETURN     all  --  anywhere             10.0.0.0/8
RETURN     all  --  anywhere             100.64.0.0/10
RETURN     all  --  anywhere             127.0.0.0/8
RETURN     all  --  anywhere             169.254.0.0/16
RETURN     all  --  anywhere             172.16.0.0/12
RETURN     all  --  anywhere             192.0.0.0/24
RETURN     all  --  anywhere             192.0.2.0/24
RETURN     all  --  anywhere             192.88.99.0/24
RETURN     all  --  anywhere             198.18.0.0/15
RETURN     all  --  anywhere             198.51.100.0/24
RETURN     all  --  anywhere             203.0.113.0/24
RETURN     all  --  anywhere             base-address.mcast.net/4
RETURN     all  --  anywhere             240.0.0.0/4
RETURN     all  --  anywhere             255.255.255.255
RETURN     tcp  --  anywhere             192.168.0.0/16       tcp dpt:!3389
RETURN     udp  --  anywhere             192.168.0.0/16       udp dpt:!3389
RETURN     all  --  anywhere             anywhere             mark match 0xff
TPROXY     udp  --  anywhere             anywhere             TPROXY redirect 0.0.0.0:12345 mark 0x1/0xffffffff
TPROXY     tcp  --  anywhere             anywhere             TPROXY redirect 0.0.0.0:12345 mark 0x1/0xffffffff
```

tproxy转发脚本如下：
```bash
#!/bin/sh

# 设置策略路由
ip route add local default dev lo table 100
ip rule add fwmark 1 lookup 100

# 代理局域网设备
iptables -t mangle -N XRAY

iptables -t mangle -A XRAY -d 0.0.0.0/8 -j RETURN
iptables -t mangle -A XRAY -d 10.0.0.0/8 -j RETURN
iptables -t mangle -A XRAY -d 100.64.0.0/10 -j RETURN
iptables -t mangle -A XRAY -d 127.0.0.0/8 -j RETURN
iptables -t mangle -A XRAY -d 169.254.0.0/16 -j RETURN
iptables -t mangle -A XRAY -d 172.16.0.0/12 -j RETURN
iptables -t mangle -A XRAY -d 192.0.0.0/24 -j RETURN
iptables -t mangle -A XRAY -d 192.0.2.0/24 -j RETURN
iptables -t mangle -A XRAY -d 192.88.99.0/24 -j RETURN
# iptables -t mangle -A XRAY -d  192.168.0.0/16 -j RETURN
iptables -t mangle -A XRAY -d 198.18.0.0/15 -j RETURN
iptables -t mangle -A XRAY -d 198.51.100.0/24 -j RETURN
iptables -t mangle -A XRAY -d 203.0.113.0/24 -j RETURN
iptables -t mangle -A XRAY -d 224.0.0.0/4 -j RETURN
iptables -t mangle -A XRAY -d 240.0.0.0/4 -j RETURN
iptables -t mangle -A XRAY -d 255.255.255.255/32 -j RETURN
iptables -t mangle -A XRAY -d 192.168.0.0/16 -p tcp ! --dport 3389 -j RETURN
iptables -t mangle -A XRAY -d 192.168.0.0/16 -p udp ! --dport 3389 -j RETURN

iptables -t mangle -A XRAY -m mark --mark 0xff -j RETURN 
iptables -t mangle -A XRAY -p udp -j TPROXY --on-port 12345 --tproxy-mark 1 # 给 UDP 打标记 1，转发至 12345 端口
iptables -t mangle -A XRAY -p tcp -j TPROXY --on-port 12345 --tproxy-mark 1 # 给 TCP 打标记 1，转发至 12345 端口
iptables -t mangle -A PREROUTING -j XRAY # 应用规则
```

xray已开debug，无任何流量进入：
```log
2021/09/26 13:06:19 [Debug] app/router: MphDomainMatcher is enabled for 71354 domain rule(s)
2021/09/26 13:06:20 [Debug] app/router: MphDomainMatcher is enabled for 60813 domain rule(s)
2021/09/26 13:06:20 [Debug] app/proxyman/inbound: creating stream worker on 0.0.0.0:2053
2021/09/26 13:06:20 [Debug] app/proxyman/inbound: creating stream worker on 0.0.0.0:12345
2021/09/26 13:06:20 [Info] transport/internet/tcp: listening TCP on 0.0.0.0:12345
2021/09/26 13:06:20 [Info] transport/internet/udp: listening UDP on 0.0.0.0:12345
2021/09/26 13:06:20 [Info] transport/internet/tcp: listening TCP on 0.0.0.0:2053
2021/09/26 13:06:20 [Info] transport/internet/udp: listening UDP on 0.0.0.0:2053
2021/09/26 13:06:20 [Warning] core: Xray 1.4.5 started
```

麻烦各位大佬指点一下，谢谢！

w311ang, 2022-12-10:
是不是请求也超时？

w311ang, 2022-12-11:
楼主解决了吗，我K2P貌似也遇到了同样的问题

w311ang, 2022-12-11:
经过我的研究初步认定应该是hnat导致的问题，回退到20210801代码后就正常工作了

w311ang, 2022-12-12:
已通过移除hnat模块解决

devioa, 2024-04-09:
> 已通过移除hnat模块解决

你好我最近折腾 openwrt 透明代理也遇到了这个问题，你说的 hnat 模块是指的什么，谷歌搜了下好像是硬件 NAT 相关的，我是 x86 软路由，应该没有这个东西吧？

coolsnowwolf, 2026-09-05:
该报告针对 2021 年构建。当前内核模块定义已将 `iptables-mod-tproxy` 正确依赖到 `kmod-ipt-tproxy`/`kmod-nf-tproxy`，并由 `include/netfilter.mk` 生成 `xt_TPROXY`、`nf_tproxy_ipv4/6` 模块；选择包后会同步启用对应 Kconfig。旧版“勾选但模块丢失”问题已被后续模块依赖整理覆盖。请确认固件实际安装了 iptables-mod-tproxy 与 kmod-ipt-tproxy；该旧 issue 现关闭。

closed 2026-09-05 by coolsnowwolf