coolsnowwolf/lede #7942: How to build with auto/en interface?

uamarchuan, 2021-09-26:
Hello, 

Thank you for the great job and a nice OpenWrt build.
I have a question: What do I need to change/comment/add to have the default English version in the Luci interface?
Thank you in advance.

Regards.

Her0R0cket, 2021-09-27:
In this file 'package/lean/default-settings/files/zzz-default-settings',
You can delete  line 3~8 .

Boos4721, 2021-09-27:
https://github.com/coolsnowwolf/lede/blob/be087a4f7e6d088a8bfa31247e0ae6dcd3712891/package/lean/default-settings/files/zzz-default-settings#L3

uamarchuan, 2021-09-27:
Super, thank you. Today evening i will try it and give you a feedback if it's work or not.

1715173329, 2021-09-29:
Well, for general purpose, you may simply unselect `default-settings` package, which is in `LuCI` menu.

uamarchuan, 2021-09-29:
> 
> 
> In this file 'package/lean/default-settings/files/zzz-default-settings', You can delete line 3~8 .

For me somehow works this:
>sed -i "s/zh_cn/en/g" package/lean/default-settings/files/zzz-default-settings
sed -i "s/+@LUCI_LANG_zh-cn/+@LUCI_LANG_en/g" package/lean/default-settings/Makefile
sed -i "/po2lmo .\/po\/zh-cn\/default.po/d" package/lean/default-settings/Makefile

After upgrade i receive english interface. But in the /etc/config/luci file still zh-cn
>config core 'main'
        option mediaurlbase '/luci-static/bootstrap'
        option resourcebase '/luci-static/resources'
        option lang 'zh-cn'

lphgor, 2021-09-30:
> After upgrade i receive english interface. But in the /etc/config/luci file still zh-cn

https://github.com/coolsnowwolf/luci/blob/83494a9a0bb26cadd339bb73389c486242788bad/modules/luci-base/root/etc/uci-defaults/luci-base#L12-L15