coolsnowwolf/lede #7944: 无法进入编译设置界面

pangfeng161, 2021-09-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：无法进入编译设置界面

### (2) 路由器型号和固件版本

A：X86

### (3) 详细日志

A：make menuconfig
tmp/.config-package.in:65188: syntax error
tmp/.config-package.in:65187: invalid statement
tmp/.config-package.in:65188: invalid statement
tmp/.config-package.in:65189: invalid statement
tmp/.config-package.in:65190: invalid statement
/home/pangzhao/lede/include/toplevel.mk:133: recipe for target 'menuconfig' failed
make: *** [menuconfig] Error 1

s0fx2, 2021-09-27:
我也遇到了，看来是lede那边的问题，等更新吧

ostrichii, 2021-09-27:
同样的问题

Boos4721, 2021-09-27:
https://github.com/fw876/helloworld/issues/670

Dimitri2020007, 2021-09-27:
我也遇到这个问题了

xayane, 2021-09-27:
我的云编译报错，看了一眼，是一样的错误

Dimitri2020007, 2021-09-27:
> 我的云编译报错，看了一眼，是一样的错误

我已经发issue了

Django9264, 2021-09-28:
我也是云编译出错了，试了下本地编译同样的错误

closed 2021-09-28 by coolsnowwolf