coolsnowwolf/lede #7945: Actions最新源码在线编译出错（卡在下载了）

Dimitri2020007, 2021-09-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：卡下载了

### (2) 路由器型号和固件版本

A：x86_64-efi

### (3) 详细日志

A：

2021-09-27T07:42:02.0371862Z ##[group]Run cd openwrt
2021-09-27T07:42:02.0372442Z [36;1mcd openwrt[0m
2021-09-27T07:42:02.0372851Z [36;1mmake defconfig[0m
2021-09-27T07:42:02.0373294Z [36;1mmake download -j8[0m
2021-09-27T07:42:02.0373896Z [36;1mfind dl -size -1024c -exec ls -l {} \;[0m
2021-09-27T07:42:02.0374521Z [36;1mfind dl -size -1024c -exec rm -f {} \;[0m
2021-09-27T07:42:26.6064286Z tmp/.config-package.in:65474: syntax error
2021-09-27T07:42:26.6065459Z tmp/.config-package.in:65473: invalid statement
2021-09-27T07:42:26.6066511Z tmp/.config-package.in:65474: invalid statement
2021-09-27T07:42:26.6067519Z tmp/.config-package.in:65475: invalid statement
2021-09-27T07:42:26.6068489Z tmp/.config-package.in:65476: invalid statement
2021-09-27T07:42:26.6423488Z make: *** [/workdir/openwrt/include/toplevel.mk:120: defconfig] Error 1
2021-09-27T07:42:26.6434276Z ##[error]Process completed with exit code 2.

zjkhy94, 2021-09-27:
https://github.com/fw876/helloworld/issues/670

admin6750217, 2021-09-28:
源码减少了调度使用。开启了节能模式。节能。卧槽。

closed 2021-09-28 by coolsnowwolf