coolsnowwolf/lede #7950: ZTE MF269 活动连接数只有8192

xfmly, 2021-09-28:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
ZTE MF269 活动连接数只有8192，查看了设置文件，数值已经是65535

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：ZTE MF269
       内核：5.10.64

(3) 详细日志

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.10.64 (runner@fv-az275-655) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 8.4.0 r3744-ecab8d5e1) 8.4.0, GNU ld (GNU Binutils) 2.34) #0 SMP Mon Sep 27 14:36:28 2021
[    0.000000] Machine model: ZTE MF269
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000040000000-0x000000005fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x0000000040ffffff]
[    0.000000]   node   0: [mem 0x0000000041000000-0x000000004a5fffff]
[    0.000000]   node   0: [mem 0x000000004a600000-0x000000004e7fffff]
[    0.000000]   node   0: [mem 0x000000004e800000-0x00000000504fffff]
[    0.000000]   node   0: [mem 0x0000000050500000-0x00000000506fffff]
[    0.000000]   node   0: [mem 0x0000000050700000-0x000000005fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000005fffffff]
[    0.000000] On node 0 totalpages: 131072
[    0.000000]   DMA zone: 2048 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 131072 pages, LIFO batch:31
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.0
[    0.000000] percpu: Embedded 20 pages/cpu s41112 r8192 d32616 u81920
[    0.000000] pcpu-alloc: s41112 r8192 d32616 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129024
[    0.000000] Kernel command line: console=ttyMSM0,115200n8 ubi.mtd=rootfs root=mtd:ubi_rootfs rootfstype=squashfs rootwait root=/dev/ubiblock0_1
[    0.000000] Dentry cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 416268K/524288K available (7294K kernel code, 838K rwdata, 1872K rodata, 1280K init, 301K bss, 108020K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: 	CONFIG_RCU_FANOUT set to non-default value of 32.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] random: get_random_bytes called from start_kernel+0x350/0x570 with crng_init=0
[    0.000000] arch_timer: cp15 and mmio timer(s) running at 19.20MHz (virt/virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000005] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000155] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=76800)
[    0.000171] pid_max: default: 32768 minimum: 301
[... 336 lines cut ...]
[   55.409635] br-lan: port 2(wlan0) entered disabled state
[   55.490101] device wlan1 left promiscuous mode
[   55.490223] br-lan: port 3(wlan1) entered disabled state
[   55.735145] wlan0: Destroyed NSS virtual interface
[   55.735178] wlan0: RX-Q purged.
[   55.787132] wlan1: Destroyed NSS virtual interface
[   55.787166] wlan1: RX-Q purged.
[   56.039723] debugfs: File 'virt_if' in directory 'stats' already present!
[   56.039895] wlan0: Created a NSS virtual interface.
[   56.045679] wlan0: RX-Q and tasklet initialized.
[   56.054257] debugfs: File 'virt_if' in directory 'stats' already present!
[   56.055297] wlan1: Created a NSS virtual interface.
[   56.061854] wlan1: RX-Q and tasklet initialized.
[   56.089403] br-lan: port 2(wlan0) entered blocking state
[   56.089436] br-lan: port 2(wlan0) entered disabled state
[   56.094305] device wlan0 entered promiscuous mode
[   56.099306] br-lan: port 2(wlan0) entered blocking state
[   56.103740] br-lan: port 2(wlan0) entered forwarding state
[   56.128166] br-lan: port 3(wlan1) entered blocking state
[   56.128204] br-lan: port 3(wlan1) entered disabled state
[   56.132911] device wlan1 entered promiscuous mode
[   56.138101] br-lan: port 3(wlan1) entered blocking state
[   56.142503] br-lan: port 3(wlan1) entered forwarding state
[   56.156278] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   56.445190] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready

Boos4721, 2021-09-28:
https://github.com/coolsnowwolf/lede/blob/0088bb160151d065d184224c9659711e9e133afa/package/qca/nss/qca-nss-ecm-64/Makefile#L50

xfmly, 2021-09-28:
已经解决，感谢大佬解答。

closed 2021-09-28 by xfmly