coolsnowwolf/lede #7952: luci-app-netspeedtest报错

rongcenter, 2021-09-28:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：luci-app-netspeedtest报错

### (2) 路由器型号和固件版本

A：x86_64

### (3) 详细日志

Create index file './feeds/packages.index'
Create index file './feeds/luci.index'
Create index file './feeds/routing.index'
Create index file './feeds/telephony.index'
Create index file './feeds/helloworld.index'
ERROR: please fix package/lean/luci-app-netspeedtest/Makefile - see logs/package/lean/luci-app-netspeedtest/dump.txt for details

qjiang888, 2021-09-28:
make[3]: Entering directory '/home/qjiang/lede/package/lean/luci-app-frpc'
mkdir -p /home/qjiang/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/qjiang/lede/tmp/.root-copy.flock -c 'cp -fpR /home/qjiang/lede/build_dir/target-x86_64_musl/luci-app-frpc/.pkgdir/luci-app-frpc/. /home/qjiang/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/qjiang/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.luci-app-frpc_installed
echo "luci-app-frpc" >> /home/qjiang/lede/staging_dir/target-x86_64_musl/pkginfo/luci-app-frpc.default.install
mkdir -p /home/qjiang/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/qjiang/lede/tmp/.root-copy.flock -c 'cp -fpR /home/qjiang/lede/build_dir/target-x86_64_musl/luci-app-frpc/.pkgdir/luci-i18n-frpc-zh-cn/. /home/qjiang/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/qjiang/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.luci-i18n-frpc-zh-cn_installed
echo "luci-i18n-frpc-zh-cn" >> /home/qjiang/lede/staging_dir/target-x86_64_musl/pkginfo/luci-app-frpc.default.install
make[3]: Leaving directory '/home/qjiang/lede/package/lean/luci-app-frpc'
time: package/lean/luci-app-frpc/compile#0.13#0.04#0.17
make[3]: Entering directory '/home/qjiang/lede/package/lean/luci-app-jd-dailybonus'
mkdir -p /home/qjiang/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/qjiang/lede/tmp/.root-copy.flock -c 'cp -fpR /home/qjiang/lede/build_dir/target-x86_64_musl/luci-app-jd-dailybonus/.pkgdir/luci-app-jd-dailybonus/. /home/qjiang/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/qjiang/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.luci-app-jd-dailybonus_installed
echo "luci-app-jd-dailybonus" >> /home/qjiang/lede/staging_dir/target-x86_64_musl/pkginfo/luci-app-jd-dailybonus.default.install
make[3]: Leaving directory '/home/qjiang/lede/package/lean/luci-app-jd-dailybonus'
time: package/lean/luci-app-jd-dailybonus/compile#0.14#0.15#0.29
make[3]: Entering directory '/home/qjiang/lede/package/lean/luci-app-netspeedtest'
Makefile:47: *** DESCRIPTION:= is obsolete, use Package/PKG_NAME/description.  Stop.
make[3]: Leaving directory '/home/qjiang/lede/package/lean/luci-app-netspeedtest'
time: package/lean/luci-app-netspeedtest/compile#0.11#0.02#0.13
    ERROR: package/lean/luci-app-netspeedtest failed to build.
make[2]: *** [package/Makefile:116: package/lean/luci-app-netspeedtest/compile] Error 1
make[2]: Leaving directory '/home/qjiang/lede'
make[1]: *** [package/Makefile:110: /home/qjiang/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/qjiang/lede'
make: *** [/home/qjiang/lede/include/toplevel.mk:230：world] 错误 2

MilesPoupart, 2021-09-28:
我github action也报错 r2s
 make[2] -C package/lean/luci-app-netspeedtest download
 make[2] -C package/lean/luci-app-nfs download
    ERROR: package/lean/luci-app-netspeedtest failed to build.

qjiang888, 2021-09-28:
大佬神速，感谢大佬！！

WYC-2020, 2021-09-28:
瞎搞哈哈，有的完全没必要都加进来，需要的自己加入就可以了

closed 2021-09-28 by coolsnowwolf