coolsnowwolf/lede #7955: r4s用新uboot打包报错

Jason6111, 2021-09-28:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：r4s

### (3) 详细日志

A：+ set 33554432 16777216 67108864 167772160
3203
+ KERNELOFFSET=65536
3204
+ KERNELSIZE=16777216
3205
+ ROOTFSOFFSET=131072
3206
+ ROOTFSSIZE=327680
3207
+ [ -n  ]
3208
+ dd if=/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/root.squashfs of=/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/tmp/openwrt-rockchip-armv8-friendlyarm_nanopi-r4s-squashfs-sysupgrade.img.gz bs=512 seek=131072 conv=notrunc
3209
62250+1 records in
3210
62250+1 records out
3211
31872185 bytes (32 MB, 30 MiB) copied, 0.165413 s, 193 MB/s
3212
+ [ -n  ]
3213
+ make_ext4fs -J -L kernel -l 16777216 /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/tmp/openwrt-rockchip-armv8-friendlyarm_nanopi-r4s-squashfs-sysupgrade.img.gz.kernel /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/tmp/openwrt-rockchip-armv8-friendlyarm_nanopi-r4s-squashfs-sysupgrade.img.gz.boot
3214
+ dd if=/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/tmp/openwrt-rockchip-armv8-friendlyarm_nanopi-r4s-squashfs-sysupgrade.img.gz.kernel of=/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/tmp/openwrt-rockchip-armv8-friendlyarm_nanopi-r4s-squashfs-sysupgrade.img.gz bs=512 seek=65536 conv=notrunc
3215
Creating filesystem with parameters:
3216
    Size: 16777216
3217
    Block size: 4096
3218
    Blocks per group: 32768
3219
    Inodes per group: 1024
3220
    Inode size: 256
3221
    Journal blocks: 0
3222
    Label: kernel
3223
    Blocks: 4096
3224
    Block groups: 1
3225
    Reserved blocks: 0
3226
    Reserved block group size: 7
3227
Created filesystem with 14/1024 inodes and 3706/4096 blocks
3228
32768+0 records in
3229
32768+0 records out
3230
16777216 bytes (17 MB, 16 MiB) copied, 0.0700319 s, 240 MB/s
3231
+ rm -f /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/tmp/openwrt-rockchip-armv8-friendlyarm_nanopi-r4s-squashfs-sysupgrade.img.gz.kernel
3232
# Copy the idbloader and the u-boot image to the image at sector 0x40 and 0x4000
3233
dd if="/workdir/openwrt/staging_dir/target-aarch64_generic_musl/image"/nanopi-r4s-rk3399-idbloader.img of="/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/tmp/openwrt-rockchip-armv8-friendlyarm_nanopi-r4s-squashfs-sysupgrade.img.gz" seek=64 conv=notrunc
3234
dd: failed to open '/workdir/openwrt/staging_dir/target-aarch64_generic_musl/image/nanopi-r4s-rk3399-idbloader.img': No such file or directory
3235
make[5]: *** [Makefile:80: /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/tmp/openwrt-rockchip-armv8-friendlyarm_nanopi-r4s-squashfs-sysupgrade.img.gz] Error 1
3236
make[4]: *** [Makefile:24: install] Error 2
3237
make[3]: *** [Makefile:11: install] Error 2
3238
make[5]: Leaving directory '/workdir/openwrt/target/linux/rockchip/image'
3239
make[4]: Leaving directory '/workdir/openwrt/target/linux/rockchip'
3240
make[3]: Leaving directory '/workdir/openwrt/target/linux'
3241
time: target/linux/install#31.30#2.36#24.49
3242
    ERROR: target/linux failed to build.
3243
make[2]: *** [target/Makefile:25: target/linux/install] Error 1
3244
make[2]: Leaving directory '/workdir/openwrt'
3245
make[1]: *** [target/Makefile:19: /workdir/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.target_install] Error 2
3246
make[1]: Leaving directory '/workdir/openwrt'
3247
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
3248
Error: Process completed with exit code 2.

1715173329, 2021-09-29:
#7957

Jason6111, 2021-09-29:
> #7957

多谢

PR #7957 (2021-09-29): rockchip: update binary name of idbloader
In rockchip's proprietary ddrloader, the idbloader can be general and
no longer limited to the specific device.
This matches the behavior in arm-trusted-firmware-rockchip-vendor.

Fixes: dd71a38acb78ef ("uboot-rockchip: update package")

Fixes: #7955
closed 2021-09-29 by Jason6111