coolsnowwolf/lede #7958: 添加netspeedtest后打开出错

qqsir-dev, 2021-09-29:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
点开后页面就是下面这样
### (1) 具体问题
/usr/lib/lua/luci/dispatcher.lua:469: Failed to execute cbi dispatcher target for entry '/admin/network/netspeedtest'.
The called action terminated with an exception:
/usr/lib/lua/luci/cbi.lua:53: /usr/lib/lua/luci/model/cbi/netspeedtest.lua:15: ')' expected near 'translate'
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:469: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>
A：

### (2) 路由器型号和固件版本
软路由
A：

### (3) 详细日志

A：

Jason6111, 2021-09-29:
一样 求解

closed 2021-09-29 by AmadeusGhost

WYC-2020, 2021-09-29:
我重新修改了makefile 没遇到这个问题
[image]