coolsnowwolf/lede #7962: 编译时开启ssr-plus，但实际没安装上

ZPerling, 2021-09-29:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译时开启了ssr-plus应用，但是刷完固件后发现管理页面没有这个应用的配置

### (2) 路由器型号和固件版本

A：树莓派3B+, 固件版本：lede-20200408
```
CONFIG_HAS_SUBTARGETS=y
CONFIG_HAS_DEVICES=y
CONFIG_TARGET_BOARD="bcm27xx"
CONFIG_TARGET_SUBTARGET="bcm2710"
CONFIG_TARGET_PROFILE="DEVICE_rpi-3"
CONFIG_TARGET_ARCH_PACKAGES="aarch64_cortex-a53"
CONFIG_DEFAULT_TARGET_OPTIMIZATION="-Os -pipe -mcpu=cortex-a53"
CONFIG_CPU_TYPE="cortex-a53"
```

### (3) 详细日志
[image]
[image]

A：
[image]
我是必需要开启 CONFIG_PACKAGE_luci-app-ssrserver-python 这个选项吗？

deplives, 2021-09-29:
你在说什么？ssrp 是哪一个你都不知道。。。

ZPerling, 2021-09-29:
> 你在说什么？ssrp 是哪一个你都不知道。。。

小白用户，还请指教，具体应该怎么装

suizhenmin, 2021-09-29:
添加源 https://github.com/fw876/helloworld.git

ZPerling, 2021-09-29:
> 添加源 https://github.com/fw876/helloworld.git

具体怎么加呢 🙏

TerryWang159, 2021-10-11:
> 
> 
> ```shell
> cd lede
> sed -i '$a src-git NueXini_Packages https://github.com/NueXini/NueXini_Packages.git' feeds.conf.default
> ./scripts/feeds update -a && ./scripts/feeds install -a
> ```

可以编译在原生的Openwrt项目下吗 ？