coolsnowwolf/lede #7965: nlbwmon插件一直提示 Out of memory

tone88, 2021-09-30:
流量监控插件一直提示 Out of memory，见下方日志，X86的机子 内存不应该出现溢出问题吧。
之前也有这种提示，后来修改提交间隔后不提示了，现在间隔已经修改为1小时了还是会一直提示。。。

### (2) 路由器型号和固件版本
X86，最新源码编译

### (3) 详细日志
Wed Sep 29 19:14:56 2021 daemon.err nlbwmon[4591]: Netlink receive failure: Out of memory
Wed Sep 29 19:14:56 2021 daemon.err nlbwmon[4591]: Unable to dump conntrack: No buffer space available
Wed Sep 29 22:03:00 2021 daemon.err nlbwmon[4591]: Netlink receive failure: Out of memory
Wed Sep 29 22:03:00 2021 daemon.err nlbwmon[4591]: Unable to dump conntrack: No buffer space available
Thu Sep 30 00:51:03 2021 daemon.err nlbwmon[4591]: Netlink receive failure: Out of memory
Thu Sep 30 00:51:03 2021 daemon.err nlbwmon[4591]: Unable to dump conntrack: No buffer space available
Thu Sep 30 03:39:07 2021 daemon.err nlbwmon[4591]: Netlink receive failure: Out of memory
Thu Sep 30 03:39:07 2021 daemon.err nlbwmon[4591]: Unable to dump conntrack: No buffer space available
Thu Sep 30 06:27:10 2021 daemon.err nlbwmon[4591]: Netlink receive failure: Out of memory
Thu Sep 30 06:27:10 2021 daemon.err nlbwmon[4591]: Unable to dump conntrack: No buffer space available
Thu Sep 30 09:47:23 2021 daemon.err nlbwmon[30760]: Netlink receive failure: Out of memory
Thu Sep 30 09:47:23 2021 daemon.err nlbwmon[30760]: Unable to dump conntrack: No buffer space available

davidwu111, 2021-09-30:
https://github.com/coolsnowwolf/lede/issues/7219 
我把nlbwmon直接删掉了

kleisth, 2021-10-01:
一直有这个警告，很久了。。
不用nlbwmon又不知道还有啥能查看实时流量的

realconnor, 2021-10-01:
以前改netlink缓存或提交间隔确实能解决问题，因为默认的参数设置主要是适用于低性能、低带宽设备的，但是最近升级到`R21.9.18`时又出现了这个问题，网上找了下，这是nlbwmon开发者近期的答复：

> [A warning has been implemented in d82c910 pointing out the need to increase net.core.rmem_max as well.](https://github.com/jow-/nlbwmon/issues/43#:~:text=A%20warning%20has%20been%20implemented%20in%20d82c910%20pointing%20out%20the%20need%20to%20increase%20net.core.rmem_max%20as%20well.)

所以我在适当增大缓存（比如加个0改成10M），并保持当前默认的提交间隔（4h）的基础上，把套接字接收/发送缓存从系统默认的“212992”（208K）增大到了“16777216”（16M）。当然，这个参数也不知道怎么根据带宽来计算比较合适，是根据网络上Linux内核参数优化的推荐来设置的，也许带宽特别大的情况下还需要适当增大，当然低内存设备可以自己试试小一些的缓存，或者干脆把这软件删掉，没啥必要。我目前用下来，再没出过问题了。
```
root@OpenWrt:~# cat /etc/sysctl.conf
# Defaults are configured in /etc/sysctl.d/* and can be customized in this file
net.core.wmem_max=16777216
net.core.rmem_max=16777216
```
用`sysctl -p`生效，之后再reload一下nlbwmon。自从上次重启，没出过相关的问题，
```
root@OpenWrt:~# uptime
 10:13:10 up 8 days, 18:05,  load average: 0.00, 0.00, 0.00
```
上月的数据：
```
root@OpenWrt:~# nlbw -c show -g mac -o rx -t 2021-09-01
              MAC      Conn.   < Downld. (   Pkts. )      Upload (   Pkts. )
c4:09:**:f4:1b:8f    20.24 K        52 B (       1 )   257.95 MB ( 559.94 K)
e4:f4:**:d4:2a:d5        76     13.47 KB (      98 )    29.67 KB (     165 )
00:00:**:00:00:00     3.28 K   622.95 KB (   4.93 K)    77.60 KB (     683 )
00:bb:**:b0:f8:66     1.63 K     7.86 MB (   6.16 K)     1.30 MB (   6.31 K)
48:5a:**:14:19:dd    24.73 K     1.96 GB (   1.00 M)   526.67 MB ( 853.66 K)
a0:57:**:d4:59:ae   413.98 K    14.35 GB (   7.13 M)   810.90 MB (   5.00 M)
28:d2:**:57:7b:14   644.80 K    49.78 GB (  27.14 M)     6.37 GB (  17.29 M)
```

tone88, 2021-10-03:
> 以前改netlink缓存或提交间隔确实能解决问题，因为默认的参数设置主要是适用于低性能、低带宽设备的，但是最近升级到`R21.9.18`时又出现了这个问题，网上找了下，这是nlbwmon开发者近期的答复： [...]
按兄弟的方法测试了几天再没出现内存溢出的提示，👍

closed 2021-10-03 by tone88