coolsnowwolf/lede #7966: ubuntu 2104 卡ffmpeg

by-JohnChen, 2021-09-30:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述
编译固件卡ffmpeg 是库不对吗
### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：newifi-d2

### (3) 详细日志

A：ede/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /home/masmk/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.miniupnpd_installed
echo "miniupnpd" >> /home/masmk/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/miniupnpd.default.install
make[3]: Leaving directory '/home/masmk/lede/feeds/packages/net/miniupnpd'
time: package/feeds/packages/miniupnpd/compile#0.20#0.08#0.26
make[3]: Entering directory '/home/masmk/lede/feeds/luci/applications/luci-app-upnp'
mkdir -p /home/masmk/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
SHELL= flock /home/masmk/lede/tmp/.root-copy.flock -c 'cp -fpR /home/masmk/lede/build_dir/target-mipsel_24kc_musl/luci-app-upnp/.pkgdir/luci-app-upnp/. /home/masmk/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /home/masmk/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.luci-app-upnp_installed
echo "luci-app-upnp" >> /home/masmk/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/luci-app-upnp.default.install
mkdir -p /home/masmk/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
SHELL= flock /home/masmk/lede/tmp/.root-copy.flock -c 'cp -fpR /home/masmk/lede/build_dir/target-mipsel_24kc_musl/luci-app-upnp/.pkgdir/luci-i18n-upnp-zh-cn/. /home/masmk/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /home/masmk/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.luci-i18n-upnp-zh-cn_installed
echo "luci-i18n-upnp-zh-cn" >> /home/masmk/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/luci-app-upnp.default.install
make[3]: Leaving directory '/home/masmk/lede/feeds/luci/applications/luci-app-upnp'
time: package/feeds/luci/luci-app-upnp/compile#1.34#0.77#1.95
make[3]: Entering directory '/home/masmk/lede/package/network/utils/wireguard-tools'
mkdir -p /home/masmk/lede/dl
SHELL= flock /home/masmk/lede/tmp/.wireguard-tools-1.0.20210914.tar.xz.flock -c '       /home/masmk/lede/scripts/download.pl "/home/masmk/lede/dl" "wireguard-tools-1.0.20210914.tar.xz" "97ff31489217bb265b7ae850d3d0f335ab07d2652ba1feec88b734bc96bd05ac" "" "https://git.zx2c4.com/wireguard-tools/snapshot/"    '
mkdir -p /home/masmk/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
SHELL= flock /home/masmk/lede/tmp/.root-copy.flock -c 'cp -fpR /home/masmk/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/wireguard-tools-1.0.20210914/.pkgdir/wireguard-tools/. /home/masmk/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/'touch /home/masmk/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.wireguard-tools_installed
echo "wireguard-tools" >> /home/masmk/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/wireguard-tools.default.install
make[3]: Leaving directory '/home/masmk/lede/package/network/utils/wireguard-tools'
time: package/network/utils/wireguard-tools/compile#0.22#0.07#0.27
make[3]: Entering directory '/home/masmk/lede/feeds/luci/protocols/luci-proto-wireguard'
mkdir -p /home/masmk/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
SHELL= flock /home/masmk/lede/tmp/.root-copy.flock -c 'cp -fpR /home/masmk/lede/build_dir/target-mipsel_24kc_musl/luci-proto-wireguard/.pkgdir/luci-proto-wireguard/. /home/masmk/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /home/masmk/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.luci-proto-wireguard_installed
echo "luci-proto-wireguard" >> /home/masmk/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/luci-proto-wireguard.default.install
make[3]: Leaving directory '/home/masmk/lede/feeds/luci/protocols/luci-proto-wireguard'
time: package/feeds/luci/luci-proto-wireguard/compile#0.28#0.10#0.35
make[3]: Entering directory '/home/masmk/lede/feeds/luci/applications/luci-app-wireguard'
mkdir -p /home/masmk/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
SHELL= flock /home/masmk/lede/tmp/.root-copy.flock -c 'cp -fpR /home/masmk/lede/build_dir/target-mipsel_24kc_musl/luci-app-wireguard/.pkgdir/luci-app-wireguard/. /home/masmk/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /home/masmk/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.luci-app-wireguard_installed
echo "luci-app-wireguard" >> /home/masmk/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/luci-app-wireguard.default.install
mkdir -p /home/masmk/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
SHELL= flock /home/masmk/lede/tmp/.root-copy.flock -c 'cp -fpR /home/masmk/lede/build_dir/target-mipsel_24kc_musl/luci-app-wireguard/.pkgdir/luci-i18n-wireguard-zh-cn/. /home/masmk/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /home/masmk/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.luci-i18n-wireguard-zh-cn_installed
echo "luci-i18n-wireguard-zh-cn" >> /home/masmk/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/luci-app-wireguard.default.install
make[3]: Leaving directory '/home/masmk/lede/feeds/luci/applications/luci-app-wireguard'
time: package/feeds/luci/luci-app-wireguard/compile#0.47#0.27#0.72
make[3]: Entering directory '/home/masmk/lede/feeds/packages/net/etherwake'
mkdir -p /home/masmk/lede/dl
SHELL= flock /home/masmk/lede/tmp/.etherwake_1.09.orig.tar.gz.flock -c '        /home/masmk/lede/scripts/download.pl "/home/masmk/lede/dl" "etherwake_1.09.orig.tar.gz" "54241c7689579dc86e29e6afbc6d60e69f97135091a1395c8a10f6d5a2daec1d" "" "http://ftp.debian.org/debian/pool/main/e/etherwake"    '
mkdir -p /home/masmk/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
SHELL= flock /home/masmk/lede/tmp/.root-copy.flock -c 'cp -fpR /home/masmk/lede/build_dir/target-mipsel_24kc_musl/etherwake-1.09.orig/.pkgdir/etherwake/. /home/masmk/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /home/masmk/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.etherwake_installed
echo "etherwake" >> /home/masmk/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/etherwake.default.install
make[3]: Leaving directory '/home/masmk/lede/feeds/packages/net/etherwake'
time: package/feeds/packages/etherwake/compile#0.21#0.06#0.25
[... 480 lines cut ...]
make[2]: Leaving directory '/home/masmk/lede'
make[1]: *** [package/Makefile:110: /home/masmk/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/masmk/lede'
make: *** [/home/masmk/lede/include/toplevel.mk:230: world] Error 2
╭─masmk@k8s-masmk-storage-master in ~/lede on master ✘ (origin/master)
╰$ 
mipsel-openwrt-linux-musl-gcc -Llibavcodec -Llibavdevice -Llibavfilter -Llibavformat -Llibavresample -Llibavutil -Llibpostproc -Llibswscale -Llibswresample -L/home/masmk/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/lib -L/home/masmk/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib -znow -zrelro  -EL -march=24kc -Wl,--as-needed -Wl,-z,noexecstack -Wl,--warn-common -Wl,-rpath-link=:libpostproc:libswresample:libswscale:libavfilter:libavdevice:libavformat:libavcodec:libavutil:libavresample   -o ffmpeg_g fftools/ffmpeg_opt.o fftools/ffmpeg_filter.o fftools/ffmpeg_hw.o fftools/cmdutils.o fftools/ffmpeg.o  -lavdevice -lavfilter -lavformat -lavcodec -lswresa

4kc_musl/usr/lib -lz -L/home/masmk/lede/staging_dir/target-mipsel_24kc_musl/usr/lib -lgnutls -pthread -lm -L/home/masmk/lede/staging_dir/target-mipsel_24kc_musl/usr/lib -lz -L/home/masmk/lede/staging_dir/target-mipsel_24kc_musl/usr/lib -lfdk-aac -L/home/masmk/lede/staging_dir/target-mipsel_24kc_musl/usr/lib -lopus -L/home/masmk/lede/staging_dir/target-mipsel_24kc_musl/usr/lib -lm -lshine -lm -lm -pthread -lm
/home/masmk/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib/gcc/mipsel-openwrt-linux-musl/8.4.0/../../../../mipsel-openwrt-linux-musl/bin/ld: libavformat/libavformat.so: undefined reference to `__atomic_fetch_sub_8'
/home/masmk/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib/gcc/mipsel-openwrt-linux-musl/8.4.0/../../../../mipsel-openwrt-linux-musl/bin/ld: libavformat/libavformat.so: undefined reference to `__atomic_store_8'
/home/masmk/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib/gcc/mipsel-openwrt-linux-musl/8.4.0/../../../../mipsel-openwrt-linux-musl/bin/ld: libavformat/libavformat.so: undefined reference to `__atomic_load_8'
[... cut ...]
time: package/feeds/packages/ffmpeg/full/compile#0.72#0.13#0.84
    ERROR: package/feeds/packages/ffmpeg failed to build (build variant: full).
make[2]: *** [package/Makefile:116: package/feeds/packages/ffmpeg/compile] Error 1
make[2]: Leaving directory '/home/masmk/lede'
make[1]: *** [package/Makefile:110: /home/masmk/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/masmk/lede'
make: *** [/home/masmk/lede/include/toplevel.mk:230: world] Error 2

kleisth, 2021-10-01:
单线程编译一遍试试

by-JohnChen, 2021-10-01:
> 单线程编译一遍试试

make -j1 V=sc出来就是这

yonggedebaqi, 2021-10-01:
[image]
 --enable-libmp3lame --enable-gpl --enable-libx264 --enable-libx265 --enable-libfdk-aac
我的编译没有问题。

hzjnet, 2021-10-05:
> [image] --enable-libmp3lame --enable-gpl --enable-libx264 --enable-libx265 --enable-libfdk-aac 我的编译没有问题。

看情况的，我尝试同一套源码在mtk7621上就有问题，但是ipq的没问题

closed 2022-02-12 by Beginner-Go