coolsnowwolf/lede #7970: Turbo ACC 网络加速设置，每次更新系统都强制开启Shortcut-FE 流量分载，哪怕已经设置了关闭

db-one, 2021-10-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：Turbo ACC 网络加速设置，每次更新系统都强制开启Shortcut-FE 流量分载，哪怕已经设置了关闭
之前的版本正常的，最近半个月编译的好像都有这个问题，已经设置关闭了Shortcut-FE 流量分载，但是更新之后就会变成运行状态，除非再次点击保存应用，才会关闭

### (2) 路由器型号和固件版本

A：X86

### (3) 详细日志

A：
[image]

OldCoding, 2021-10-01:
同样有这个问题，重启或升级后都要重新点一下保存才能让它停止

dsyo2008, 2021-10-01:
同样问题，不知道哪个设置关闭，升级后是这样。

mysteryman2045, 2021-10-01:
不知道是不是 luci-app-turboacc/root/etc/config/turboacc中 option sfe_flow '1'   把1改成0再编译 应该就是默认关闭状态了
如果用不到这东西 完全也可以make menuconfig的时候 取消Include Shortcut-FE

db-one, 2021-10-01:
> 不知道是不是 luci-app-turboacc/root/etc/config/turboacc中 option sfe_flow '1' 把1改成0再编译 应该就是默认关闭状态了 如果用不到这东西 完全也可以make menuconfig的时候 取消Include Shortcut-FE

应该不是这个原因，保留配置文件升级，应该是会直接继承之前的配置文件的

hcym, 2021-10-01:
arm的容易挂，x86有抗药性

imy7, 2021-10-03:
@mysteryman2045  那个路径配置是指编译之后首次的软件运行状态, 1为开启  0为关闭   ,恢复出厂设置也是同样的!

helllkz, 2021-10-03:
X86用不上，所有我都只弄个NAT1，其他offload就不编译了

Ailick, 2021-10-08:
请测试此修复（任选）：
[turboacc.txt](https://github.com/coolsnowwolf/lede/files/7309608/turboacc.txt)请重命名为turboacc
**编译前**
替换源码内：
 `package/lean/luci-app-turboacc/root/etc/init.d/turboacc`
**保留配置文件升级**
替换路由器内：
`etc/init.d/turboacc`

Ailick, 2021-10-09:
> @Ailick 确实这样

nat的可以简单处理，也不会有什么后遗症。
dns这个，我怎么说呢，它功能本来就是这样的，dns缓存加速本来就依赖这个东西，而使用者非要不使用这个东西。我觉得它既是问题，但也不是问题。

smdx, 2021-10-09:
请问Turbo ACC的dns缓存加速功能在不开启的情况下还是每次重启系统就重置dhcp里面的相关设置是什么原因？

WYC-2020, 2021-10-09:
@Ailick 在使用trubo的dns的时候会忽略dhcp的dns，使用trubo指定的dns，是为了实现这个

smdx, 2021-10-10:
> @Ailick 它停止的时候就会恢复，为了保证后面没有开启dns的时候恢复到默认值，如果去掉，他开起dns然后再去掉dns那个忽略dns配置文件就没法复位了

只开启了ACC中的全锥形 NAT功能，每次重启路由器都会重置dhcp中的dns配置（例如：DNS 查询缓存的大小设置为0重启后又重置成150）.

Ailick, 2021-10-11:
> > @Ailick 它停止的时候就会恢复，为了保证后面没有开启dns的时候恢复到默认值，如果去掉，他开起dns然后再去掉dns那个忽略dns配置文件就没法复位了
> 
> 只开启了ACC中的全锥形 NAT功能，每次重启路由器都会重置dhcp中的dns配置（例如：DNS 查询缓存的大小设置为0重启后又重置成150）.
> 
> Sun Oct 10 16:09:20 2021 daemon.notice procd: /etc/rc.d/S90turboacc: net.ipv4.tcp_congestion_control = cubic Sun Oct 10 16:09:20 2021 daemon.notice procd: /etc/rc.d/S90turboacc: killall: dnscache: no process killed Sun Oct 10 16:09:20 2021 daemon.notice procd: /etc/rc.d/S90turboacc: sh: you need to specify whom to kill Sun Oct 10 16:09:20 2021 daemon.notice procd: /etc/rc.d/S90turboacc: Stop DNS Caching Sun Oct 10 16:09:21 2021 daemon.notice procd: /etc/rc.d/S90turboacc: DNSMASQ change

这个让大雕来吧，感觉这里面暗藏玄机。估计我一改，能炸一片。

WYC-2020, 2021-10-12:
@lgh200  我就是最新版了，昨晚才编译的

WYC-2020, 2021-10-12:
@lgh200  你是在防火墙那边做了什么操作吗？增加端口转发 保存后就会？我试下看下

lgh200, 2021-10-12:
> 我x86 我只是在测试防火墙和turbo 全锥状态是否同步 ，就在他们两之间一端开启一端关闭全锥

X86直接无视，Turbo ACC都可以不用编译进去。我测试的是ARM设备

lgh200, 2021-10-12:
> @lgh200 你是在防火墙那边做了什么操作吗？增加端口转发 保存后就会？我试下看下

不用添加，就保存应用就会。你试arm设备，X86没试过

lgh200, 2021-10-12:
> @lgh200 我看这个新增的turbo逻辑，防火墙那块就是获取配置信息，应该不至于吧，这个还是等大佬看吧，目前我手里没有arm设备没法测试，只能看有人还遇到没

我几台arm设备都会，看看有没有人遇到这样的问题，我ax6 mf269都会

WYC-2020, 2021-10-13:
全新的逻辑那个状态更改依赖于防火墙的reload触发，我测试防火墙在修改配置后不知道是没发送reload还是怎么，我来回测试四次就会出现不触发，尴尬,此时用reload 手动重新加载firewall也不行，必须手工修改一下firewall的配置文件，在进行reload就可以了，怀疑是不是这几次后他们的hash值一样了，procd查阅说凭借的就是什么md5还是hash进行比较的

Ailick, 2021-10-14:
> > > > > 请教一下重启路由后忽略解析文件被重置另外个issues解决了，DHCP/DNS高级设置-DNS 查询缓存的大小被重置需要修改turboacc那条代码啊？谢谢
> > > > 
> > > > 
> > > > 去掉`revert_dns`
> > > 
> > > 
> > > 谢谢，请问是去掉这一项吗？ [image]
> > > 另外这些需要动吗？ [image]
> > 
> > 
> > 去掉start()和stop()里的那一行，其他的不用管
> 
> 你好，刚刚去掉这两个地方的revert_dns后重启路由DNS 查询缓存的大小还是重置了，设置为0重启后又变成150了。 [image]

把最开头的 `restart_utils="true"` 删了，如果还不行那就与acc无关了。

Ailick, 2021-10-14:
> @Ailick 找到原因了，那个turbo修改firwall的方式uci commit firewall并不会更新config的md5所以导致这个状态不触发，在commit之后加入更新md5的就可以了reload_config firewall，我这边还把start里面那个turboacc_fss调用去掉了，因为我这边测试其实没必要调用这个，因为turbo本身的状态是没有问题的，只会出现fire修改了turbo状态不对，因为现在用了事件触发，已经修改了turbo的配置文件，所以当打开turbo的时候状态已经更新，不用在手工调用了

其实这个服务的机制不是这样的。总之正确使用它，不会出错。
修复这个问题主要是因为，防火墙修改后，如果状态不同步，那么重启以后会被acc重新启用或者禁用。

closed 2022-10-31 by aiamadeus