coolsnowwolf/lede #7971: lede在线编译passwall报错x509: certificate has expired or is not yet valid

Dimitri2020007, 2021-10-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：x86_64

### (3) 详细日志

A：

2021-10-01T06:44:18.4928379Z Finding targets
2021-10-01T06:44:18.5168890Z go: downloading github.com/golang/protobuf v1.5.2
2021-10-01T06:44:18.5232139Z go: downloading google.golang.org/protobuf v1.27.1
2021-10-01T06:44:18.6642124Z go: downloading google.golang.org/grpc v1.40.0
2021-10-01T06:44:18.8054451Z go: downloading golang.org/x/net v0.0.0-20210917221730-978cfadd31cf
2021-10-01T06:44:18.8080340Z go: downloading go.starlark.net v0.0.0-20210901212718-87f333178d59
2021-10-01T06:44:18.9485952Z go: downloading golang.org/x/sys v0.0.0-20210917161153-d61c044b1678
2021-10-01T06:44:18.9510911Z go: downloading golang.org/x/crypto v0.0.0-20210920023735-84f357641f63
2021-10-01T06:44:19.1023063Z go: downloading github.com/xtls/go v0.0.0-20210920065950-d4af136d3672
2021-10-01T06:44:19.1029439Z go: downloading github.com/lucas-clemente/quic-go v0.23.0
2021-10-01T06:44:19.2345280Z go: downloading github.com/refraction-networking/utls v0.0.0-20210713165636-0b2885c8c0d4
2021-10-01T06:44:19.2367825Z go: downloading github.com/pires/go-proxyproto v0.6.1
2021-10-01T06:44:19.3693683Z go: downloading github.com/ghodss/yaml v1.0.1-0.20190212211648-25d852aebe32
2021-10-01T06:44:19.3695152Z go: downloading github.com/pelletier/go-toml v1.9.4
2021-10-01T06:44:19.5154946Z go: downloading github.com/seiflotfy/cuckoofilter v0.0.0-20201222105146-bc6005554a0c
2021-10-01T06:44:22.3808017Z 
2021-10-01T06:44:22.3809024Z Building targets
2021-10-01T06:44:22.4023733Z go: downloading github.com/golang/protobuf v1.5.2
2021-10-01T06:44:22.4106899Z go: downloading google.golang.org/protobuf v1.27.1
2021-10-01T06:44:22.5582569Z go: downloading google.golang.org/grpc v1.40.0
2021-10-01T06:44:22.6912111Z go: downloading golang.org/x/net v0.0.0-20210917221730-978cfadd31cf
2021-10-01T06:44:22.6955062Z go: downloading go.starlark.net v0.0.0-20210901212718-87f333178d59
2021-10-01T06:44:22.8385739Z go: downloading golang.org/x/sys v0.0.0-20210917161153-d61c044b1678
2021-10-01T06:44:22.8417651Z go: downloading golang.org/x/crypto v0.0.0-20210920023735-84f357641f63
2021-10-01T06:44:22.9757747Z go: downloading github.com/xtls/go v0.0.0-20210920065950-d4af136d3672
2021-10-01T06:44:22.9762735Z go: downloading github.com/lucas-clemente/quic-go v0.23.0
2021-10-01T06:44:23.1159073Z go: downloading github.com/refraction-networking/utls v0.0.0-20210713165636-0b2885c8c0d4
2021-10-01T06:44:23.1178752Z go: downloading github.com/pires/go-proxyproto v0.6.1
2021-10-01T06:44:23.2532915Z go: downloading github.com/ghodss/yaml v1.0.1-0.20190212211648-25d852aebe32
2021-10-01T06:44:23.2559560Z go: downloading github.com/pelletier/go-toml v1.9.4
2021-10-01T06:44:23.3933072Z go: downloading github.com/seiflotfy/cuckoofilter v0.0.0-20201222105146-bc6005554a0c
2021-10-01T06:44:26.2633297Z ../../infra/conf/serial/loader.go:9:2: github.com/ghodss/yaml@v1.0.1-0.20190212211648-25d852aebe32: Get "https://goproxy.io/github.com/ghodss/yaml/@v/v1.0.1-0.20190212211648-25d852aebe32.zip": x509: certificate has expired or is not yet valid: current time 2021-10-01T06:44:23Z is after 2021-09-25T00:00:00Z
2021-10-01T06:44:26.2635590Z ../../common/serial/typed_message.go:7:2: github.com/golang/protobuf@v1.5.2: Get "https://goproxy.io/github.com/golang/protobuf/@v/v1.5.2.zip": x509: certificate has expired or is not yet valid: current time 2021-10-01T06:44:22Z is after 2021-09-25T00:00:00Z
2021-10-01T06:44:26.2637773Z ../../transport/internet/quic/conn.go:9:2: github.com/lucas-clemente/quic-go@v0.23.0: Get "https://goproxy.io/github.com/lucas-clemente/quic-go/@v/v0.23.0.zip": x509: certificate has expired or is not yet valid: current time 2021-10-01T06:44:23Z is after 2021-09-25T00:00:00Z
2021-10-01T06:44:26.2640338Z ../../infra/conf/serial/loader.go:10:2: github.com/pelletier/go-toml@v1.9.4: Get "https://goproxy.io/github.com/pelletier/go-toml/@v/v1.9.4.zip": x509: certificate has expired or is not yet valid: current time 2021-10-01T06:44:23Z is after 2021-09-25T00:00:00Z
2021-10-01T06:44:26.2642514Z ../../transport/internet/system_listener.go:8:2: github.com/pires/go-proxyproto@v0.6.1: Get "https://goproxy.io/github.com/pires/go-proxyproto/@v/v0.6.1.zip": x509: certificate has expired or is not yet valid: current time 2021-10-01T06:44:23Z is after 2021-09-25T00:00:00Z
2021-10-01T06:44:26.2644978Z ../../transport/internet/tls/tls.go:6:2: github.com/refraction-networking/utls@v0.0.0-20210713165636-0b2885c8c0d4: Get "https://goproxy.io/github.com/refraction-networking/utls/@v/v0.0.0-20210713165636-0b2885c8c0d4.zip": x509: certificate has expired or is not yet valid: current time 2021-10-01T06:44:23Z is after 2021-09-25T00:00:00Z
2021-10-01T06:44:26.2647586Z ../../common/antireplay/replayfilter.go:7:2: github.com/seiflotfy/cuckoofilter@v0.0.0-20201222105146-bc6005554a0c: Get "https://goproxy.io/github.com/seiflotfy/cuckoofilter/@v/v0.0.0-20201222105146-bc6005554a0c.zip": x509: certificate has expired or is not yet valid: current time 2021-10-01T06:44:23Z is after 2021-09-25T00:00:00Z
2021-10-01T06:44:26.2649928Z ../../transport/internet/tcp/hub.go:11:2: github.com/xtls/go@v0.0.0-20210920065950-d4af136d3672: Get "https://goproxy.io/github.com/xtls/go/@v/v0.0.0-20210920065950-d4af136d3672.zip": x509: certificate has expired or is not yet valid: current time 2021-10-01T06:44:23Z is after 2021-09-25T00:00:00Z
2021-10-01T06:44:26.2652088Z ../../app/router/condition.go:6:2: go.starlark.net@v0.0.0-20210901212718-87f333178d59: Get "https://goproxy.io/go.starlark.net/@v/v0.0.0-20210901212718-87f333178d59.zip": x509: certificate has expired or is not yet valid: current time 2021-10-01T06:44:22Z is after 2021-09-25T00:00:00Z
2021-10-01T06:44:26.2654187Z ../../app/router/condition.go:7:2: go.starlark.net@v0.0.0-20210901212718-87f333178d59: Get "https://goproxy.io/go.starlark.net/@v/v0.0.0-20210901212718-87f333178d59.zip": x509: certificate has expired or is not yet valid: current time 2021-10-01T06:44:22Z is after 2021-09-25T00:00:00Z
2021-10-01T06:44:26.2656304Z ../../proxy/shadowsocks/config.go:13:2: golang.org/x/crypto@v0.0.0-20210920023735-84f357641f63: Get "https://goproxy.io/golang.org/x/crypto/@v/v0.0.0-20210920023735-84f357641f63.zip": x509: certificate has expired or is not yet valid: current time 2021-10-01T06:44:22Z is after 2021-09-25T00:00:00Z
2021-10-01T06:44:26.2659216Z ../../proxy/shadowsocks/config.go:14:2: golang.org/x/crypto@v0.0.0-20210920023735-84f357641f63: Get "https://goproxy.io/golang.org/x/crypto/@v/v0.0.0-20210920023735-84f357641f63.zip": x509: certificate has expired or is not yet valid: current time 2021-10-01T06:44:22Z is after 2021-09-25T00:00:00Z
2021-10-01T06:44:26.2661284Z ../../common/ocsp/ocsp.go:11:2: golang.org/x/crypto@v0.0.0-20210920023735-84f357641f63: Get "https://goproxy.io/golang.org/x/crypto/@v/v0.0.0-20210920023735-84f357641f63.zip": x509: certificate has expired or is not yet valid: current time 2021-10-01T06:44:22Z is after 2021-09-25T00:00:00Z
2021-10-01T06:44:26.2663357Z ../../proxy/vmess/encoding/auth.go:10:2: golang.org/x/crypto@v0.0.0-20210920023735-84f357641f63: Get "https://goproxy.io/golang.org/x/crypto/@v/v0.0.0-20210920023735-84f357641f63.zip": x509: certificate has expired or is not yet valid: current time 2021-10-01T06:44:22Z is after 2021-09-25T00:00:00Z
2021-10-01T06:44:26.2665451Z ../../common/protocol/dns/io.go:7:2: golang.org/x/net@v0.0.0-20210917221730-978cfadd31cf: Get "https://goproxy.io/golang.org/x/net/@v/v0.0.0-20210917221730-978cfadd31cf.zip": x509: certificate has expired or is not yet valid: current time 2021-10-01T06:44:22Z is after 2021-09-25T00:00:00Z
2021-10-01T06:44:26.2667639Z ../../proxy/http/client.go:14:2: golang.org/x/net@v0.0.0-20210917221730-978cfadd31cf: Get "https://goproxy.io/golang.org/x/net/@v/v0.0.0-20210917221730-978cfadd31cf.zip": x509: certificate has expired or is not yet valid: current time 2021-10-01T06:44:22Z is after 2021-09-25T00:00:00Z
2021-10-01T06:44:26.2669780Z ../../transport/internet/http/hub.go:11:2: golang.org/x/net@v0.0.0-20210917221730-978cfadd31cf: Get "https://goproxy.io/golang.org/x/net/@v/v0.0.0-20210917221730-978cfadd31cf.zip": x509: certificate has expired or is not yet valid: current time 2021-10-01T06:44:22Z is after 2021-09-25T00:00:00Z
[... cut ...]
2021-10-01T06:44:26.2788125Z make[2]: *** [package/Makefile:116: package/feeds/passwall/xray-core/compile] Error 1
2021-10-01T06:44:26.2788984Z make[2]: Leaving directory '/workdir/openwrt'
2021-10-01T06:44:26.2834932Z make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2021-10-01T06:44:26.2835862Z make[1]: Leaving directory '/workdir/openwrt'
2021-10-01T06:44:26.2858101Z make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
2021-10-01T06:44:26.3007803Z ##[error]Process completed with exit code 2.

qqsir-dev, 2021-10-01:
编译之前执行这两条命令试试
sudo apt-get install apt-transport-https ca-certificates -y
sudo update-ca-certificates

Dimitri2020007, 2021-10-01:
> 编译之前执行这两条命令试试 sudo apt-get install apt-transport-https ca-certificates -y sudo update-ca-certificates

好吧我先试试

Dimitri2020007, 2021-10-01:
> 编译之前执行这两条命令试试 sudo apt-get install apt-transport-https ca-certificates -y sudo update-ca-certificates

已经亲测方法是无效的

ZoldanYao, 2021-10-05:
我的没有问题，你用科学上网试试，可能是科学上网的问题

Dimitri2020007, 2021-10-08:
> 我的没有问题，你用科学上网试试，可能是科学上网的问题

我用的actions编译

问题已经找到了

https://github.com/goproxyio/goproxy/issues/185