coolsnowwolf/lede #7973: 5.10内核无法拨号

Luastonsu, 2021-10-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述
红米AC2100路由器，编译testing内核，版本5.10
输入账号密码后拨号，没有任何反应，一直连不通，且没有任何错误信息
### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

XiaoliChan, 2021-10-01:
No log, no issue.

Luastonsu, 2021-10-02:
> No log, no issue.

I have looked through the kernel log and system log for this problem. There is no information about this problem.

Luastonsu, 2021-10-02:
> No log, no issue.

However, I can provide a screenshot of the error at that time
[image]

XiaoliChan, 2021-10-02:
> However, I can provide a screenshot of the error at that time [image]

Well, I have no idea what going on :) .

AmadeusGhost, 2021-10-02:
没有精力支持 5.10，而且 5.10 上默认少了一部分有线驱动。

closed 2021-10-03 by AmadeusGhost

Luastonsu, 2021-10-03:
> 没有精力支持 5.10，而且 5.10 上默认少了一部分有线驱动。

所以说5.10内核就抛弃非x86用户了么？

AmadeusGhost, 2021-10-03:
> 所以说5.10内核就抛弃非x86用户了么？

并没有抛弃，有这个精力可以选择自行修复

Luastonsu, 2021-10-05:
> > 所以说5.10内核就抛弃非x86用户了么？
> 
> 并没有抛弃，有精力的可以选择自行修复

那还是用着5.4内核养老算了。