coolsnowwolf/lede #7982: 麻烦大神们来看看，5.4.150还是不能用，完整系统/内核日志弄出来了

rongcenter, 2021-10-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

一直在使用5.4.140内核，这个内核版本使用无问题。
从5.4.142开始出现的问题，eth2直通口一直有问题。
是否esxi直通口挂掉是通病？

求大神们帮看看，谢谢。

### (2) 路由器型号和固件版本

X86
1、esxi+openwrt
2、wan口：正常（eth0）
3、lan口：2口，vmxnet3（eth1，正常）和直通82576（eth2，不正常）

### (3) 详细日志

【内核日志】
[    0.000000] Linux version 5.4.150 (xx@xx) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r3772-45963979b)) #0 SMP Sat Oct 2 16:44:31 2021
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=3ab0ce11-cca9-9b7c-5352-b3fe434ca202 rootwait console=tty0 console=ttyS0,115200n8 noinitrd
[    0.000000] Disabled fast string operations
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Enabled xstate features 0x3, context size is 576 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000000fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000000001000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000c0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000000cfa8fff] usable
[    0.000000] BIOS-e820: [mem 0x000000000cfa9000-0x000000000cfb1fff] type 20
[    0.000000] BIOS-e820: [mem 0x000000000cfb2000-0x000000000cfc4fff] usable
[    0.000000] BIOS-e820: [mem 0x000000000cfc5000-0x000000000cfcafff] type 20
[    0.000000] BIOS-e820: [mem 0x000000000cfcb000-0x000000000cfdafff] usable
[    0.000000] BIOS-e820: [mem 0x000000000cfdb000-0x000000000cfdffff] type 20
[    0.000000] BIOS-e820: [mem 0x000000000cfe0000-0x000000000def8fff] usable
[    0.000000] BIOS-e820: [mem 0x000000000def9000-0x000000000df18fff] type 20
[    0.000000] BIOS-e820: [mem 0x000000000df19000-0x000000000df68fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000df69000-0x000000000df84fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000000df85000-0x000000000df88fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000000df89000-0x000000003fffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000ffc00000-0x00000000ffc29fff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.31 by VMware, Inc.
[    0.000000] efi:  SMBIOS=0xdf86000  ACPI 2.0=0xdf75000  MEMATTR=0xbbd8298 
[    0.000000] SMBIOS 2.7 present.
[    0.000000] DMI: VMware, Inc. VMware7,1/440BX Desktop Reference Platform, BIOS VMW71.00V.16707776.B64.2008070230 08/07/2020
[    0.000000] vmware: hypercall mode: 0x00
[    0.000000] Hypervisor detected: VMware
[    0.000000] vmware: TSC freq read from hypervisor : 1497.600 MHz
[    0.000000] vmware: Host bus clock speed read from hypervisor : 66000000 Hz
[    0.000000] vmware: using sched offset of 3977310771 ns
[    0.000002] tsc: Detected 1497.600 MHz processor
[    0.007493] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.007496] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.007501] last_pfn = 0x40000 max_arch_pfn = 0x400000000
[    0.007528] MTRR default type: uncachable
[    0.007528] MTRR fixed ranges enabled:
[    0.007530]   00000-9FFFF write-back
[    0.007531]   A0000-FFFFF uncachable
[    0.007531] MTRR variable ranges enabled:
[    0.007533]   0 base 00000000000 mask 7FFC0000000 write-back
[... 1844 lines cut ...]
Sun Oct  3 10:33:10 2021 kern.warn kernel: [   25.260844] RBP: ffffc9000007fea0 R08: 0000000000000000 R09: 0000000000000000
Sun Oct  3 10:33:10 2021 kern.warn kernel: [   25.266618] R10: 0000000000000000 R11: 0000000000000000 R12: ffffffff82294680
Sun Oct  3 10:33:10 2021 kern.warn kernel: [   25.268187] R13: 0000000000000000 R14: 0000000000000000 R15: 0000000000000000
Sun Oct  3 10:33:10 2021 kern.warn kernel: [   25.269743]  default_idle+0x9/0x10
Sun Oct  3 10:33:10 2021 kern.warn kernel: [   25.276260]  arch_cpu_idle+0x10/0x20
Sun Oct  3 10:33:10 2021 kern.warn kernel: [   25.277532]  default_idle_call+0x27/0x30
Sun Oct  3 10:33:10 2021 kern.warn kernel: [   25.278819]  do_idle+0x19f/0x1c0
Sun Oct  3 10:33:10 2021 kern.warn kernel: [   25.280011]  cpu_startup_entry+0x18/0x20
Sun Oct  3 10:33:10 2021 kern.warn kernel: [   25.286453]  start_secondary+0x145/0x170
Sun Oct  3 10:33:10 2021 kern.warn kernel: [   25.287639]  secondary_startup_64+0xa4/0xb0
Sun Oct  3 10:33:10 2021 kern.warn kernel: [   25.288829] ---[ end trace 001b08fdb8895acb ]---
Sun Oct  3 10:33:10 2021 kern.err kernel: [   25.290415] igb 0000:03:00.0 eth2: Reset adapter
Sun Oct  3 10:33:10 2021 daemon.notice netifd: Network device 'eth2' link is down
Sun Oct  3 10:33:10 2021 kern.info kernel: [   25.485268] br-lan: port 2(eth2) entered disabled state
Sun Oct  3 10:33:13 2021 kern.info kernel: [   28.083094] igb 0000:03:00.0 eth2: igb: eth2 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
Sun Oct  3 10:33:13 2021 kern.info kernel: [   28.198405] br-lan: port 2(eth2) entered blocking state
Sun Oct  3 10:33:13 2021 kern.info kernel: [   28.204675] br-lan: port 2(eth2) entered forwarding state
Sun Oct  3 10:33:13 2021 daemon.notice netifd: Network device 'eth2' link is up
Sun Oct  3 10:34:09 2021 kern.err kernel: [   83.876729] igb 0000:03:00.0 eth2: Reset adapter
Sun Oct  3 10:34:09 2021 daemon.notice netifd: Network device 'eth2' link is down
Sun Oct  3 10:34:09 2021 kern.info kernel: [   84.068914] br-lan: port 2(eth2) entered disabled state
Sun Oct  3 10:34:12 2021 kern.info kernel: [   86.710090] igb 0000:03:00.0 eth2: igb: eth2 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
Sun Oct  3 10:34:12 2021 daemon.notice netifd: Network device 'eth2' link is up
Sun Oct  3 10:34:12 2021 kern.info kernel: [   86.821575] br-lan: port 2(eth2) entered blocking state
Sun Oct  3 10:34:12 2021 kern.info kernel: [   86.827413] br-lan: port 2(eth2) entered forwarding state

rongcenter, 2021-10-03:
@coolsnowwolf @Beginner-Go 5.10也是一样的情况。

smallprogram, 2021-10-03:
为啥我刚编译的x86就没这毛病呢。你网线是不是松了？

guqingyuan, 2021-10-03:
esxi 下直通i350 t4 有相同问题。旧版本就没事儿

guqingyuan, 2021-10-03:
> 为啥我刚编译的x86就没这毛病呢。你网线是不是松了？

我编译的x86 也是这个问题，换用桥接就没事儿，直通就会不断重启

WYC-2020, 2021-10-03:
那就用旧版本呗，修也不是一会的事

rongcenter, 2021-10-03:
> 为啥我刚编译的x86就没这毛病呢。你网线是不是松了？

你也是exsi直通？物理机估计没问题。

smallprogram, 2021-10-03:
> > 为啥我刚编译的x86就没这毛病呢。你网线是不是松了？
> 
> 
> 
> 你也是exsi直通？物理机估计没问题。

我是物理机。exsi出问题，应该不是内核的事吧。

rongcenter, 2021-10-03:
> > > 为啥我刚编译的x86就没这毛病呢。你网线是不是松了？
> > 
> > 
> > 你也是exsi直通？物理机估计没问题。
> 
> 我是物理机。exsi出问题，应该不是内核的事吧。

估计是内核的事

rongcenter, 2021-10-03:
> 那就用旧版本呗，修也不是一会的事

旧版已用着了，～(￣▽￣～)~，这个好像确实不是一会的事，先反馈出来。

rongcenter, 2021-10-03:
> esxi 下直通i350 t4 有相同问题。旧版本就没事儿

> > 为啥我刚编译的x86就没这毛病呢。你网线是不是松了？
> 
> 我编译的x86 也是这个问题，换用桥接就没事儿，直通就会不断重启

https://www.intel.cn/content/www/cn/zh/download/14098/30647/intel-network-adapter-driver-for-82575-6-82580-i350-and-i210-211-based-gigabit-network-connections-for-linux.html
用这个驱动试试，我用的5.5.2解决了，最新5.7.2不知道为啥我那个信号时不时的出现感叹号。我是用笨方法手动替换驱动的。

rongcenter, 2021-10-03:
@coolsnowwolf 大神，我用intel官方的驱动暂时解决了。

rongcenter, 2021-10-03:
[    0.000000] Linux version 5.4.150 (xx@xx) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r3772-45963979b)) #0 SMP Sat Oct 2 16:44:31 2021
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=3ab0ce11-cca9-9b7c-5352-b3fe434ca202 rootwait console=tty0 console=ttyS0,115200n8 noinitrd
[    0.000000] Disabled fast string operations
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Enabled xstate features 0x3, context size is 576 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000000fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000000001000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000c0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000000cfa8fff] usable
[    0.000000] BIOS-e820: [mem 0x000000000cfa9000-0x000000000cfb1fff] type 20
[    0.000000] BIOS-e820: [mem 0x000000000cfb2000-0x000000000cfc4fff] usable
[    0.000000] BIOS-e820: [mem 0x000000000cfc5000-0x000000000cfcafff] type 20
[    0.000000] BIOS-e820: [mem 0x000000000cfcb000-0x000000000cfdafff] usable
[    0.000000] BIOS-e820: [mem 0x000000000cfdb000-0x000000000cfdffff] type 20
[    0.000000] BIOS-e820: [mem 0x000000000cfe0000-0x000000000def8fff] usable
[    0.000000] BIOS-e820: [mem 0x000000000def9000-0x000000000df18fff] type 20
[    0.000000] BIOS-e820: [mem 0x000000000df19000-0x000000000df68fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000df69000-0x000000000df84fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000000df85000-0x000000000df88fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000000df89000-0x000000003fffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000ffc00000-0x00000000ffc29fff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.31 by VMware, Inc.
[    0.000000] efi:  SMBIOS=0xdf86000  ACPI 2.0=0xdf75000  MEMATTR=0xbbd8298 
[    0.000000] SMBIOS 2.7 present.
[    0.000000] DMI: VMware, Inc. VMware7,1/440BX Desktop Reference Platform, BIOS VMW71.00V.16707776.B64.2008070230 08/07/2020
[    0.000000] vmware: hypercall mode: 0x00
[    0.000000] Hypervisor detected: VMware
[    0.000000] vmware: TSC freq read from hypervisor : 1497.600 MHz
[    0.000000] vmware: Host bus clock speed read from hypervisor : 66000000 Hz
[    0.000000] vmware: using sched offset of 4310684818 ns
[    0.000002] tsc: Detected 1497.600 MHz processor
[    0.007575] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.007578] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.007583] last_pfn = 0x40000 max_arch_pfn = 0x400000000
[    0.007625] MTRR default type: uncachable
[    0.007626] MTRR fixed ranges enabled:
[    0.007628]   00000-9FFFF write-back
[    0.007629]   A0000-FFFFF uncachable
[    0.007630] MTRR variable ranges enabled:
[    0.007631]   0 base 00000000000 mask 7FFC0000000 write-back
[    0.007632]   1 disabled
[    0.007632]   2 disabled
[    0.007633]   3 disabled
[    0.007633]   4 disabled
[    0.007634]   5 disabled
[    0.007634]   6 disabled
[    0.007635]   7 disabled
[    0.007679] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.012976] Using GB pages for direct mapping
[    0.013609] Secure boot disabled
[    0.013612] ACPI: Early table checksum verification disabled
[    0.013617] ACPI: RSDP 0x000000000DF75000 000024 (v02 VMWARE)
[    0.013621] ACPI: XSDT 0x000000000DF75064 00005C (v01 INTEL  440BX    06040000 VMW  01324272)
[    0.013627] ACPI: SRAT 0x000000000DF750C0 0000A0 (v03 VMWARE EFISRAT  06040001 VMW  000007CE)
[    0.013631] ACPI: FACP 0x000000000DF83DB6 0000F4 (v04 INTEL  440BX    06040000 PTL  000F4240)
[    0.013636] ACPI: DSDT 0x000000000DF75160 00EC56 (v01 PTLTD  Custom   00000000 INTL 20130823)
[    0.013640] ACPI: FACS 0x000000000DF85000 000040
[    0.013643] ACPI: FACS 0x000000000DF85000 000040
[    0.013645] ACPI: APIC 0x000000000DF83F1E 00005E (v03 VMWARE EFIAPIC  06040001 VMW  000007CE)
[    0.013648] ACPI: MCFG 0x000000000DF83F7C 00003C (v01 VMWARE EFIMCFG  06040001 VMW  000007CE)
[    0.013651] ACPI: HPET 0x000000000DF83FB8 000038 (v01 VMWARE VMW HPET 00000000 VMW  00000000)
[    0.013655] ACPI: WAET 0x000000000DF84000 000028 (v01 VMWARE VMW WAET 06040001 VMW  00000001)
[    0.013658] ACPI: WSMT 0x000000000DF84028 000028 (v01 VMWARE VMW WSMT 06040001 VMW  00000001)
[    0.013661] ACPI: Reserving SRAT table memory at [mem 0xdf750c0-0xdf7515f]
[    0.013662] ACPI: Reserving FACP table memory at [mem 0xdf83db6-0xdf83ea9]
[    0.013663] ACPI: Reserving DSDT table memory at [mem 0xdf75160-0xdf83db5]
[    0.013665] ACPI: Reserving FACS table memory at [mem 0xdf85000-0xdf8503f]
[... 1114 lines cut ...]
[    6.422487] PPP MPPE Compression module registered
[    6.424248] NET: Registered protocol family 24
[    6.435345] usbcore: registered new interface driver rt2800usb
[    6.437856] usbcore: registered new interface driver ath9k_htc
[    6.440505] kmodloader: done loading kernel modules from /etc/modules.d/*
[    9.606970] random: crng init done
[   10.264273] vmxnet3 0000:15:00.0 eth1: intr type 3, mode 0, 3 vectors allocated
[   10.266363] vmxnet3 0000:15:00.0 eth1: NIC Link is Up 10000 Mbps
[   10.267900] 8021q: adding VLAN 0 to HW filter on device eth1
[   10.269872] br-lan: port 1(eth1) entered blocking state
[... cut ...]
[   10.474600] vmxnet3 0000:0d:00.0 eth0: intr type 3, mode 0, 3 vectors allocated
[   10.476723] vmxnet3 0000:0d:00.0 eth0: NIC Link is Up 10000 Mbps
[   10.478235] 8021q: adding VLAN 0 to HW filter on device eth0
[   11.287030] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   12.291619] igb 0000:03:00.0 eth2: igb: eth2 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
[   12.293330] br-lan: port 2(eth2) entered blocking state
[   12.294528] br-lan: port 2(eth2) entered forwarding state

closed 2021-10-04 by rongcenter

tdg2088, 2021-11-12:
自行编译igb.ko并手动替换解决么？ @rongcenter