coolsnowwolf/lede #7983: 1K最近固件的 端口转发 NAT环回 不生效？

82kg, 2021-10-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
最近编译的两版RAC2V1K的固件，端口转发里的 NAT环回 功能失效了。怀疑是不是新版的QCA-NSS-ECM-SFE驱动造成

### (1) 具体问题

A：端口转发，内网无法使用ddns的域名访问，只能用内网ip访问某个端口

### (2) 路由器型号和固件版本

A：型号：RAC2V1K，Askey RT4230W REV6
   固件版本 OpenWrt R21.10.1 / LuCI Master (git-21.273.48290-ff4b0f6)

### (3) 详细日志

A：插件就一个ssr+,动态dns，解锁网易灰色，frps，迅雷快鸟，全能推送。

Ailick, 2021-10-07:
QCA-NSS-ECM-SFE其实目前并不会使用sfe加速，也就是说你勾选SFE FOR ECM与否，并不会导致结果有什么不同，你可以试试不选择sfe编译。
SFE FOR ECM目前还有一些别的问题需要解决，这原因比较复杂，简而言之你遇到的问题与SFE无关。

Ailick, 2021-10-08:
> > QCA-NSS-ECM-SFE其实目前并不会使用sfe加速，也就是说你勾选SFE FOR ECM与否，并不会导致结果有什么不同，你可以试试不选择sfe编译。 SFE FOR ECM目前还有一些别的问题需要解决，这原因比较复杂，简而言之你遇到的问题与SFE无关。
> 
> lean说旧版没这问题。我晚上回去试试 不勾选 ecm，用旧版的试试。 [image]

有nss的设备可以不用勾选sfe，另外新版sfe也不存在端口转发问题，因为我自己的设备正常运行，你需要担心的是你的防火墙而不是这些转发加速的驱动。
目前806x和807x遇到的问题与sfe-cm无关，sfe for ecm使用的是sfe-drv驱动，并且此驱动目前尚未生效。

closed 2021-10-16 by 82kg