coolsnowwolf/lede #7985: 编译老是报错，

Yofo-me, 2021-10-03:
> Ubuntu 20.04 lts 编译时一直报错，提示“ath11k.ko is missing.”但是之前执行` make download -j8 V=S`没有报错啊，`make -j1 V=s`就不行

> 编译环境：
- Ubuntu 20.04 LTS
- 国外的千兆独服

> 相关日志：

```shell
...
rstrip.sh: /home/yokimu/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/ipkg-x86_64/kmod-ath10k/lib/modules/5.4.145/ath10k_pci.ko: relocatable
(cd /home/yokimu/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/ipkg-x86_64/kmod-ath10k/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm;  )
install -d -m0755 /home/yokimu/lede/bin/targets/x86/64/packages
/home/yokimu/lede/staging_dir/host/bin/fakeroot /home/yokimu/lede/scripts/ipkg-build -m "" /home/yokimu/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/ipkg-x86_64/kmod-ath10k /home/yokimu/lede/bin/targets/x86/64/packages
Packaged contents of /home/yokimu/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/ipkg-x86_64/kmod-ath10k into /home/yokimu/lede/bin/targets/x86/64/packages/kmod-ath10k_5.4.145+5.10.42-1-3_x86_64.ipk
mkdir -p /home/yokimu/lede/bin/targets/x86/64/packages /home/yokimu/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/ipkg-x86_64/kmod-ath11k/CONTROL /home/yokimu/lede/staging_dir/target-x86_64_musl/pkginfo
ERROR: module '/home/yokimu/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath11k/ath11k.ko' is missing.
make[3]: *** [Makefile:574: /home/yokimu/lede/bin/targets/x86/64/packages/kmod-ath11k_5.4.145+5.10.42-1-3_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/yokimu/lede/package/kernel/mac80211'
time: package/kernel/mac80211/compile#583.05#83.94#661.68
    ERROR: package/kernel/mac80211 failed to build.
make[2]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[2]: Leaving directory '/home/yokimu/lede'
make[1]: *** [package/Makefile:110: /home/yokimu/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/yokimu/lede'
make: *** [/home/yokimu/lede/include/toplevel.mk:230: world] Error 2

```
---

> 执行四次`make clean` && `make -j1 V=s`后还是不行

by-JohnChen, 2021-10-04:
ath11.ko这个内核驱动文件缺失了

kokeri, 2021-10-17:
同样的错误，不知道怎么解决。换过内核版本，换了mac80211的源码成openwrt的，都不行

TonyTeng22, 2022-05-05:
5.15.8 遇到同样的问题,不知怎么解决
ouch /media/tony/HDD/lede/build_dir/target-arm_xscale_musl_eabi/linux-kirkwood/backports-5.15.8-1/.pkgdir/kmod-ath10k.installed
mkdir -p /media/tony/HDD/lede/bin/targets/kirkwood/generic/packages /media/tony/HDD/lede/build_dir/target-arm_xscale_musl_eabi/linux-kirkwood/backports-5.15.8-1/ipkg-arm_xscale/kmod-ath11k/CONTROL /media/tony/HDD/lede/staging_dir/target-arm_xscale_musl_eabi/pkginfo
ERROR: module '/media/tony/HDD/lede/build_dir/target-arm_xscale_musl_eabi/linux-kirkwood/backports-5.15.8-1/drivers/net/wireless/ath/ath11k/ath11k.ko' is missing.
make[3]: *** [Makefile:555: /media/tony/HDD/lede/bin/targets/kirkwood/generic/packages/kmod-ath11k_5.4.191+5.15.8-1-2_arm_xscale.ipk] Error 1
make[3]: Leaving directory '/media/tony/HDD/lede/package/kernel/mac80211'
time: package/kernel/mac80211/regular/compile#308.10#84.04#249.18
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[2]: Leaving directory '/media/tony/HDD/lede'
make[1]: *** [package/Makefile:110: /media/tony/HDD/lede/staging_dir/target-arm_xscale_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/media/tony/HDD/lede'
make: *** [/media/tony/HDD/lede/include/toplevel.mk:230：world] 错误 2