coolsnowwolf/lede #7986: docker bridge 无法联网

bilibili20001, 2021-10-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：docker host正常，但是bridge就无法联网，参考了issue里面的一些设置方法，都没有用，应该还是防火墙的问题
https://github.com/coolsnowwolf/lede/issues/7408
https://github.com/coolsnowwolf/lede/issues/1760#issuecomment-542985373

### (2) 路由器型号和固件版本

A：x86-64 (N3150工控机)
      好几个月没有编译过固件，之前的没问题，最近这个月编译的都不行

### (3) 详细日志

A：
Thu Sep 30 19:59:19 2021 daemon.notice netifd: bridge 'docker0' link is up
Thu Sep 30 19:59:19 2021 daemon.notice netifd: Interface 'docker' has link connectivity
Thu Sep 30 19:59:53 2021 daemon.notice ttyd[4467]: HTTP / - 192.168.5.236
Thu Sep 30 19:59:53 2021 daemon.notice ttyd[4467]: HTTP /token - 192.168.5.236
Thu Sep 30 19:59:53 2021 daemon.notice ttyd[4467]: WS   /ws - 192.168.5.236, clients: 1
Thu Sep 30 19:59:53 2021 daemon.notice ttyd[4467]: started process, pid: 10578
Thu Sep 30 20:00:22 2021 daemon.notice ttyd[4467]: WS closed from 192.168.5.236, clients: 0
Thu Sep 30 20:00:22 2021 daemon.notice ttyd[4467]: killing process 10578 with signal: 1 (SIGHUP)
Thu Sep 30 20:00:22 2021 daemon.notice ttyd[4467]: process killed with signal 1 (SIGHUP), pid: 10578
Thu Sep 30 20:00:44 2021 user.info : Restarting firewall on custom /etc/firewall.user change
Thu Sep 30 20:00:56 2021 daemon.notice ttyd[4467]: HTTP / - 192.168.5.236
Thu Sep 30 20:00:56 2021 daemon.notice ttyd[4467]: HTTP /token - 192.168.5.236
Thu Sep 30 20:00:56 2021 daemon.notice ttyd[4467]: WS   /ws - 192.168.5.236, clients: 1
Thu Sep 30 20:00:56 2021 daemon.notice ttyd[4467]: started process, pid: 10832
Thu Sep 30 20:01:31 2021 daemon.err dockerd[3886]: time="2021-09-30T12:01:31.591346249Z" level=error msg="copy shim log" error="read /proc/self/fd/14: file already closed"
Thu Sep 30 20:01:31 2021 kern.info kernel: [  316.320292] docker0: port 1(veth3094fec) entered disabled state
Thu Sep 30 20:01:31 2021 kern.info kernel: [  316.321241] veth74c2294: renamed from eth0
Thu Sep 30 20:01:31 2021 kern.info kernel: [  316.381194] docker0: port 1(veth3094fec) entered disabled state
Thu Sep 30 20:01:31 2021 kern.info kernel: [  316.384856] device veth3094fec left promiscuous mode
Thu Sep 30 20:01:31 2021 kern.info kernel: [  316.385272] docker0: port 1(veth3094fec) entered disabled state
Thu Sep 30 20:01:31 2021 daemon.notice netifd: bridge 'docker0' link is down
Thu Sep 30 20:01:31 2021 daemon.notice netifd: Interface 'docker' has link connectivity loss
Thu Sep 30 20:01:31 2021 kern.info kernel: [  316.574463] docker0: port 1(veth3a20643) entered blocking state
Thu Sep 30 20:01:31 2021 kern.info kernel: [  316.574951] docker0: port 1(veth3a20643) entered disabled state
Thu Sep 30 20:01:31 2021 kern.info kernel: [  316.575569] device veth3a20643 entered promiscuous mode
Thu Sep 30 20:01:31 2021 kern.info kernel: [  316.576221] docker0: port 1(veth3a20643) entered blocking state
Thu Sep 30 20:01:31 2021 kern.info kernel: [  316.576721] docker0: port 1(veth3a20643) entered forwarding state
Thu Sep 30 20:01:31 2021 daemon.err dockerd[3886]: time="2021-09-30T12:01:31.966423169Z" level=warning msg="Failed to allocate and map port 5678-5678:  (iptables failed: iptables -t nat -A DOCKER -p tcp -d 0/0 --dport 5678 -j DNAT --to-destination 172.17.0.2:5678 ! -i docker0: iptables: No chain/target/match by that name.\n (exit status 1))"
Thu Sep 30 20:01:31 2021 kern.info kernel: [  316.668859] docker0: port 1(veth3a20643) entered disabled state
Thu Sep 30 20:01:31 2021 kern.info kernel: [  316.671181] device veth3a20643 left promiscuous mode
Thu Sep 30 20:01:31 2021 kern.info kernel: [  316.671579] docker0: port 1(veth3a20643) entered disabled state

EnnawYang, 2021-10-03:
https://github.com/lisaac/luci-app-dockerman/issues/126#issuecomment-922695135

EnnawYang, 2021-10-03:
还有一个方法就是开机之后手动重启下 `dockerd` (在使用了 `luci-app-docker` 并开放了wan的情况下)

bilibili20001, 2021-10-04:
> [lisaac/luci-app-dockerman#126 (comment)](https://github.com/lisaac/luci-app-dockerman/issues/126#issuecomment-922695135)

删除 config firewall 'firewall' 那部分后还是不行……
重启 dockerd服务 也不行……

bilibili20001, 2021-10-04:
目前重新编译固件问题已解决，应该是大佬们更新修复了bug
（config文件没有改动，开机后也没有更改任何设置，直接就可以正常使用了）
https://github.com/bilibili20001/Actions-OpenWrt/actions/runs/1302222145

Deleted-Account, 2021-10-16:
Firewall-->General Settings-->Forward-->Accept

coolsnowwolf, 2026-09-06:
Docker bridge 网络依赖 Docker、firewall、宿主机路由和自定义 firewall.user 配置，报告没有完整配置、iptables/nft 规则或容器路由信息，无法定位当前包缺陷，关闭为 not planned。

closed 2026-09-06 by coolsnowwolf