coolsnowwolf/lede #7987: RAC2V1K Rev:9.3 不开机，重复报krait-cache soc:l2-cache: Failed

coeoe, 2021-10-03:
反馈bug/问题模板，提建议请删除

1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
 [X ] 没有类似的issue

 2. 详细叙述

 (1) 具体问题

RAC2V1K 通电后不能正常启动，五个网口均没反应，不亮灯。
接TTL线后通电，发现循环报

[    3.720408] krait-cache soc:l2-cache: Failed to enable regulator: -5
[    3.720436] cpufreq: __target_index: Failed to change cpu frequency: -5

指示灯状态：通电先亮红灯约2s，然后蓝灯闪烁，最后蓝灯常亮。
尝试过uboot重新刷机，但因网口不识别失败。

感谢各位

 (2) 路由器型号和固件版本

型号：RAC2V1K Rev:9.3 （Askey RT4230W REV9.3 ）
闪存：Macronix MX30UF2G18AC 256 MiB, SLC
固件：OpenWrt R21.10.1 / LuCI Master (git-21.273.48290-ff4b0f6) (编译时选Rev:9.3)

 (3) 详细日志

U-Boot 1.0.6 [spf.5.0.2_csu1.1] (Jan 15 2020 - 03:36:44)

smem ram ptable found: ver: 0 len: 5
DRAM:  1003 MiB
setting 0x177e as machine type from smem
NAND:  SF: Unsupported manufacturer 00
ipq_spi: SPI Flash not found (bus/cs/speed/mode) = (0/0/48000000/0)
256 MiB
MMC:
In:    serial
Out:   serial
Err:   serial
MMC Device 0 not found
cdp: get part failed for 0:HLOS
Net:   MAC0 addr:c8:b4:22:cb:54:7
switch init failed. reset switch...
Switch inits failed
Board Net Initialization Failed
No ethernet found.
Hit space key to stop autoboot:  0
Creating 1 MTD partitions on "nand0":
0x000002400000-0x000010000000 : "mtd=0"
UBI: attaching mtd1 to ubi0
UBI: physical eraseblock size:   131072 bytes (128 KiB)
UBI: logical eraseblock size:    126976 bytes
UBI: smallest flash I/O unit:    2048
UBI: VID header offset:          2048 (aligned 2048)
UBI: data offset:                4096
UBI: attached mtd1 to ubi0
UBI: MTD device name:            "mtd=0"
UBI: MTD device size:            220 MiB
UBI: number of good PEBs:        1760
UBI: number of bad PEBs:         0
UBI: max. allowed volumes:       128
UBI: wear-leveling threshold:    4096
UBI: number of internal volu[Y.'鈻抝
                                  UBI: number of user volumes:     3
UBI: available PEBs:             23
UBI: total number of reserved PEBs: 1737
UBI: number of PEBs reserved for bad PEB handling: 17
[... 2644 lines cut ...]
[   89.498814] cpufreq: __target_index: Failed to change cpu frequency: -5
[   89.627384] krait-cache soc:l2-cache: Failed to enable regulator: -5
[   89.627413] cpufreq: __target_index: Failed to change cpu frequency: -5
[   89.688873] krait-cache soc:l2-cache: Failed to enable regulator: -5
[   89.688903] cpufreq: __target_index: Failed to change cpu frequency: -5
[   89.704151] krait-cache soc:l2-cache: Failed to enable regulator: -5
[   89.704181] cpufreq: __target_index: Failed to change cpu frequency: -5
[   89.783928] krait-cache soc:l2-cache: Failed to enable regulator: -5
[   89.783958] cpufreq: __target_index: Failed to change cpu frequency: -5
[   90.141328] krait-cache soc:l2-cache: Failed to enable regulator: -5
[   90.141360] cpufreq: __target_index: Failed to change cpu frequency: -5
[   90.150628] krait-cache soc:l2-cache: Failed to enable regulator: -5
[   90.153125] cpufreq: __target_index: Failed to change cpu frequency: -5
[   90.172669] krait-cache soc:l2-cache: Failed to enable regulator: -5
[   90.172700] cpufreq: __target_index: Failed to change cpu frequency: -5
[   90.202478] krait-cache soc:l2-cache: Failed to enable regulator: -5
[   90.202509] cpufreq: __target_index: Failed to change cpu frequency: -5
[   90.318346] krait-cache soc:l2-cache: Failed to enable regulator: -5
[   90.318375] cpufreq: __target_index: Failed to change cpu frequency: -5
[   91.208365] krait-cache soc:l2-cache: Failed to enable regulator: -5
[   91.208399] cpufreq: __target_index: Failed to change cpu frequency: -5
[   91.257777] krait-cache soc:l2-cache: Failed to enable regulator: -5
[   91.257809] cpufreq: __target_index: Failed to change cpu frequency: -5
[   91.267757] krait-cache soc:l2-cache: Failed to enable regulator: -5
[   91.269684] cpufreq: __target_index: Failed to change cpu frequency: -5

AmadeusGhost, 2021-10-07:
这个是你硬件坏了。

closed 2021-11-02 by AmadeusGhost

hzjixiang, 2021-11-20:
> 重复插电断电有几率启动系统

不接电放一个小时，再接电，很大概率可以开机。如果短时间拔插电源，基本都开不了机。

coeoe, 2021-12-03:
> > 重复插电断电有几率启动系统
> 
> 不接电放一个小时，再接电，很大概率可以开机。如果短时间拔插电源，基本都开不了机。

谢谢，目前已恢复正常。拆掉无线网卡并重新刷机后可正常使用。