coolsnowwolf/lede #7989: 运行lighttpd与nginx报错symbol not found

jomplin, 2021-10-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：运行lighttpd与nginx报错

### (2) 路由器型号和固件版本

A：竞斗云2.0 P&W R619AC 128M
OpenWrt R21.10.1 / LuCI Master (git-21.273.48290-ff4b0f6)
云编译完成

### (3) 详细日志

A：Error relocating /usr/sbin/lighttpd: __gmtime64_r: symbol not found
Error relocating /usr/sbin/lighttpd: __select_time64: symbol not found
Error relocating /usr/sbin/lighttpd: __fstat_time64: symbol not found
Error relocating /usr/sbin/lighttpd: __clock_gettime64: symbol not found
Error relocating /usr/sbin/lighttpd: __timegm_time64: symbol not found
Error relocating /usr/sbin/lighttpd: __localtime64_r: symbol not found
Error relocating /usr/sbin/lighttpd: __stat_time64: symbol not found
Error relocating /usr/sbin/lighttpd: __time64: symbol not found
Error relocating /usr/sbin/lighttpd: __lstat_time64: symbol not found
Error relocating /usr/sbin/lighttpd: __dlsym_time64: symbol not found

Error relocating /lib/libubox.so.20210819: __clock_gettime64: symbol not found
Error relocating /lib/libubus.so.20210630: __clock_gettime64: symbol not found
Error relocating /usr/bin/nginx-util: _ZNSt15__exception_ptr13exception_ptr9_M_addrefEv: symbol not found
Error relocating /usr/bin/nginx-util: _ZSt28__throw_bad_array_new_lengthv: symbol not found
Error relocating /usr/bin/nginx-util: _ZNSt15__exception_ptr13exception_ptr10_M_releaseEv: symbol not found
Error relocating /usr/bin/nginx-util: __lstat_time64: symbol not found
Error relocating /usr/bin/nginx-util: __time64: symbol not found
Error relocating /lib/libubox.so.20210819: __clock_gettime64: symbol not found
Error relocating /lib/libubus.so.20210630: __clock_gettime64: symbol not found
Error relocating /usr/bin/nginx-util: _ZNSt15__exception_ptr13exception_ptr9_M_addrefEv: symbol not found
Error relocating /usr/bin/nginx-util: _ZSt28__throw_bad_array_new_lengthv: symbol not found
Error relocating /usr/bin/nginx-util: _ZNSt15__exception_ptr13exception_ptr10_M_releaseEv: symbol not found
Error relocating /usr/bin/nginx-util: __lstat_time64: symbol not found
Error relocating /usr/bin/nginx-util: __time64: symbol not found

dfbucy090, 2021-10-17:
root@OpenWrt:~# chmod +x cf-auto-passwall.sh && bash cf-auto-passwall.sh
Error relocating /bin/bash: __select_time64: symbol not found
Error relocating /bin/bash: __fstat_time64: symbol not found
Error relocating /bin/bash: __getrusage_time64: symbol not found
Error relocating /bin/bash: __stat_time64: symbol not found
Error relocating /bin/bash: __time64: symbol not found
Error relocating /bin/bash: __pselect_time64: symbol not found
Error relocating /bin/bash: __gettimeofday_time64: symbol not found
Error relocating /bin/bash: __localtime64: symbol not found
Error relocating /bin/bash: __lstat_time64: symbol not found
Error relocating /bin/bash: __dlsym_time64: symbol not found
Error relocating /bin/bash: __setitimer_time64: symbol not found
root@OpenWrt:~# apt-get update
-ash: apt-get: not found
root@OpenWrt:~#
同样的问题，是32位时间问题，是不是要添加什么更新插件？

jomplin, 2021-10-30:
> 你的问题解决了吗？怎么解决的？

编译的时候就把lighttpd编译进去

closed 2021-11-12 by AmadeusGhost