coolsnowwolf/lede #7998: ZTE MF269 编译报错

ldy8182, 2021-10-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译报错

### (2) 路由器型号和固件版本

A：ZTE MF269

### (3) 详细日志

A：Applying /home/ldy8182/lede/target/linux/ipq807x/patches-5.10/118-arm64-dts-ipq8074-Fixup-PCIe-dts-nodes.patch using plaintext: 
patching file arch/arm64/boot/dts/qcom/ipq8074.dtsi
Hunk #2 FAILED at 703.
Hunk #4 FAILED at 764.
2 out of 6 hunks FAILED -- saving rejects to file arch/arm64/boot/dts/qcom/ipq8074.dtsi.rej
Patch failed!  Please fix /home/ldy8182/lede/target/linux/ipq807x/patches-5.10/118-arm64-dts-ipq8074-Fixup-PCIe-dts-nodes.patch!
make[4]: *** [Makefile:31: /home/ldy8182/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.70/.prepared_6161c9299e275af7d90715c4a64e2557] Error 1
make[4]: Leaving directory '/home/ldy8182/lede/target/linux/ipq807x'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/ldy8182/lede/target/linux'
time: target/linux/compile#1.12#8.72#7.28
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/compile] Error 1
make[2]: Leaving directory '/home/ldy8182/lede'
make[1]: *** [target/Makefile:18: /home/ldy8182/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/ldy8182/lede'
make: *** [/home/ldy8182/lede/include/toplevel.mk:230: world] Error 2

Boos4721, 2021-10-05:
`target/linux/ipq807x/patches-5.10/118-arm64-dts-ipq8074-Fixup-PCIe-dts-nodes.patch`
> 把这个删掉  https://github.com/Boos4721/openwrt/commit/3e68644ebc187e97c89e4601c6923df3fbdfc1a0#diff-555e2653d41c620220e804ee7afb4b645d801339a6aa886f6ad3d0f58f57d652 在5.10.67就修复了

ldy8182, 2021-10-05:
HOSTCC  scripts/asn1_compiler
  DTC     arch/arm64/boot/dts/qcom/apq8016-sbc.dtb
  DTC     arch/arm64/boot/dts/qcom/apq8096-db820c.dtb
  DTC     arch/arm64/boot/dts/qcom/apq8096-ifc6640.dtb
  DTC     arch/arm64/boot/dts/qcom/ipq6018-cp01-c1.dtb
  DTC     arch/arm64/boot/dts/qcom/ipq8071-ax6.dtb
  DTC     arch/arm64/boot/dts/qcom/ipq8071-ax3600.dtb
  DTC     arch/arm64/boot/dts/qcom/ipq8071-mf269.dtb
  DTC     arch/arm64/boot/dts/qcom/ipq8072-ax9000.dtb
  DTC     arch/arm64/boot/dts/qcom/ipq8074-sxr80.dtb
  DTC     arch/arm64/boot/dts/qcom/ipq8074-hk01.dtb
  DTC     arch/arm64/boot/dts/qcom/msm8916-mtp.dtb
  DTC     arch/arm64/boot/dts/qcom/msm8916-longcheer-l8150.dtb
  DTC     arch/arm64/boot/dts/qcom/msm8916-samsung-a3u-eur.dtb
  DTC     arch/arm64/boot/dts/qcom/msm8916-samsung-a5u-eur.dtb
Error: arch/arm64/boot/dts/qcom/ipq8071-ax3600.dts:43.1-15 Label or path qmp_pcie_phy0 not found
FATAL ERROR: Syntax error parsing input tree
make[7]: *** [scripts/Makefile.lib:326: arch/arm64/boot/dts/qcom/ipq8071-ax3600.dtb] Error 1
make[7]: *** Waiting for unfinished jobs....
Error: arch/arm64/boot/dts/qcom/ipq8072-ax9000.dts:227.1-15 Label or path qmp_pcie_phy0 not found
Error: arch/arm64/boot/dts/qcom/ipq8072-ax9000.dts:237.1-15 Label or path qmp_pcie_phy1 not found
FATAL ERROR: Syntax error parsing input tree
make[7]: *** [scripts/Makefile.lib:326: arch/arm64/boot/dts/qcom/ipq8072-ax9000.dtb] Error 1
make[6]: *** [scripts/Makefile.build:497: arch/arm64/boot/dts/qcom] Error 2
make[5]: *** [Makefile:1359: dtbs] Error 2
make[5]: *** Waiting for unfinished jobs....
make[5]: Leaving directory '/home/ldy8182/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.70'
make[4]: *** [Makefile:22: /home/ldy8182/lede/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.10.70/.modules] Error 2
make[4]: Leaving directory '/home/ldy8182/lede/target/linux/ipq807x'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/ldy8182/lede/target/linux'
time: target/linux/compile#10.22#10.51#14.90
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/compile] Error 1
make[2]: Leaving directory '/home/ldy8182/lede'
make[1]: *** [target/Makefile:18: /home/ldy8182/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/ldy8182/lede'
make: *** [/home/ldy8182/lede/include/toplevel.mk:230: world] Error 2

Boos4721, 2021-10-05:
对着后面改下https://github.com/Boos4721/openwrt/commit/2e4e4c9ef64162b8d769bf3a95760796a8f5a12a
我等下提个pr修

ldy8182, 2021-10-05:
那我等大佬pr了，辛苦了~~

Boos4721, 2021-10-05:
https://we.tl/t-lj1llusVvk用这个patch试试

baodp, 2021-10-05:
帮帮我们吧！！！

Boos4721, 2021-10-06:
> > [用这个patch试试](https://we.tl/t-lj1llusVvk)
> 
> 链接网址不对吧

没有问题

ldy8182, 2021-10-06:
大佬，这个patch应该怎么用？

Boos4721, 2021-10-06:
> 大佬，这个patch应该怎么用？

git apply a.patch

ldy8182, 2021-10-06:
谢谢大佬~编译成功了

baodp, 2021-10-06:
感谢。编译成功了

Boos4721, 2021-10-06:
> > [用这个patch试试](https://we.tl/t-lj1llusVvk)
> 
> pr下吧，自己打补丁也不是个事

need more test

closed 2021-10-07 by coolsnowwolf