coolsnowwolf/lede #8000: 在拉取源码，进行updata，日志会提示没有找到文件或目录

forever2019, 2021-10-05:
在拉取源码，进行updata时，日志会提示没有找到文件或目录

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述
在运行 ./scripts/feeds update -a 之后，日志提示的

猜测或许是他们的问题
src-git routing https://git.openwrt.org/feed/routing.git
src-git telephony https://git.openwrt.org/feed/telephony.git

[image]

### (2) 路由器型号和固件版本

A：D2550 x86-64

ZoldanYao, 2021-10-05:
应该是源的问题，我用的是以下的源，可以正常update和install
src-git packages https://github.com/coolsnowwolf/packages
src-git luci https://github.com/coolsnowwolf/luci
src-git routing https://git.openwrt.org/feed/routing.git
src-git telephony https://git.openwrt.org/feed/telephony.git
src-git small https://github.com/kenzok8/small
src-git kenzo https://github.com/kenzok8/openwrt-packages

tuzhq, 2021-10-05:
遇到同样问题：
Updating feed 'routing' from 'https://git.openwrt.org/feed/routing.git' ...
fatal: unable to access 'https://git.openwrt.org/feed/routing.git/': server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
failed.
Updating feed 'telephony' from 'https://git.openwrt.org/feed/telephony.git' ...
fatal: unable to access 'https://git.openwrt.org/feed/telephony.git/': server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
failed.

Luastonsu, 2021-10-06:
同 还以为是自己网络环境问题

Her0R0cket, 2021-10-07:
更新下编译系统的软件包,可能ca证书包要更新了

DanielChen008, 2021-10-12:
sudo apt update
sudo apt upgrade
更新一下即可

closed 2021-10-14 by forever2019