coolsnowwolf/lede #8001: /etc/config/network新增接口

ZoldanYao, 2021-10-05:
编译时添加了gre grev6还有kmod-gre,kmod-gre6

之后在network中如下新增接口，重启服务或者重启机器后，无法在shell下使用ip add查看到接口
config interface 'ip6gre10'
	option ip6addr ' FC00:1::0'
	option peer6addr 'FC00:2::0'
	option mtu '1400'
	option proto 'ip6gre'
 
config interface 'tun10'
	option proto 'static'
	option ifname '@ip6gre10'
	option ipaddr '172.16.100.1'
	option netmask '255.255.255.252'

如果使用命令行直接添加

ip link add name gre10 type ip6gre local FC00:1::0 remote  FC00:2::0
ip link set dev gre10 up
ip addr add 172.16.100.1/30 dev gre10

则可以成功创建接口，并且经过测试，grev6是生效的。
如何将接口信息写死在network文件中并使其生效呢？

WYC-2020, 2021-10-06:
怕是哪里每写对吧，把你可以的时候的network文件拿来对比一下

coolsnowwolf, 2026-09-06:
该配置使用了错误的 ip6gre 选项名（ip6addr/peer6addr 并不是 netifd 当前协议参数），且没有 netifd 日志；命令行创建成功不能证明 UCI 配置缺陷，关闭为 not planned。

closed 2026-09-06 by coolsnowwolf