coolsnowwolf/lede #8006: OP断网再联上，QEMU虚拟机就无法联网

feng5064, 2021-10-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：以OP作为宿主机，然后用QEMU虚拟了WIN10和黑群。
现在的问题是，OP一旦断网（例如营运商强制断网），即使OP再联上网，但QEMU的虚拟机再也没法联网，也没法获得IP。所有虚拟机都这样。即使重启虚拟机也不行（win10可以用VNC控制重启），只能KILL掉qeum的进程重新启动才可以正常。
我想可能不是OP的问题，感觉可能是QEMU本身的问题。内核版本5.4和5.10都试过，都存在这个问题。

我win10虚拟机的运行命令如下：
qemu-system-x86_64 -enable-kvm -nodefaults \
        -machine q35,accel=kvm -vga virtio \
        -cpu host -smp 8,cores=4,threads=2 -m 6G \
         -drive file=/mnt/sdb2/qemu/Win10x64.qcow2 \
       -device virtio-net-pci,mac=EE:FF:01:02:01:02,netdev=br1 -netdev bridge,br=br-lan,id=br1 \
        -usb -device usb-kbd -device usb-tablet \
        -vnc :1,websocket -k en-us\
        -monitor stdio

### (2) 路由器型号和固件版本

A：软路由Genuine Intel(R) CPU 0000 @ 2.40GHz : 6 Core 12 Thread 
固件版本	OpenWrt R21.10.1 / LuCI Master (git-21.273.48290-ff4b0f6)
内核版本	5.4.145
电信单线双拨
### (3) 详细日志

A：Wed Oct  6 12:18:27 2021 daemon.info pppd[16427]: LCP terminated by peer
Wed Oct  6 12:18:27 2021 daemon.info pppd[16427]: Connect time 2885.0 minutes.
Wed Oct  6 12:18:27 2021 daemon.info pppd[16427]: Sent 4247615661 bytes, received 786114058 bytes.
Wed Oct  6 12:18:27 2021 daemon.notice netifd: Interface 'vwan1' has lost the connection
Wed Oct  6 12:18:27 2021 daemon.notice netifd: Interface 'vwan1_6' is now down
Wed Oct  6 12:18:27 2021 daemon.notice netifd: Interface 'vwan1_6' is disabled
Wed Oct  6 12:18:27 2021 daemon.notice netifd: Interface 'vwan1_6' has link connectivity loss
Wed Oct  6 12:18:27 2021 daemon.notice pppd[16427]: Modem hangup
Wed Oct  6 12:18:27 2021 daemon.notice pppd[16427]: Connection terminated.
Wed Oct  6 12:18:27 2021 daemon.info pppd[16427]: Connect time 2885.0 minutes.
Wed Oct  6 12:18:27 2021 daemon.info pppd[16427]: Sent 4247615661 bytes, received 786114058 bytes.
Wed Oct  6 12:18:27 2021 kern.info kernel: [346247.956613] netifd[2963]: segfault at 501 ip 00000000004189e8 sp 00007fff90488fe0 error 4 in netifd[404000+1b000]
Wed Oct  6 12:18:27 2021 kern.info kernel: [346247.967329] Code: 55 53 48 89 fb 52 48 83 bf 70 01 00 00 00 74 0c 48 8d bf 48 01 00 00 e8 ec 86 ff ff 48 8b ab f0 04 00 00 4c 8d ab f0 04 00 00 <4c> 8b 65 00 48 8b 83 f8 04 00 00 48 39 45 08 74 1e 48 89 ee 4c 89
Wed Oct  6 12:18:28 2021 daemon.info pppd[16427]: Sent PADT
Wed Oct  6 12:18:28 2021 daemon.info pppd[16426]: LCP terminated by peer
Wed Oct  6 12:18:28 2021 daemon.info pppd[16426]: Connect time 2885.0 minutes.
Wed Oct  6 12:18:28 2021 daemon.info pppd[16426]: Sent 522979315 bytes, received 2551782136 bytes.
Wed Oct  6 12:18:28 2021 daemon.notice pppd[16426]: Modem hangup
Wed Oct  6 12:18:28 2021 daemon.notice pppd[16426]: Connection terminated.
Wed Oct  6 12:18:28 2021 daemon.info pppd[16426]: Connect time 2885.0 minutes.
Wed Oct  6 12:18:28 2021 daemon.info pppd[16426]: Sent 522979315 bytes, received 2551782136 bytes.
Wed Oct  6 12:18:28 2021 daemon.info pppd[16426]: Sent PADT
Wed Oct  6 12:18:28 2021 daemon.info pppd[16426]: Exit.
Wed Oct  6 12:18:28 2021 user.notice mwan3[2421]: Execute ifdown event on interface vwan1 (unknown)
Wed Oct  6 12:18:28 2021 user.info mwan3track[27895]: Detect ifdown event on interface vwan1 (pppoe-vwan1)
Wed Oct  6 12:18:28 2021 user.notice mwan3track[27895]: Interface vwan1 (pppoe-vwan1) is offline
Wed Oct  6 12:18:28 2021 user.notice ddns-scripts[29642]: 3322: PID '29642' terminated by 'SIGTERM' at 2021-10-06 12:18
Wed Oct  6 12:18:28 2021 user.notice ddns-scripts[29643]: jmfrl: PID '29643' terminated by 'SIGTERM' at 2021-10-06 12:18
Wed Oct  6 12:18:31 2021 user.info mwan3track[17304]: Check (ping) failed for target "www.baidu.com" on interface vwan2 (pppoe-vwan2)
Wed Oct  6 12:18:31 2021 user.info mwan3track[17304]: Check (ping) failed for target "114.114.114.114" on interface vwan2 (pppoe-vwan2)
Wed Oct  6 12:18:31 2021 user.info mwan3track[17304]: Check (ping) failed for target "119.29.29.29" on interface vwan2 (pppoe-vwan2)
Wed Oct  6 12:18:33 2021 user.notice : Added device handler type: 8021ad
Wed Oct  6 12:18:33 2021 user.notice : Added device handler type: 8021q
Wed Oct  6 12:18:33 2021 user.notice : Added device handler type: macvlan
Wed Oct  6 12:18:33 2021 user.notice : Added device handler type: veth
Wed Oct  6 12:18:33 2021 user.notice : Added device handler type: bridge
Wed Oct  6 12:18:33 2021 user.notice : Added device handler type: Network device
[... 204 lines cut ...]
Wed Oct  6 12:20:33 2021 cron.err crond[5477]: crond (busybox 1.33.1) started, log level 5
Wed Oct  6 12:20:33 2021 cron.err crond[5704]: crond (busybox 1.33.1) started, log level 5
Wed Oct  6 12:30:12 2021 user.warn ddns-scripts[29484]: pazu: Updating IP at DDNS provider failed - starting retry 1/0
Wed Oct  6 12:40:17 2021 user.warn ddns-scripts[29484]: pazu: Updating IP at DDNS provider failed - starting retry 2/0
Wed Oct  6 12:50:22 2021 user.warn ddns-scripts[29484]: pazu: Updating IP at DDNS provider failed - starting retry 3/0
Wed Oct  6 12:50:23 2021 user.err ddns-scripts[29484]: pazu: wget 错误代码: '8'
Wed Oct  6 12:50:23 2021 user.warn ddns-scripts[29484]: pazu: 传输失败 - 1/0 在 60 秒后重试
Wed Oct  6 13:01:28 2021 user.warn ddns-scripts[29484]: pazu: Updating IP at DDNS provider failed - starting retry 4/0
Wed Oct  6 13:11:35 2021 user.warn ddns-scripts[29484]: pazu: Updating IP at DDNS provider failed - starting retry 5/0
Wed Oct  6 13:21:41 2021 user.warn ddns-scripts[29484]: pazu: Updating IP at DDNS provider failed - starting retry 6/0
Wed Oct  6 13:31:46 2021 user.warn ddns-scripts[29484]: pazu: Updating IP at DDNS provider failed - starting retry 7/0
Wed Oct  6 13:41:52 2021 user.warn ddns-scripts[29484]: pazu: Updating IP at DDNS provider failed - starting retry 8/0
Wed Oct  6 13:41:54 2021 user.err ddns-scripts[29484]: pazu: wget 错误代码: '8'
Wed Oct  6 13:41:55 2021 user.warn ddns-scripts[29484]: pazu: 传输失败 - 1/0 在 60 秒后重试
Wed Oct  6 13:53:02 2021 user.warn ddns-scripts[29484]: pazu: Updating IP at DDNS provider failed - starting retry 9/0
Wed Oct  6 14:03:08 2021 user.warn ddns-scripts[29484]: pazu: Updating IP at DDNS provider failed - starting retry 10/0
Wed Oct  6 14:13:14 2021 user.warn ddns-scripts[29484]: pazu: Updating IP at DDNS provider failed - starting retry 11/0
Wed Oct  6 14:23:20 2021 user.warn ddns-scripts[29484]: pazu: Updating IP at DDNS provider failed - starting retry 12/0
Wed Oct  6 14:23:22 2021 user.err ddns-scripts[29484]: pazu: wget 错误代码: '8'
Wed Oct  6 14:23:23 2021 user.warn ddns-scripts[29484]: pazu: 传输失败 - 1/0 在 60 秒后重试
Wed Oct  6 14:24:23 2021 user.err ddns-scripts[29484]: pazu: wget 错误代码: '8'
Wed Oct  6 14:24:24 2021 user.warn ddns-scripts[29484]: pazu: 传输失败 - 2/0 在 60 秒后重试
Wed Oct  6 14:30:05 2021 daemon.err nlbwmon[3418]: Netlink receive failure: Out of memory
Wed Oct  6 14:30:05 2021 daemon.err nlbwmon[3418]: Unable to dump conntrack: No buffer space available
Wed Oct  6 14:35:29 2021 user.warn ddns-scripts[29484]: pazu: Updating IP at DDNS provider failed - starting retry 13/0

feng5064, 2021-10-08:
> 你也是intel的网卡?

的确是intel的网卡，主板板载的和外加的都是intel的网卡。这跟intel网卡有关？
但我觉得宿主机的OP还是很稳的，就是虚拟机有时会断网，而且是断网后再也连不上那种。

coolsnowwolf, 2026-09-06:
日志中的 netifd segfault 发生在 2021 年双拨 PPPoE 断线场景；当前 netifd 已经过多轮更新，旧版崩溃问题无法在现行代码复现，关闭为 completed。

closed 2026-09-06 by coolsnowwolf