coolsnowwolf/lede #8009: 无法休眠,请大佬指点

kl9800, 2021-10-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

x86，sata硬盘，硬盘成功识别，尚未copy任何资料到硬盘里，开启hd-idle，设置10分钟，但都过去20分钟了，硬盘无法进入休眠状态，请问我该如何解决，强调一下，我不是虚拟机环境，我的lede系统在U盘里，sata接口直通的其他硬盘，整套主机完全是给lede用的，没有任何虚拟环境干扰。

### (2) 路由器型号和固件版本

x86 

### (3) 详细日志

A：

kl9800, 2021-10-10:
自问自答。。最终的解决方案是抛弃hd-idle，改用hdparm，可以成功，用hd-idle怎么都不行。。也许是自己手太潮懂得少。。总之抛弃了hd-idle就好了。。

kl9800, 2021-10-10:
可惜 hdparm只能休眠一次，唤醒后就无法再次自动休眠，
所以又重新启用 hd-idle，
最终命令是：
hd-idle -a /dev/sda1 -i 600 -a /dev/sdb1 -i 600 -a /dev/sdc1 -i 600 -a /dev/sdd1 -i 600 -a /dev/sde1 -i 600
完美解决，实测唤醒后可以再次自动休眠

WYC-2020, 2021-10-11:
@kl9800 这个不知道luci 下面的hd-ilde的默认启动参数有没关系，他的默认启动参数是/usr/bin/hd-idle -d -i 0 -a sda -i 600  这里面有一个-i 0 ,查询说是这么一个意思
[image]
 -d 是debug模式

kl9800, 2021-10-11:
> @kl9800 这个不知道luci 下面的hd-ilde的默认启动参数有没关系，他的默认启动参数是/usr/bin/hd-idle -d -i 0 -a sda -i 600 这里面有一个-i 0 ,查询说是这么一个意思 [image] -d 是debug模式

亲测只去掉 -d不行，-i 0也一起去掉就可以了，但是即使这样，其中一块盘被唤醒后也无法再次休眠，但这不是必现，是有机率，再试一次就没问题了，现在就一直保持这样用了，也无大碍，lede当一个常年开机的网盘用，不敢弄群晖，感觉群晖的休眠更玄学...