coolsnowwolf/lede #8044: AX3600的2

devioa, 2021-10-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

xiaopang0v0, 2021-10-15:
应该是因为你没有设备接入进去吧，openwrt没有设备接入进去的话，信号栏就是空的

devioa, 2021-10-15:
> 应该是因为你没有设备接入进去吧，openwrt没有设备接入进去的话，信号栏就是空的

你再好好看看

devioa, 2021-10-15:
> > > > > 应该是因为你没有设备接入进去吧，openwrt没有设备接入进去的话，信号栏就是空的
> > > > 
> > > > 
> > > > 你再好好看看
> > > 
> > > 
> > > 好吧，大意了，你看看是不是跟访客WiFi冲突了呢？
> > 
> > 
> > 没创建那个访客ssid之前也是一样，再者我没装访客网络那个luci-app，我手动创建的那个ssid压根没启用
> 
> 这样，你先备份现在的设置，然后把所有的2.4GWiFi删了，包括lot的，然后在非lot下新建WiFi，应该就能启动2.4G的那个热点了

不是，我觉得你可能题没审对，我的意思是它显示有问题，但实际使用没有任何问题，热点一直都能正常启用，所以我觉得这是个显示BUG

xiaopang0v0, 2021-10-15:
> > > > > > 应该是因为你没有设备接入进去吧，openwrt没有设备接入进去的话，信号栏就是空的
> > > > > 
> > > > > 
> > > > > 你再好好看看
> > > > 
> > > > 
> > > > 好吧，大意了，你看看是不是跟访客WiFi冲突了呢？
> > > 
> > > 
> > > 没创建那个访客ssid之前也是一样，再者我没装访客网络那个luci-app，我手动创建的那个ssid压根没启用
> > 
> > 
> > 这样，你先备份现在的设置，然后把所有的2.4GWiFi删了，包括lot的，然后在非lot下新建WiFi，应该就能启动2.4G的那个热点了
> 
> 不是，我觉得你可能题没审对，我的意思是它显示有问题，但实际使用没有任何问题，热点一直都能正常启用，所以我觉得这是个显示BUG 不过还是谢谢你了哈，看看有没有大佬修复这个BUG吧

其实我现在还是不能理解，路由器有三个可用的SSID，分别是5G、2.4G常规、2.4G alot，你的意思是这 三个 都可以用是吧，如果是这个意思，那还是建议你修改一下描述

xiaopang0v0, 2021-10-15:
哦，那就让开发者修一下就行了，我还以为是alot能用，2.4G没法使用呢

darkjackx, 2021-11-15:
> > > > > > > > 应该是因为你没有设备接入进去吧，openwrt没有设备接入进去的话，信号栏就是空的
> > > > > > > 
> > > > > > > 
> > > > > > > 你再好好看看
> > > > > > 
> > > > > > 
> > > > > > 好吧，大意了，你看看是不是跟访客WiFi冲突了呢？
> > > > > 
> > > > > 
> > > > > 没创建那个访客ssid之前也是一样，再者我没装访客网络那个luci-app，我手动创建的那个ssid压根没启用
> > > > 
> > > > 
> > > > 这样，你先备份现在的设置，然后把所有的2.4GWiFi删了，包括lot的，然后在非lot下新建WiFi，应该就能启动2.4G的那个热点了
> > > 
> > > 
> > > 不是，我觉得你可能题没审对，我的意思是它显示有问题，但实际使用没有任何问题，热点一直都能正常启用，所以我觉得这是个显示BUG 不过还是谢谢你了哈，看看有没有大佬修复这个BUG吧
> > 
> > 
> > 其实我现在还是不能理解，路由器有三个可用的SSID，分别是5G、2.4G常规、2.4G alot，你的意思是这 三个 都可以用是吧，如果是这个意思，那还是建议你修改一下描述
> 
> 我说的是**显示**有问题，功能都正常

我用小米 AX9000 也出現 2.4G WiFi LuCi界面信号一直显示的是0%，传输速率也一直显示的是 1 Mbit/s 的問題

closed 2022-10-31 by aiamadeus