coolsnowwolf/lede #8046: 使用MT7620芯片的小米Mini刷入最新lede后会 有线、无线 一起间歇性断网

xiaopang0v0, 2021-10-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：路由器间隔一段时间就会出现断网的现象，有线、无线一起中断，ping路由器无法ping通
我百度了一下，似乎是交换机驱动的问题，请问是这个原因吗？
如果是这个原因，那我应该如何解决呢？麻烦你们了

### (2) 路由器型号和固件版本

型号：小米Mini
固件版本：OpenWrt R21.10.1 / LuCI Master (git-21.280.14233-9b796b1)
内核版本：5.4.150

### (3) 详细日志

系统日志：
Fri Oct 15 09:20:08 2021 kern.warn kernel: [ 4189.971081] ------------[ cut here ]------------
Fri Oct 15 09:20:08 2021 kern.warn kernel: [ 4189.975841] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:473 dev_watchdog+0x26c/0x274
Fri Oct 15 09:20:08 2021 kern.info kernel: [ 4189.984303] NETDEV WATCHDOG: eth0 (ralink_soc_eth): transmit queue 0 timed out
Fri Oct 15 09:20:08 2021 kern.warn kernel: [ 4189.991682] Modules linked in: xt_FULLCONENAT rt2800soc rt2800mmio rt2800lib pppoe ppp_async iptable_nat xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT wireguard rt2x00soc rt2x00mmio rt2x00lib pppox ppp_mppe ppp_generic nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_nat nf_flow_table_hw nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_amanda nf_conntrack mt76x2e mt76x2_common mt76x02_lib mt76 mac80211 libchacha20poly1305 libblake2s ipt_REJECT cfg80211 xt_time xt_tcpudp xt_socket xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TPROXY xt_TCPMSS xt_LOG ts_kmp ts_fsm ts_bm slhc poly1305_mips nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 macvlan libcurve25519_generic libblake2s_generic iptable_raw iptable_mangle
Fri Oct 15 09:20:08 2021 kern.warn kernel: [ 4189.991948]  iptable_filter ip_tables crc_ccitt compat chacha_mips asn1_decoder cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ip6_udp_tunnel udp_tunnel crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha256_generic libsha256 sha1_generic seqiv jitterentropy_rng drbg hmac ghash_generic gf128mul gcm ecb ctr cmac ccm authenc arc4 leds_gpio ohci_platform ohci_hcd fsl_mph_dr_of ehci_platform ehci_fsl ehci_hcd gpio_button_hotplug usbcore nls_base usb_common aead cryptomgr crypto_null crypto_hash
Fri Oct 15 09:20:08 2021 kern.warn kernel: [ 4190.160374] CPU: 0 PID: 0 Comm: swapper Not tainted 5.4.150 #0
Fri Oct 15 09:20:08 2021 kern.warn kernel: [ 4190.166330] Stack : 805f0000 8058dc60 00000000 00000000 8058cd58 87c0be1c 805c5120 805c4d03
Fri Oct 15 09:20:08 2021 kern.warn kernel: [ 4190.174876]         805379a4 00000000 807233cc 805e0000 00200000 00000001 87c0bdd0 48aa789b
Fri Oct 15 09:20:08 2021 kern.warn kernel: [ 4190.183416]         00000000 00000000 80750000 00000000 00000030 00000103 312e342e 23203035
Fri Oct 15 09:20:08 2021 kern.warn kernel: [ 4190.191963]         00000000 000000d0 00000000 00027276 00000000 00000009 00000000 8037f084
Fri Oct 15 09:20:08 2021 kern.warn kernel: [ 4190.200494]         00000009 805e0000 00200000 00000122 00000001 802ae710 00000000 80720000
Fri Oct 15 09:20:08 2021 kern.warn kernel: [ 4190.209033]         ...
Fri Oct 15 09:20:08 2021 kern.warn kernel: [ 4190.211535] Call Trace:
Fri Oct 15 09:20:08 2021 kern.warn kernel: [ 4190.214049] [<8000b3d0>] show_stack+0x30/0x100
Fri Oct 15 09:20:08 2021 kern.warn kernel: [ 4190.218601] [<80025988>] __warn+0xc0/0x10c
Fri Oct 15 09:20:08 2021 kern.warn kernel: [ 4190.222797] [<80025a60>] warn_slowpath_fmt+0x8c/0xac
Fri Oct 15 09:20:08 2021 kern.warn kernel: [ 4190.227869] [<8037f084>] dev_watchdog+0x26c/0x274
Fri Oct 15 09:20:08 2021 kern.warn kernel: [ 4190.232709] [<80069b94>] call_timer_fn.isra.34+0x20/0x90
Fri Oct 15 09:20:08 2021 kern.warn kernel: [ 4190.238134] [<80069da8>] run_timer_softirq+0x1a4/0x1ec
Fri Oct 15 09:20:08 2021 kern.warn kernel: [ 4190.243399] [<804b0ae8>] __do_softirq+0x110/0x298
Fri Oct 15 09:20:08 2021 kern.warn kernel: [ 4190.248204] [<80005e90>] except_vec_vi_end+0xb8/0xc4
Fri Oct 15 09:20:08 2021 kern.warn kernel: [ 4190.253284] [<804b08b4>] r4k_wait_irqoff+0x18/0x24
Fri Oct 15 09:20:08 2021 kern.warn kernel: [ 4190.258175] ---[ end trace 9f0c1892de5c4ba2 ]---
Fri Oct 15 09:20:08 2021 kern.err kernel: [ 4190.262908] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
Fri Oct 15 09:20:08 2021 kern.info kernel: [ 4190.269484] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
Fri Oct 15 09:20:08 2021 kern.info kernel: [ 4190.275899] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=06b20000, max=1024, ctx=146, dtx=146, fdx=103, next=146
Fri Oct 15 09:20:08 2021 kern.info kernel: [ 4190.287185] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=06b0c000, max=1024, calc=856, drx=873
Fri Oct 15 09:20:14 2021 kern.err kernel: [ 4192.319379] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
Fri Oct 15 09:20:14 2021 kern.info kernel: [ 4192.325981] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
Fri Oct 15 09:20:14 2021 kern.info kernel: [ 4192.332413] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=06b20000, max=1024, ctx=319, dtx=319, fdx=64, next=319
Fri Oct 15 09:20:14 2021 kern.info kernel: [ 4192.343609] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=06b00000, max=1024, calc=406, drx=407
Fri Oct 15 09:20:24 2021 kern.err kernel: [ 4195.760491] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
Fri Oct 15 09:20:24 2021 kern.info kernel: [ 4195.767092] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
Fri Oct 15 09:20:24 2021 kern.info kernel: [ 4195.773525] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=06b20000, max=1024, ctx=49, dtx=49, fdx=32, next=49
Fri Oct 15 09:20:24 2021 kern.info kernel: [ 4195.784454] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=06b0c000, max=1024, calc=229, drx=230
Fri Oct 15 09:20:34 2021 kern.err kernel: [ 4199.289017] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
Fri Oct 15 09:20:34 2021 kern.info kernel: [ 4199.295614] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
Fri Oct 15 09:20:34 2021 kern.info kernel: [ 4199.302072] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=06b20000, max=1024, ctx=42, dtx=42, fdx=32, next=42
Fri Oct 15 09:20:34 2021 kern.info kernel: [ 4199.313014] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=06b00000, max=1024, calc=264, drx=266
内核日志：
[ 4189.971081] ------------[ cut here ]------------
[ 4189.975841] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:473 dev_watchdog+0x26c/0x274
[ 4189.984303] NETDEV WATCHDOG: eth0 (ralink_soc_eth): transmit queue 0 timed out
[ 4189.991682] Modules linked in: xt_FULLCONENAT rt2800soc rt2800mmio rt2800lib pppoe ppp_async iptable_nat xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT wireguard rt2x00soc rt2x00mmio rt2x00lib pppox ppp_mppe ppp_generic nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_nat nf_flow_table_hw nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_amanda nf_conntrack mt76x2e mt76x2_common mt76x02_lib mt76 mac80211 libchacha20poly1305 libblake2s ipt_REJECT cfg80211 xt_time xt_tcpudp xt_socket xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TPROXY xt_TCPMSS xt_LOG ts_kmp ts_fsm ts_bm slhc poly1305_mips nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 macvlan libcurve25519_generic libblake2s_generic iptable_raw iptable_mangle
[ 4189.991948]  iptable_filter ip_tables crc_ccitt compat chacha_mips asn1_decoder cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_rejec [... cut ...]
[ 4195.773525] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=06b20000, max=1024, ctx=49, dtx=49, fdx=32, next=49
[ 4195.784454] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=06b0c000, max=1024, calc=229, drx=230
[ 4199.289017] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
[ 4199.295614] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
[ 4199.302072] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=06b20000, max=1024, ctx=42, dtx=42, fdx=32, next=42
[ 4199.313014] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=06b00000, max=1024, calc=264, drx=266

xiaopang0v0, 2021-10-16:
> 好像是内核的问题，退回140之前试试？

请问要如何只退回内核呢？是修改makefile吗？还是说直接退回lede的版本，退回到使用140内核之前的版本？

LINGJP, 2021-10-19:
我的k2也有这个问题，就是间歇性重启，目前找不到原因，降级到4.14.x好点了

xiaopang0v0, 2021-10-20:
> 我已经出现了。退回了我我原来用的 R21.4.18 了

> 我的k2也有这个问题，就是间歇性重启，目前找不到原因，降级到4.14.x好点了

经测试，最新的5.10.74内核也存在间歇性断网的原因，查看日志结果跟上面是一样的，现在准备切到以前的提交，重新编译试试老版本看有没有这个问题

hcym, 2021-10-20:
明天到货，瑟瑟发抖，官方只能用master的也烦人

xiaopang0v0, 2021-10-21:
[image]
5.4.140测试有相同问题，正准备继续下降

JusticeNet, 2021-10-30:
有解决的吗？

xiaopang0v0, 2021-11-09:
> 有解决的吗？

现在应该无解吧，我编译刷机弄了一两个月了，我现在直接放弃了，我去用lienol的源码了，lienol的源码会重启，但是wifi还是正常的

xiaopang0v0, 2022-02-27:
> > [image] 5.4.140测试有相同问题，正准备继续下降
> 
> 收到 我也准备回4.14

5.4.124就没有这个问题了，现在准备长时间使用，测试一下稳定性
[image]

xiaopang0v0, 2022-03-13:
回退到2021年8月21号编译的固件版本是R21.8.6，内核版本是5.4.138，会出现断流的问题
[image]

Cathgao, 2022-03-13:
手上的MT762X都有断流问题，newifimini的MT7620、新3的MT7621

LINGJP, 2022-03-13:
玩4.14＋闭源驱动

xiaopang0v0, 2022-03-14:
R21.4.8,Linux内核5.4.113，没有问题
[image]

xiaopang0v0, 2022-03-14:
> 玩4.14＋闭源驱动

请问怎么编译闭源驱动呢？

ixinshang, 2022-04-10:
> 

请问一下，怎么切换到这个版本呢？

xiaopang0v0, 2022-04-10:
> > 
> 
> 请问一下，怎么切换到这个版本呢？

先克隆lede的分支，然后使用`git checkout 041d7abc4aa47d732b264dc50973555af5cf980e`切换到最后一个带4.14内核的版本
然后编辑feeds.conf.default
```
src-git packages https://github.com/coolsnowwolf/packages^428cbdc6cb790fc50d494a8397c38df692b39c48.git
src-git luci https://github.com/coolsnowwolf/luci^974fb045f86ac0f3667eff6a6b973434db707dba
src-git routing https://git.openwrt.org/feed/routing.git
src-git telephony https://git.openwrt.org/feed/telephony.git
src-git kenzo https://github.com/kenzok8/openwrt-packages
src-git small https://github.com/kenzok8/small
src-git small8 https://github.com/kenzok8/small-package
```
然后就按照教程的操作编译就可以了

chobitsos, 2022-05-22:
已换SDK
测试30分钟双向跑满没有断流，适配lean的packges仓库
[image]

[image]

[image]

airlander, 2022-06-24:
> 先克隆lede的分支，然后使用`git checkout 041d7abc4aa47d732b264dc50973555af5cf980e`切换到最后一个带4.14内核的版本 然后编辑feeds.conf.default
>
> ```
> src-git packages [...]

如何找到合适的commitid呀，尤其是package的commitid，头大，想找到5.4.124的lede大版本

coolsnowwolf, 2026-09-05:
小米 Mini 的断网日志来自旧版 MT7620/5.4.150 `ralink_soc_eth` watchdog，当前目标和内核已演进，缺少现行版本复现与硬件对照，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf