coolsnowwolf/lede #8063: R4s编译错误

capxax, 2021-10-16:
arch/arm64/boot/dts/rockchip/rk3328-doornet1.dts:10:10: fatal error: rk3328-dram-default-timing.dtsi: No such file or directory
   10 | #include "rk3328-dram-default-timing.dtsi"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[7]: *** [scripts/Makefile.lib:285: arch/arm64/boot/dts/rockchip/rk3328-doornet1.dtb] Error 1
make[6]: *** [scripts/Makefile.build:497: arch/arm64/boot/dts/rockchip] Error 2
make[5]: *** [Makefile:1274: dtbs] Error 2
make[5]: Leaving directory '/home/wanghao/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.150'
make[4]: *** [Makefile:26: /home/wanghao/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.150/.modules] Error 2
make[4]: Leaving directory '/home/wanghao/r4s/lede/target/linux/rockchip'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/wanghao/r4s/lede/target/linux'
time: target/linux/compile#10.71#6.17#16.62
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/compile] Error 1
make[2]: Leaving directory '/home/wanghao/r4s/lede'
make[1]: *** [target/Makefile:18: /home/wanghao/r4s/lede/staging_dir/target-aarch64_generic_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/wanghao/r4s/lede'
make: *** [/home/wanghao/r4s/lede/include/toplevel.mk:230：world] 错误 2

8827974, 2021-10-16:
insn-automata.c:155450:1: fatal error: error writing to /home/yang/lede/tmp/cc5fWfPM.s: No space left on device
 }
 ^
compilation terminated.
Makefile:1110: recipe for target 'insn-automata.o' failed
make[5]: *** [insn-automata.o] Error 1
make[5]: Leaving directory '/home/yang/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0-final/gcc'
Makefile:4270: recipe for target 'all-gcc' failed
make[4]: *** [all-gcc] Error 2
make[4]: Leaving directory '/home/yang/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0-final'
Makefile:905: recipe for target 'all' failed
make[3]: *** [all] Error 2
make[3]: Leaving directory '/home/yang/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0-final'
Makefile:89: recipe for target '/home/yang/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0-final/.built' failed
make[2]: *** [/home/yang/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0-final/.built] Error 2
make[2]: Leaving directory '/home/yang/lede/toolchain/gcc/final'
time: toolchain/gcc/final/compile#205.34#26.43#288.21
    ERROR: toolchain/gcc/final failed to build.
toolchain/Makefile:95: recipe for target 'toolchain/gcc/final/compile' failed
make[1]: *** [toolchain/gcc/final/compile] Error 1
make[1]: Leaving directory '/home/yang/lede'
Build failed - please re-run with -j1 to see the real error message
/home/yang/lede/include/toplevel.mk:228: recipe for target 'toolchain/compile' failed
make: *** [toolchain/compile] Error 1

8827974, 2021-10-16:
我是这个错误

KLT007, 2021-10-16:
https://github.com/coolsnowwolf/lede/pull/8062

capxax, 2021-10-16:
> 已修复了 #8062 重新编译吧，如果是本地重新，重新拉

THX