coolsnowwolf/lede #8065: mariadb-connector-c-3.1.12-src 在线编译报错

Dimitri2020007, 2021-10-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：x86_64

### (3) 详细日志

A：

2021-10-16T13:14:05.2132342Z mkdir -p /workdir/openwrt/dl
2021-10-16T13:14:05.2167157Z SHELL= flock /workdir/openwrt/tmp/.mariadb-connector-c-3.1.12-src.tar.gz.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "mariadb-connector-c-3.1.12-src.tar.gz" "2f5ae14708b4813e4ff6857d152c22e6fc0e551c9fa743c1ef81a68e3254fe63" "" "https://mirror.netcologne.de/mariadb/connector-c-3.1.12" "https://mirror.lstn.net/mariadb/connector-c-3.1.12" "https://ftp.yz.yamagata-u.ac.jp/pub/dbms/mariadb/connector-c-3.1.12" "https://downloads.mariadb.org/interstitial/connector-c-3.1.12"    '
2021-10-16T13:14:05.2648445Z rm -f /workdir/openwrt/build_dir/target-x86_64_musl/mariadb-connector-c-3.1.12-src/.configured_*
2021-10-16T13:14:05.2676398Z rm -f /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.libmariadb_installed
2021-10-16T13:14:05.2717795Z mkdir -p /workdir/openwrt/build_dir/target-x86_64_musl/mariadb-connector-c-3.1.12-src
2021-10-16T13:14:05.2821009Z (cd /workdir/openwrt/build_dir/target-x86_64_musl/mariadb-connector-c-3.1.12-src; CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/mariadb-connector-c-3.1.12-src=mariadb-connector-c-3.1.12-src -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/workdir/openwrt/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/include -I/workdir/openwrt/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/include -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include -I/workdir/openwrt/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/include -I/workdir/openwrt/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/include " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/mariadb-connector-c-3.1.12-src=mariadb-connector-c-3.1.12-src -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/workdir/openwrt/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/include -I/workdir/openwrt/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/include -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include -I/workdir/openwrt/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/include -I/workdir/openwrt/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/include " LDFLAGS="-L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro -L/workdir/openwrt/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/lib -Wl,-rpath-link=/workdir/openwrt/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/lib -L/workdir/openwrt/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/lib -Wl,-rpath-link=/workdir/openwrt/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/lib " cmake -DCMAKE_SYSTEM_NAME=Linux -DCMAKE_SYSTEM_VERSION=1 -DCMAKE_SYSTEM_PROCESSOR=x86_64 -DCMAKE_BUILD_TYPE=Release -DCMAKE_C_FLAGS_RELEASE="-DNDEBUG" -DCMAKE_CXX_FLAGS_RELEASE="-DNDEBUG" -DCMAKE_C_COMPILER_LAUNCHER="" -DCMAKE_C_COMPILER="/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/bin/x86_64-openwrt-linux-musl-gcc" -DCMAKE_CXX_COMPILER_LAUNCHER="" -DCMAKE_CXX_COMPILER="/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/bin/x86_64-openwrt-linux-musl-g++" -DCMAKE_ASM_COMPILER_LAUNCHER="" -DCMAKE_ASM_COMPILER="/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/bin/x86_64-openwrt-linux-musl-gcc" -DCMAKE_EXE_LINKER_FLAGS:STRING="-L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro -L/workdir/openwrt/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/lib -Wl,-rpath-link=/workdir/openwrt/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/lib -L/workdir/openwrt/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/lib -Wl,-rpath-link=/workdir/openwrt/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/lib" -DCMAKE_MODULE_LINKER_FLAGS:STRING="-L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro -L/workdir/openwrt/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/lib -Wl,-rpath-link=/workdir/openwrt/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/lib -L/workdir/openwrt/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/lib -Wl,-rpath-link=/workdir/openwrt/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/lib -Wl,-Bsymbolic-functions" -DCMAKE_SHARED_LINKER_FLAGS:STRING="-L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro -L/workdir/openwrt/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/lib -Wl,-rpath-link=/workdir/openwrt/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/lib -L/workdir/openwrt/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/lib -Wl,-rpath-link=/workdir/openwrt/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/lib -Wl,-Bsymbolic-functions" -DCMAKE_AR="/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/bin/x86_64-openwrt-linux-musl-gcc-ar" -DCMAKE_NM="/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/bin/x86_64-openwrt-linux-musl-gcc-nm" -DCMAKE_RANLIB="/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/bin/x86_64-openwrt-linux-musl-gcc-ranlib" -DCMAKE_FIND_ROOT_PATH="/workdir/openwrt/staging_dir/target-x86_64_musl/usr;/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl" -DCMAKE_FIND_ROOT_PATH_MODE_PROGRAM=BOTH -DCMAKE_FIND_ROOT_PATH_MODE_LIBRARY=ONLY -DCMAKE_FIND_ROOT_PATH_MODE_INCLUDE=ONLY -DCMAKE_STRIP=: -DCMAKE_INSTALL_PREFIX=/usr -DDL_LIBRARY=/workdir/openwrt/staging_dir/target-x86_64_musl -DCMAKE_PREFIX_PATH=/workdir/openwrt/staging_dir/target-x86_64_musl -DCMAKE_SKIP_RPATH=TRUE -DCMAKE_EXPORT_PACKAGE_REGISTRY=FALSE -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=TRUE -DCMAKE_FIND_USE_PACKAGE_REGISTRY=FALSE -DCMAKE_FIND_PACKAGE_NO_PACKAGE_REGISTRY=TRUE -DCMAKE_FIND_USE_SYSTEM_PACKAGE_REGISTRY=FALSE -DCMAKE_FIND_PACKAGE_NO_SYSTEM_PACKAGE_REGISTRY=TRUE -DCMAKE_GENERATOR="Ninja" -DWITH_UNIT_TESTS=0 -DCMAKE_CROSSCOMPILING=1 -DINSTALL_INCLUDEDIR=include/mysql -DINSTALL_LIBDIR=lib -DINSTALL_PLUGINDIR=lib/mariadb/plugin -DMARIADB_UNIX_ADDR=/var/run/mysqld/mysqld.sock -DWITH_EXTERNAL_ZLIB=YES -DWITH_MYSQLCOMPAT=ON -DWITH_SSL=OPENSSL -DICONV_INCLUDE_DIR=/workdir/openwrt/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/include -DICONV_LIBRARIES=/workdir/openwrt/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/lib/libiconv.a -DCLIENT_PLUGIN_AUTH_GSSAPI_CLIENT=OFF -DCLIENT_PLUGIN_REMOTE_IO=OFF /workdir/openwrt/build_dir/target-x86_64_musl/mariadb-connector-c-3.1.12-src )
2021-10-16T13:14:05.2880404Z CMake Deprecation Warning at CMakeLists.txt:5 (CMAKE_MINIMUM_REQUIRED):
2021-10-16T13:14:05.2881201Z   Compatibility with CMake < 2.8.12 will be removed from a future version of
2021-10-16T13:14:05.2881740Z   CMake.
2021-10-16T13:14:05.2881962Z 
2021-10-16T13:14:05.2882432Z   Update the VERSION argument <min> value or use a ...<max> suffix to tell
2021-10-16T13:14:05.2883169Z   CMake that the project does not need compatibility with older versions.
2021-10-16T13:14:05.2883616Z 
2021-10-16T13:14:05.2883809Z 
2021-10-16T13:14:05.4033181Z CMake Error at cmake/ConnectorName.cmake:30 (ENDMACRO):
2021-10-16T13:14:05.4034139Z   Flow control statements are not properly nested.
2021-10-16T13:14:05.4034685Z Call Stack (most recent call first):
2021-10-16T13:14:05.4035245Z   CMakeLists.txt:423 (INCLUDE)
2021-10-16T13:14:05.4035552Z 
2021-10-16T13:14:05.4035749Z 
2021-10-16T13:14:05.4041500Z -- Configuring incomplete, errors occurred!
[... cut ...]
2021-10-16T13:14:05.4090222Z make[2]: *** [package/Makefile:116: package/feeds/packages/libmariadb/compile] Error 1
2021-10-16T13:14:05.4105397Z make[2]: Leaving directory '/workdir/openwrt'
2021-10-16T13:14:05.4154936Z make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2021-10-16T13:14:05.4155898Z make[1]: Leaving directory '/workdir/openwrt'
2021-10-16T13:14:05.4179600Z make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
2021-10-16T13:14:05.4370476Z ##[error]Process completed with exit code 2.

Boos4721, 2021-10-17:
JUST UPDATE **CMakeLists.txt** TO 2.8.12

Boos4721, 2021-10-17:
[image]
ya, fixed with https://github.com/coolsnowwolf/packages/pull/257

Dimitri2020007, 2021-10-17:
ok i know

closed 2021-11-02 by AmadeusGhost