coolsnowwolf/lede #8066: ERROR: tools/m4 failed to build.

xiangfeidexiaohuo, 2021-10-17:
这个怎么报错了 好突然

make[7]: *** [Makefile:1910: c-stack.o] Error 1
make[7]: *** Waiting for unfinished jobs....
make[7]: Leaving directory '/home/yaodao/lede/build_dir/host/m4-1.4.18/lib'
make[6]: *** [Makefile:1674: all] Error 2
make[6]: Leaving directory '/home/yaodao/lede/build_dir/host/m4-1.4.18/lib'
make[5]: *** [Makefile:1572: all-recursive] Error 1
make[5]: Leaving directory '/home/yaodao/lede/build_dir/host/m4-1.4.18'
make[4]: *** [Makefile:1528: all] Error 2
make[4]: Leaving directory '/home/yaodao/lede/build_dir/host/m4-1.4.18'
make[3]: *** [Makefile:30: /home/yaodao/lede/build_dir/host/m4-1.4.18/.built] Error 2
make[3]: Leaving directory '/home/yaodao/lede/tools/m4'
time: tools/m4/compile#0.46#0.35#0.30
    ERROR: tools/m4 failed to build.
make[2]: *** [tools/Makefile:159: tools/m4/compile] Error 1
make[2]: *** Waiting for unfinished jobs....
make[3]: Leaving directory '/home/yaodao/lede/tools/mkimage'
time: tools/mkimage/compile#0.06#0.15#0.16
make[3]: Leaving directory '/home/yaodao/lede/tools/cmake'
time: tools/cmake/compile#0.06#0.18#0.19
make[2]: Leaving directory '/home/yaodao/lede'
make[1]: *** [tools/Makefile:155: /home/yaodao/lede/staging_dir/host/stamp/.tools_compile_yyynyynnyyyynyyyyyynyynnyyynyyyyyyyyyyyyyyyyynynnyyyyyyyyyy] Error 2
make[1]: Leaving directory '/home/yaodao/lede'
make: *** [/home/yaodao/lede/include/toplevel.mk:230：world] 错误 2

Boos4721, 2021-10-17:
UPDATE TO 1.4.19

xiangfeidexiaohuo, 2021-10-17:
> UPDATE TO 1.4.19

应该不是这个错误，错误了 再运行，就换了一个error，但是 每次都有lede/staging_dir/host/stamp/.tools_compile_yyynyynnyyyynyyyyyynyynnyyynyyyyyyyyyyyyyyyyynynnyyyyyyyyyy] Error 2

closed 2021-10-17 by xiangfeidexiaohuo