coolsnowwolf/lede #8069: 红米ax6更新到最新版2.4g出现问题

AXBNG, 2021-10-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：红米ax6从5.10.70升级到5.10.73之后，2.4g经常出现连接上但无法上网的情况，断开重连后恢复正常，5g没有问题

### (2) 路由器型号和固件版本

A：红米ax6    最新版

### (3) 详细日志

A：

Boos4721, 2021-10-17:
[image]
[image]
no issuse with Linux 5.10.74 on AX3600, iOS and Android.

AXBNG, 2021-10-17:
[image]

在connected和disconnected之间反复横跳